ETLCPP/etl #6: Build with IAR failed

DaveBian, 2015-10-29:
Hi,

When I built etl iar project, it failed and give message list below: 
IAR version: IAR Embedded Workbench for ARM, 7.40.2.8570
How to fix this problem? Thanks a lot.

Building configuration: etl - Debug 
Updating build tree... 
test_compile.cpp  
Error[Pe065]: expected a ";" C:\01.Projets\03.RefCodes\etl\algorithm.h 296 
Error[Pe020]: identifier "n" is undefined C:\01.Projets\03.RefCodes\etl\algorithm.h 298 
Error[Pe065]: expected a ";" C:\01.Projets\03.RefCodes\etl\algorithm.h 323 
Error[Pe020]: identifier "n" is undefined C:\01.Projets\03.RefCodes\etl\algorithm.h 325 
Error[Pe065]: expected a ";" C:\01.Projets\03.RefCodes\etl\algorithm.h 354 
Error[Pe020]: identifier "n" is undefined C:\01.Projets\03.RefCodes\etl\algorithm.h 356 
Error[Pe065]: expected a ";" C:\01.Projets\03.RefCodes\etl\algorithm.h 381 
Error[Pe020]: identifier "n" is undefined C:\01.Projets\03.RefCodes\etl\algorithm.h 383 
Warning[Pe940]: missing return statement at end of non-void function "etl::io_port_rw<T, ADDRESS>::iterator::operator=" C:\01.Projets\03.RefCodes\etl\io_port.h 76 
Error[Pe519]: class "etl::crc8_ccitt" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 188 
Error[Pe519]: class "etl::crc8_ccitt" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 189 
Error[Pe276]: name followed by "::" must be a class or namespace name C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 189 
Error[Pe519]: class "etl::crc16" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 191 
Error[Pe519]: class "etl::crc16" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 192 
Error[Pe276]: name followed by "::" must be a class or namespace name C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 192 
Error[Pe519]: class "etl::crc16_ccitt" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 194 
Error[Pe519]: class "etl::crc16_ccitt" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 195 
Error[Pe276]: name followed by "::" must be a class or namespace name C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 195 
Error[Pe519]: class "etl::crc16_kermit" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 197 
Error[Pe519]: class "etl::crc16_kermit" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 198 
Error[Pe276]: name followed by "::" must be a class or namespace name C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 198 
Error[Pe519]: class "etl::crc32" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 200 
Error[Pe519]: class "etl::crc32" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 201 
Error[Pe276]: name followed by "::" must be a class or namespace name C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 201 
Error[Pe519]: class "etl::crc64_ecma" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 203 
Error[Pe519]: class "etl::crc64_ecma" may not have a template argument list C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 204 
Error[Pe276]: name followed by "::" must be a class or namespace name C:\01.Projets\03.RefCodes\etl\test\test_compile.cpp 204 
Error while running C/C++ Compiler 
 
Total number of errors: 26 
Total number of warnings: 1

DaveBian, 2015-11-09:
Could some tell me how fix this issue? Thanks in advance.

jwellbelove, 2015-11-12:
IAR seems to be having a problem with std::iterator_traits<TIterator2>::difference_type
Are you able to confirm whether this type is supported by IAR? Maybe you could try substituting 'int32_t' or 'long' for the type.

I don't have access to my PC at the moment as I'm in Australia staying at campsites with almost completely useless WIFI. 've only been able to access email twice in two weeks. I shall be back in the UK on Nov 27th and I'll take a closer look when I've got over the jet lag!

One of the jobs I have to finish is to update the test files for the embedded compiler projects as they do not cover any near as much as the unit tests do. There are bound to be a few more cross platform issues due to each compiler interpreting the standard in slightly different ways. Testing for the Green Hills compiler is virtually impossible for me as they will not supply anything other than a 30 day trial. At least Keil and IAR allow 'mini' projects without time limit.

The unit tests only run under Linux with GCC and Windows with GCC and Visual Studio.

DaveBian, 2015-11-12:
> IAR seems to be having a problem with std::iterator_traits::difference_type

John,

Yes, maybe you are right. After the code roll back for former version,
substituting 'size_t ' for type 'std::iterator_traits::difference_type' it
works.

Best regards!
David Bian
*********************************************************
Business Email: David.Bian@uster.com
<David.Bian@uster.com>Private Email: idavidbian@gmail.com
<idavidbian@gmail.com>
Skype: idavidbian
P Kindly please consider the environment before printing this email!
********************************************************

jwellbelove, 2015-11-14:
It looks like all of the 'is_permutation' overloads will be a problem for IAR. It would probably be wise to change the difference types to another standard signed type like int32_t. I don't see many embedded projects being required to check containers in excess of 2GB. It's possible that there may be some other parts of the library use 'difference_type' from 'interator_traits'. I shall have some free time in December to put some effort in to getting the embedded tests up to date and completing work on etl versions of unordered_map & set. Once these are done I plan to add some cache containers based on unordered_map.

closed 2015-12-09 by DaveBian

jwellbelove, 2015-12-11:
I have just issued a new release 6.0 that should significantly reduce code size when using many of the containers.