ETLCPP/etl #7: Supported platforms and compilers

fkromer, 2016-11-10:
The README.txt states *"Any help porting the library to work under different platforms and compilers would be gratefully received."* but I was not able to figure out what platforms and compilers are supported already and what portings you or the community prioritize as "most important".

jwellbelove, 2016-11-13:
The library is written and unit tested with GCC and Microsoft compilers on Windows and, occasionally on Linux, using GCC.

To my knowledge it has been used with ARM devices (bare metal, no OS) using TI's compiler for MSP430 microcontrollers and  C2000 series DSPs and also on Arduino devices.
I have experimented a little with compiling some of the classes using Kiel and IAR compilers, but that was not very extensive. Not all embedded compiler vendors allow free trial versions of their wares, and when they do, the code size is limited.

As far as I know, it has never been tried out with any of the real time OS's like VxWorks, FreeRTOS etc.
If anyone has, they've not told me about it.

What I need are people using different embedded compilers and targets to try out the library and suggest fixes for any incompatibilities. The library uses some compiler intrinsics which are often not portable between compilers. See 'alignment_of' in type_traits.h for an example.

AshleyRoll, 2016-12-26:
Hi John, first off, thanks heaps for this nice library, I think I will be using it a lot!

I've just started using ETL on a PIC32MZ using the Microchip XC32 compiler. I had a few teething issues getting it to compile and I'm only using a few basic containers at the moment (more later). But it seems to be working. I'll let you know how I get on. This is a fairly advanced stage project and I'm trying to simplify it a bit so was very happy to run across ETL, as I am firmly in the "no dynamic memory" camp for my systems. I'll certainly provide what feedback and possibly code when I can on this platform once I get a bit further into it.

My problem was the use of `std::nullptr_t` in several places. It was not defined in the version of the std library I had. I found your `NO_NULLPTR_SUPPORT` define and tried that, but ended up with:

`../../../ExternalRepos/etl/src/nullptr.h:75:19: error: expected unqualified-id before 'nullptr' const nullptr_t nullptr = {};`

where you attempt to create a replacement. So it seems that the compiler and shipped std lib are not in sync. I ended up resolving this with my own header that I add before your ETL library that does:

```c++
namespace std {
    typedef decltype(nullptr) nullptr_t;
}
```

As far as I know, XC32 (v1.42) is actually GCC 4.8.3 (well that's what --version reports). they also ship with a std c++ library by "Dinkum Compleat". 

This is not a problem with your ETL library, and I'll raise an issue with Microchip about it, but I don't expect a fast response to that, but I wanted to let anyone else having this problem know about it.

Ash.

jwellbelove, 2016-12-26:
Hi Ash,

I'm glad to hear that you found the library useful.
I started writng it several years ago when I found myself constantly 
reinventing an 'ETL type' wheel every time I started a new embedded job.
There were just no proper embedded friendly STL like libraries out 
there, especially ones which promised never to use dynamic memory.

I found an issue with nullptr using the ETL with GCC a short while ago 
(someone else had a similar problem).
It turned out to be with how the ETL detects nullptr support in 
/platform.h/. I updated it recently; you may still have an old version.

The GCC test line should read (defined(ETL_COMPILER_GCC) && (__cplusplus 
< 201103L))

Also I dumped all of the ETL_PLATFORM macro definitions when I realised 
that I was detecting the platform the code was /compiling /on and not 
the target's platform (Doh!)
The library will no longer have any references to the OS in it on the 
next release.

I've got a whole bunch of updates and fixes to upload very soon, 
including a couple of new intrusive containers and fixes for all of the 
GCC warnings.
I very nearly got them uploaded to Github in November but had a slight 
misshap with Git and reverted alot of my changes the day before I was 
about to fly to India for a holiday! The Christmas holiday has given me 
a chance to catch up again and I hope to push a new version before the 
new year.

There are two important issues that may or may not affect you at the 
moment; etl::vector and etl:variant do not destruct their elements when 
the container is destructed.
This will also be fixed in the next release.

Most of the projects people are using the ETL for ARM and TI's MSP430 
chips, so I'm interested to see how you get on with the PIC32MZ.
Having the compiler based on GCC helps a lot with getting cross compiler 
compatibilty problem free.

I'll be letting emailing everyone when the new release is available.

John.

AshleyRoll, 2016-12-26:
Hi John,

I'm on the latest _*master*_ branch (`57bde67`). I can see the GCC test you show above. 

I honestly think it is my compiler/std c++ library - it has `nullptr` but not `std::nullptr_t`, not your problem :)

Doesn't seem to be a lot of C++ work done on PIC32s, the MPLAB IDE is woeful at it, it was only in the last release of the compiler I could even call `__builtin_*()` methods to do interrupt enable/disable.

I'm looking forward to the update. sounds great. I was aware of the vector not destructing, not an issue for me as I'm using a vector of pointers (as a container for priority_queue) and the PIC32MZ is powered down most of the time when there is no work to do.

Really appreciate the work you've put in! I'll try to help out once this project ships :)

Ash.

xymopen, 2017-08-09:
I'm using Arduino IDE. For AVR devices, the compilation won't succeed unless comment out
    // Check to see if the compiler supports C++11 type traits.
    /* #if (defined(ETL_COMPILER_MICROSOFT) && (_MSC_VER >= 1600)) || \
        (defined(ETL_COMPILER_GCC) && (__cplusplus >= 201103L))
      #define ETL_C11_TYPE_TRAITS_SUPPORTED
    #endif */
in `platform.h` and add `typedef decltype(nullptr) nullptr_t;` to `std` namespace in `<cstddef>`.
The library do compile successfully with ESP8266 which is MIPS architect.
The library doesn't compile with ARM board, tested with Arduino Due and STM32F103 package from stm32duino.

jwellbelove, 2017-08-10:
What does the compiler complain about when the code in platform.h is not commented out?

What is the compilation error when using the Arduino Due?

What parts of the ETL library were you trying to use?

Thanks,
John

From: Xu Yiming
Sent: 09 August 2017 16:46
To: ETLCPP/etl
Cc: John Wellbelove; Comment
Subject: Re: [ETLCPP/etl] Supported platforms and compilers (#7)

I'm using Arduino IDE. For AVR devices, the compilation won't succeed unless comment out
// Check to see if the compiler supports C++11 type traits.
/* #if (defined(ETL_COMPILER_MICROSOFT) && (_MSC_VER >= 1600)) || 
(defined(ETL_COMPILER_GCC) && (__cplusplus >= 201103L))
#define ETL_C11_TYPE_TRAITS_SUPPORTED
#endif */
in platform.h and add typedef decltype(nullptr) nullptr_t; to std namespace in <cstddef>.
The library do compile successfully with ESP8266 which is MIPS architect.
The library doesn't compile with ARM board, tested with Arduino Due and STM32F103 package from stm32duino.

xymopen, 2017-08-16:
Sorry for the delay, I was quite busy last days.

The sketch is following:

	#include <vector.h>
	
	etl::vector< int, 12 > ints;
	
	void setup() {
	    Serial.begin( 9600 );
	
	    ints.push_back( 1 );
	}
	
	void loop() {
	   for ( auto i = ints.begin(); i != ints.end(); ++i ) {
	    Serial.println( *i );
	   }
	}

The compiler just complain about can't find `<type_traits>` if targeting AVR. The STL I am using is [andysworkshop/avr-stl: The C++ Standard Template Library for the Arduino](https://github.com/andysworkshop/avr-stl) and [mike-matera/ArduinoSTL: An STL and iostream implementation based on uClibc++ that supports my CS-11M class.](https://github.com/mike-matera/ArduinoSTL) However they all DO have `<type_traits.h>`. Raw error output is following:

	In file included from C:\Users\xymopen\Documents\Arduino\libraries\Embedded_Template_Library\src/iterator.h:36:0,
	                 from C:\Users\xymopen\Documents\Arduino\libraries\Embedded_Template_Library\src/algorithm.h:46,
	                 from C:\Users\xymopen\Documents\Arduino\libraries\Embedded_Template_Library\src/vector.h:48,
	                 from C:\Users\xymopen\Documents\Arduino\vector\vector.ino:2:
	
	C:\Users\xymopen\Documents\Arduino\libraries\Embedded_Template_Library\src/type_traits.h:62:25: fatal error: type_traits: No such file or directory
	   #include <type_traits>
	                         ^
	
	compilation terminated.
	
There is something wrong with my board manager so my SAM boards are missing. I choose Maple which is STM32F103 instead. The compiler throw tons of error when targeting ARM. The STL using is bounded to the compiler. DUE and STM32 are all using `arm-none-eabi-gcc` but different revision. The output is following:

	"C:\Users\xymopen\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -Wall -Wextra -DDEBUG_LEVEL=DEBUG_ALL -std=gnu++11 -MMD -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8002000 -DERROR_LED_PORT=GPIOB -DERROR_LED_PIN=1 -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10803 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1  -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8  -mthumb  -march=armv7-m -D__STM32F1__ "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple" "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include" "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/stm32f1/include"                                                           "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/usb/stm32f1" "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/usb/usb_lib" "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\cores\maple" "-IC:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\variants\generic_stm32f103c" "-IC:\Users\xymopen\Documents\Arduino\libraries\Embedded_Template_Library\src" "C:\Users\xymopen\AppData\Local\Temp\arduino_build_494664\sketch\vector.ino.cpp" -o "C:\Users\xymopen\AppData\Local\Temp\arduino_build_494664\sketch\vector.ino.cpp.o"
	In file included from C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\cores\maple/boards.h:37:0,
	                 from C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\cores\maple/wirish.h:54,
	                 from C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\cores\maple/Arduino.h:30,
	                 from C:\Users\xymopen\AppData\Local\Temp\arduino_build_494664\sketch\vector.ino.cpp:1:
	C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
	 #define __io volatile
	              ^
	C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'const std::ios_base&'
	 #define __io volatile
	              ^
	C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
	 #define __io volatile
	              ^
	C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
	 #define __io volatile
	              ^
	C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
	 #define __io volatile
	              ^
	C:\Users\xymopen\AppData\Local\Arduino15\packages\stm32duino\hardware\STM32F1\2017.8.7\system/libmaple/include/libmaple/libmaple_types.h:55:14: error: 'volatile' qualifiers cannot be applied to 'std::ios_base&'
[... cut ...]
	c:\users\xymopen\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\typeinfo: At global scope:
	c:\users\xymopen\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\typeinfo:39:37: error: expected '}' before end of line
	c:\users\xymopen\appdata\local\arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1\arm-none-eabi\include\c++\4.8.3\typeinfo:39:37: error: expected declaration before end of line

xymopen, 2017-08-16:
Although it seems STL should be blamed when targeting ARM, the compiler would very happy when using `#include <vector>` and `std::vector< int > ints;` instead of `#include <vector.h>` and `etl::vector< int, 12 > ints;`, so it might require lots of effort to find the bugs in the code. ARM has quite recent STL support.

closed 2018-05-30 by jwellbelove