ETLCPP/etl #9: Too many warnings with -Wall in GCC

bkaradzic, 2016-11-20:
I just tried to use library with my project but it produces way too many warnings. Luckily there are only for types of warnings, but many instances of them. If these files were .cpp files I would just suppress these warnings for compilation, but because they are in .h files it should be fixed so suppression doesn't affect user code.

-Wundef
```
etl/intrusive_forward_list.h:935:5: warning: "_DEBUG" is not defined [-Wundef]
```

`#if _DEBUG` should be replaced with `#if defined(_DEBUG) && _DEBUG`

-Wshadow

```
etl/function.h:204:7: warning: declaration of ‘p_function’ shadows a member of ‘etl::function<void, void>’ [-Wshadow]
       : p_function(p_function)
       ^
../../../../bx/include/etl/function.h:219:24: note: shadowed declaration is here
     void (*p_function)(); ///< Pointer to the function.
```

This warning is available also in VS2015 by default (not sure why you're not seeing it).

-Wstrict-aliasing

```
etl/hash.h:369:45: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         return *reinterpret_cast<size_t*>(&v);
```

This can be fixed by casting through union.

-Wunused-parameter

```
etl/exception.h:65:47: warning: unused parameter ‘file’ [-Wunused-parameter]
     exception(string_type reason, string_type file, numeric_type line)
```
This is due to #ifdef, parameters should be just referenced.

bkaradzic, 2016-11-20:
Another one, https://github.com/ETLCPP/etl/blob/master/src/private/ivectorpointer.h#L38 include should be just `#include "pvoidvector.h"` since files is the same directory.

jwellbelove, 2016-12-08:
Thanks Branimir,
I'll look into those points you raised as soon as possible.
I've just got back from a holiday in India and I'm still a bit jet lagged!
I've also got to fix a mess of my own making with the ETL. I was about to release a new version with several fixes and enhancements before I went away, but made a small error with Git and was punished by loosing some of my work (I reverted ALL the files instead of just one). I'm going to have to check my backups and try to rebuild what I lost.
To be honest, in all my decades of C/C++ coding this is the first time that I've come across the 'strict aliasing' warning.

bkaradzic, 2016-12-08:
> I'll look into those points you raised as soon as possible.

Cool I'm looking forward to start using ETL in my projects! :)

> To be honest, in all my decades of C/C++ coding this is the first time that I've come across the 'strict aliasing' warning.

If you encounter it as non-warning, and `-fno-strict-aliasing` option fixes the issue, it's usually hell to debug it. :)

There is good overview here:
http://cellperformance.beyond3d.com/articles/2006/06/understanding-strict-aliasing.html

jwellbelove, 2016-12-08:
There seems to be some debate online as to whether the 'union' method is portable or even defined behaviour. Using 'memcpy' seems to be one of the only guaranteed methods.

bkaradzic, 2016-12-08:
I used this cast through union for years and no issues, theoretically it's undefined, but not in practice:

`Strictly speaking, reading a member of a union different from the one written to is undefined in ANSI/ISO C99 except in the special case of type-punning to a char*, similar to the example below: Casting to char*. However, it is an extremely common idiom and is well-supported by all major compilers. As a practical matter, reading and writing to any member of a union, in any order, is acceptable practice.`

jwellbelove, 2016-12-12:
I decided to go with the 'union' method.

AshleyRoll, 2017-10-30:
Hi John,

Any thoughts on the -Wshadow fixes? I'm running into this and have to manually disable the warnings around my #includes and on any cpp files from etl. 

Not a show stopper at all, but would be nice to be able to compile cleanly in a paranoid build :)

Thanks,
Ash.

jwellbelove, 2017-11-01:
Hi Ash,

I started looking at this yesterday and I should have a new release this morning (UK time)

John

jwellbelove, 2017-11-01:
Done.

annius, 2018-02-08:
Is this issue considered closed? I'm still seeing the strict aliasing warning when compiling code with gcc 5.4,  when I'm using `etl::variant`.

jwellbelove, 2018-02-08:
OK, I'll have another look at that.

jwellbelove, 2018-02-08:
I've added unit tests for testing the conversion operators but I'm not seeing the warnings.
I have these warnings set.

-Wdelete-non-virtual-dtor
-Wunused-parameter
-Wstrict-aliasing
-Wshadow
-Wundef 
-Wall

jwellbelove, 2018-02-08:
Linux: GCC 5.4.0
Windows: GCC 4.9.2

bkaradzic, 2018-02-08:
Btw, you should update your GCC, those you're using are really old (it's ok to use them to support legacy, but newer ones introduce more static analysis checks):
https://askubuntu.com/questions/26498/choose-gcc-and-g-version/26518#26518

annius, 2018-02-08:
@jwellbelove After your feedback I wrote a simple program with a variant use case and it built without warnings. The issue I had is probably root caused somewhere else in the project and I'll investigate further. If I can come up with well-encapsulated example that can serve as a (failing) test case I'll pass it on to.

Thanks so much for the prompt response.

annius, 2018-02-19:
I tried the latest commit (https://github.com/ETLCPP/etl/commit/442757698500b63bb942423cb7ac5ffae448c554) and it appears to have fixed the strict aliasing issue.

Thanks!

annius, 2018-02-19:
BTW building with `-Wunused-parameter` on MacOS fails. I'll provide a short example shortly.

annius, 2018-02-23:
OK, here's a test case. I created a file called `test.cpp` with the following code
```
#include <iostream>
#include "optional.h"

int main() {
  etl::optional<std::size_t> a = 10;
}
```
Building with
```
$/usr/bin/clang++-3.8 test.cpp -std=c++11 
```
works successfully.

Building with
```
$/usr/bin/clang++-3.8 test.cpp -std=c++11 -Wunused-parameter
```
results in
```
In file included from test.cpp:2:
./optional.h:384:58: warning: unused parameter 'rhs' [-Wunused-parameter]
bool operator ==(etl::nullopt_t, const etl::optional<T>& rhs)
                                                         ^
1 warning generated.
```

annius, 2018-03-26:
I believe this issue can be closed out as we are no longer seeing any warnings under `-Wall`, `-Wextra`, `-Wpedantic`, and `-Wfloat-equal` under `gcc` version 5.4 and `clang` version 3.8.