ETLCPP/etl #10: Container destructor does not call item destructors

BlackMATov, 2016-11-21:
It's very seriously bug in every etl container.

```cpp
#include <iostream>
#include <vector.h>

class item_t {
    item_t(const item_t&);
    item_t& operator=(const item_t&);
public:
    item_t() {
        std::cout << "item_t::item_t()" << std::endl;
    }
    ~item_t() {
        std::cout << "item_t::~item_t()" << std::endl;
    }
};

int main() {
    { // vector scope
        etl::vector<item_t, 10> v;
        v.push_back();
        v.push_back();
    }
    return 0;
}

// terminal output:
// item_t::item_t()
// item_t::item_t()
// Program ended with exit code: 0
```

jwellbelove, 2016-12-12:
Good grief, how did I miss that!
It may have been that in the original implementations, the internal data was based around arrays, which would automatically destruct, rather than the memory storage areas and pools that I now use.
Thanks.

jwellbelove, 2016-12-12:
It looks like only vector suffers from this. Most other containers use etl::pool which has its own destructor which destroys all of the elements.

BlackMATov, 2017-02-11:
Ouh! It's OK now :)
👍

closed 2017-02-11 by BlackMATov