ETLCPP/etl #11: Problems migrating from STL to ETL

nlohmann, 2016-11-24:
I really appreciate your realisation of an STL version for embedded applications. I once worked on a project where we needed to rewrite a lot of the code, because a certain STL function was not available on the target platform...

I am the maintainer of JSON for Modern C++ (https://github.com/nlohmann/json), a header-only library to make JSON a first-class datatype in C++11. I am using a lot of the STL, but at its core, the library is just a wrapper around std::map, std::vector, and std::string. As these containers are used as template types, it should be possible to replace them with the ETL versions. (see nlohmann/json#361)

What do you think?

I tried to compile a small program myself, but the moment I, for instance, include "vector.h", I get a lot of compiler errors even though I do not use the vector. Is there any migration guide available or a list of gotchas?

Program:

```cpp
#include <vector.h>

int main()
{
	return 0;
}
```

Output:

```
g++-5 -Ietl    testetl.cpp   -o testetl
In file included from etl/alignment.h:37:0,
                 from etl/vector.h:40,
                 from testetl.cpp:1:
etl/array.h: In function 'T& etl::get(etl::array<T, MAXN>&)':
etl/static_assert.h:37:77: error: there are no arguments to 'static_assert' that depend on a template parameter, so a declaration of 'static_assert' must be available [-fpermissive]
   #define STATIC_ASSERT(Condition, Message) static_assert(Condition, Message)
                                                                             ^
etl/array.h:457:5: note: in expansion of macro 'STATIC_ASSERT'
     STATIC_ASSERT(I < MAXN, "Index out of bounds");
     ^
etl/static_assert.h:37:77: note: (if you use '-fpermissive', G++ will accept your code, but allowing the use of an undeclared name is deprecated)
   #define STATIC_ASSERT(Condition, Message) static_assert(Condition, Message)
                                                                             ^
etl/array.h:457:5: note: in expansion of macro 'STATIC_ASSERT'
     STATIC_ASSERT(I < MAXN, "Index out of bounds");
     ^
etl/array.h: In function 'const T& etl::get(const etl::array<T, MAXN>&)':
etl/static_assert.h:37:77: error: there are no arguments to 'static_assert' that depend on a template parameter, so a declaration of 'static_assert' must be available [-fpermissive]
   #define STATIC_ASSERT(Condition, Message) static_assert(Condition, Message)
                                                                             ^
etl/array.h:472:5: note: in expansion of macro 'STATIC_ASSERT'
     STATIC_ASSERT(I < MAXN, "Index out of bounds");
     ^
make: *** [testetl] Error 1
```

nlohmann, 2016-12-01:
Hi there! Any reaction on this?

jwellbelove, 2016-12-09:
Sorry to be so late in replying, but I've been on holiday in India for the last two weeks and I had very limited internet access!

It looks like the GCC version you are using doesn't support static_assert. At the moment the header assumes that GCC supports this. I will have to add some macro definitions to check the GCC version, but in the mean time you could comment out the following lines in static_assert.h

#elif defined(ETL_COMPILER_GCC)
  #define STATIC_ASSERT(Condition, Message) static_assert(Condition, Message)

I have a few other fixes and improvements to upload, but a glitch with Git just before I went away meant I will have to rebuild a few of them. :-(

nlohmann, 2016-12-09:
I am using g++-5 (Homebrew gcc5 5.4.0) 5.4.0 - this version supports `static_assert`.

jwellbelove, 2016-12-12:
OK, I think I get what the problem is, but I'm not sure of the solution as yet.

jwellbelove, 2016-12-12:
Does your configuration have -std=c++11 set?

nlohmann, 2016-12-12:
Yes.

jwellbelove, 2016-12-12:
The problem seems to be that GCC doesn't like this code...

  template <std::size_t I, typename T, std::size_t MAXN>
  inline T& get(array<T, MAXN>& a)
  {
    static_assert(I < MAXN, "Index out of bounds"); <<<<<<<<<<<<<<<<<<<
    return a[I];
  }

error: there are no arguments to 'static_assert' that depend on a template parameter, so a declaration of 'static_assert' must be available [-fpermissive]

The call to static assert already depends on template non-type parameters I & MAXN.
I'm assuming that it must depend on either 'a' or maybe the type T too?
I need to experiment. 
You could either comment out the STATIC_ASSERT macro as previously mentioned of temporarily comment out the STATIC_ASSERTs in etl::array until I have a solution.

jwellbelove, 2016-12-12:
I can only get this to fail if I select "-std=c++98"
If I select "-std=c++11" then I get no warnings.

nlohmann, 2016-12-12:
I tried again with the most recent release. The error does not occur any more. I am puzzled. Sorry for troubling you.

I now try to get the JSON library to compile with ETL.

closed 2016-12-12 by nlohmann