ETLCPP/etl #12: error: '::memchr' has not been declared

nlohmann, 2016-12-12:
Using the release version 8.2.0, the following file cannot be compiled with `g++-5 -Ietl-8.2.0/src    testetl.cpp -o testetl -std=c++11` using g++-5 (Homebrew gcc5 5.4.0) 5.4.0 on macOS 10.12.1.

```cpp
#include <vector.h>
#include <string.h>

int main()
{
    return 0;
}
```

Errors:

```
In file included from etl-8.2.0/src/ibasic_string.h:39:0,
                 from etl-8.2.0/src/basic_string.h:38,
                 from etl-8.2.0/src/string.h:35,
                 from testetl.cpp:2:
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:75:11: error: '::memchr' has not been declared
   using ::memchr;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:76:11: error: '::memcmp' has not been declared
   using ::memcmp;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:77:11: error: '::memcpy' has not been declared
   using ::memcpy;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:78:11: error: '::memmove' has not been declared
   using ::memmove;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:79:11: error: '::memset' has not been declared
   using ::memset;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:80:11: error: '::strcat' has not been declared
   using ::strcat;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:81:11: error: '::strcmp' has not been declared
   using ::strcmp;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:82:11: error: '::strcoll' has not been declared
   using ::strcoll;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:83:11: error: '::strcpy' has not been declared
   using ::strcpy;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:84:11: error: '::strcspn' has not been declared
   using ::strcspn;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:85:11: error: '::strerror' has not been declared
   using ::strerror;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:86:11: error: '::strlen' has not been declared
   using ::strlen;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:87:11: error: '::strncat' has not been declared
   using ::strncat;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:88:11: error: '::strncmp' has not been declared
   using ::strncmp;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:89:11: error: '::strncpy' has not been declared
   using ::strncpy;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:90:11: error: '::strspn' has not been declared
   using ::strspn;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:91:11: error: '::strtok' has not been declared
   using ::strtok;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:92:11: error: '::strxfrm' has not been declared
   using ::strxfrm;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:93:11: error: '::strchr' has not been declared
   using ::strchr;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:94:11: error: '::strpbrk' has not been declared
   using ::strpbrk;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:95:11: error: '::strrchr' has not been declared
   using ::strrchr;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:96:11: error: '::strstr' has not been declared
   using ::strstr;
           ^
```

The same errors occur with GCC 4.9.3 and GCC 6.2.0 with the same compiler flags.

jwellbelove, 2016-12-12:
It may be that the compiler is picking up ETL's string.h instead of the C library's version.
You may have to play around with the include directory paths.
I think it can happen if you set the include path directly to etl/src/
If you set the path to the one containing the ETL library then you can #include "etl/src/..."
Hope that helps.

nlohmann, 2016-12-12:
Sorry, I don't get it to work.

I tried

```cpp
#include "etl-8.2.0/src/vector.h"
#include "etl-8.2.0/src/string.h"

int main()
{
    return 0;
}
```

and ran ` g++-5 testetl.cpp -o testetl -std=c++11` (the file `testetl.cpp` is in a directory that contains the release directory `etl-8.2.0`). The error message is

```
In file included from etl-8.2.0/src/ivector.h:904:0,
                 from etl-8.2.0/src/vector.h:38,
                 from testetl.cpp:1:
etl-8.2.0/src/private/ivectorpointer.h:38:33: fatal error: private/pvoidvector.h: No such file or directory
compilation terminated.
```

When I add `-I etl-8.2.0/src` to resolve `private/pvoidvector.h`, I get

```
In file included from etl-8.2.0/src/ibasic_string.h:39:0,
                 from etl-8.2.0/src/basic_string.h:38,
                 from etl-8.2.0/src/string.h:35,
                 from testetl.cpp:2:
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:75:11: error: '::memchr' has not been declared
   using ::memchr;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:76:11: error: '::memcmp' has not been declared
   using ::memcmp;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:77:11: error: '::memcpy' has not been declared
   using ::memcpy;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:78:11: error: '::memmove' has not been declared
   using ::memmove;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:79:11: error: '::memset' has not been declared
   using ::memset;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:80:11: error: '::strcat' has not been declared
   using ::strcat;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:81:11: error: '::strcmp' has not been declared
   using ::strcmp;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:82:11: error: '::strcoll' has not been declared
   using ::strcoll;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:83:11: error: '::strcpy' has not been declared
   using ::strcpy;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:84:11: error: '::strcspn' has not been declared
   using ::strcspn;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:85:11: error: '::strerro' has not been declared
   using ::strerror;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:86:11: error: '::strlen' has not been declared
   using ::strlen;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:87:11: error: '::strncat' has not been declared
   using ::strncat;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:88:11: error: '::strncmp' has not been declared
   using ::strncmp;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:89:11: error: '::strncpy' has not been declared
   using ::strncpy;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:90:11: error: '::strspn' has not been declared
   using ::strspn;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:91:11: error: '::strtok' has not been declared
   using ::strtok;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:92:11: error: '::strxfrm' has not been declared
   using ::strxfrm;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:93:11: error: '::strchr' has not been declared
   using ::strchr;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:94:11: error: '::strpbrk' has not been declared
   using ::strpbrk;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:95:11: error: '::strrchr' has not been declared
   using ::strrchr;
           ^
/usr/local/Cellar/gcc5/5.4.0/include/c++/5.4.0/cstring:96:11: error: '::strstr' has not been declared
   using ::strstr;
           ^
```

jwellbelove, 2016-12-13:
Hi Niel,

Sorry to hear you're still having issues.

private/pvoidvector.h is an error.
It should be just pvoidvector.h

I've updated the file in the master branch

As for the other errors, could you try changing the <cstring> in the string classes to <string.h> and see if it makes a difference?

John.

jwellbelove, 2017-01-01:
I've changed the name of string.h to cstring.h which seems to solve the issues you are seeing. The code is on theme master branch.

IAMFINN, 2020-02-06:
same question,Do you fixed?

jwellbelove, 2020-02-06:
It is fixed as far as I  know

closed 2020-02-06 by jwellbelove