ETLCPP/etl #13: Performance benchmarks ETL vs other STL implementations

bkaradzic, 2016-12-29:
I just did quick benchmark comparing different implementation of unordered_map, and ETL was the slowest. Do you have benchmarks ETL vs STL, vs EASTL, or something similar?

Code:

```cpp
	///
	{
		int64_t elapsed = -bx::getHPCounter();

		for (uint32_t ii = 0; ii < numIterations; ++ii)
		{
			typedef etl::unordered_map<uint64_t, uint16_t, numElements> EtlUnorderedMap;
			EtlUnorderedMap map;

			for (uint32_t jj = 0; jj < numElements; ++jj)
			{
				std::pair<EtlUnorderedMap::iterator, bool> ok = map.insert(std::make_pair(uint64_t(jj), uint16_t(jj) ) );
				assert(ok.second);
			}

			for (uint32_t jj = 0; jj < numElements; ++jj)
			{
				bool ok = bx::mapRemove(map, uint64_t(jj) );
				assert(ok);
			}

//			assert(map.size() == 0);
		}

		elapsed += bx::getHPCounter();
		printf("          ETL: %15f\n", double(elapsed) );
	}
```

Was added here:
https://github.com/bkaradzic/bx/blob/master/tests/handle_bench.cpp

The results were:

```
      TinyStl:     4534.000000
          STL:     3573.000000
          ETL:     9690.000000
HandleHashMap:     1569.000000
```

I also tried increasing maximum number of elements to 2x in `etl::unordered_map` but that didn't change results.

Also there might be some issue with `erase` functionality but didn't look into it.

jwellbelove, 2016-12-29:
Hi Branimir,

Thanks, I'll look into where the bottleneck is.
One possibility may be due to the fact that I'm using a forward 
intrusive list for storage.
This requires less memory space than a bidirectional list but is slower 
due to the higher management overhead.
Up to now I've been concentrating on getting the functionality correct, 
rather than tuning for speed.
If you can put together a test to show where you think erase is going 
wrong, that would be very helpful.

Regards,
John.

jwellbelove, 2016-12-29:
OK, I see the issue wth erase.

On 29/12/2016 06:22, Branimir Karadžić wrote:
> creasing maximum number of elements to 2x in |etl::unordered_map| but 
> that didn't change results.
>
> Also there might be some issue with |erase| functionality but didn't 
> look into it.
>
>

jwellbelove, 2016-12-30:
I've traced the performance bottleneck to where etl::pool searches the 
internal bitset for the next available location to contruct a new element.

John

bkaradzic, 2016-12-30:
Cool! I mentioned it because TinySTL is also designed as minimal compatible subset of STL (with better allocator model). It's primary goal was to replace allocator, but not to be faster than STL. That's why I reported it, because I expected ETL to be faster than TinySTL by default, just because ETL doesn't do any dynamic allocation.

jwellbelove, 2016-12-31:
Hi Branimir,

I have completed some initial experiments with a new pool implementation 
based on an algorithm I found online.
https://www.thinkmind.org/index.php?view=article&articleid=computation_tools_2012_1_10_80006 

The new algorithm is not as space efficient as the previous 
implementation as it uses an index for each element instead of one bit 
in a bitset, but it is *extremely* fast!

Processor: Intel i7 16GB 3.40 GHz
Operating System: Windows 10 64bit
Compiler: Visual Studio 2015
Optimisation: Maximize Speed (/O2)
Container: unordered_map

Size: 10,000
Iterations: 16
STL: 26ms
ETL: 4ms
650%

Size: 100,000
Iterations: 16
STL: 437ms
ETL: 41ms
1066%

Size: 1,000,000
Iterations: 16
STL: 6386ms
ETL: 420ms
1520%

Size: 10,000,000
Iterations: 16
STL: 88020ms
ETL: 4096ms
2149%

I'm away for the next couple of days, but when I get back I shall try to 
improve the code to allow space optimisations based on the maximum size 
of the pool.

i.e.
If the maximum pool is not greater than 127 then the index may be an 
int8_t, if not greater than 32767, then int16_t etc.
(The index must always be a signed type to allow the pool to distinguish 
between allocated and non-allocated nodes).

John.

#include "stdafx.h"
#include <windows.h>
#include <iostream>

#include <unordered_map>
#include "../../../../src/unordered_map.h"

LARGE_INTEGER frequency;
LARGE_INTEGER begin;

void StartTimer()
{
   QueryPerformanceFrequency(&frequency);
   QueryPerformanceCounter(&begin);

   frequency.QuadPart /= 1000;
}

uint64_t StopTimer()
{
   LARGE_INTEGER end;
   QueryPerformanceCounter(&end);

   return (end.QuadPart - begin.QuadPart) / frequency.QuadPart;
}

const size_t TESTSIZE = 1000000;
const size_t TESTINTERATIONS = 16;

typedef std::unordered_map<uint64_t, uint16_t> Stdmap;
typedef etl::unordered_map<uint64_t, uint16_t, TESTSIZE> Etlmap;

Stdmap stdmap;
Etlmap etlmap;

int main()
{
   StartTimer();

   for (size_t i = 0; i < TESTINTERATIONS; ++i)
   {
     for (size_t j = 0; j < TESTSIZE; ++j)
     {
       std::pair<Stdmap::iterator, bool> ok = stdmap.insert(std::make_pair(uint64_t(j), uint16_t(j)));
     }

     for (size_t j = 0; j < TESTSIZE; ++j)
     {
       stdmap.erase(j);
     }
   }

   uint64_t time;

   time = StopTimer();
   std::cout << "STD Time = " << time << "ms\n";

   StartTimer();

   for (size_t i = 0; i < TESTINTERATIONS; ++i)
   {
     for (size_t j = 0; j < TESTSIZE; ++j)
     {
       std::pair<Etlmap::iterator, bool> ok = etlmap.insert(std::make_pair(uint64_t(j), uint16_t(j)));
     }

     for (size_t j = 0; j < TESTSIZE; ++j)
     {
       etlmap.erase(j);
     }
   }

   time = StopTimer();
   std::cout << "ETL Time = " << time << "ms\n";

   return 0;
}

jwellbelove, 2017-02-01:
I have pushed a new release to the master branch with a completely re-written pool class.
It's taken a bit of time to get done as most of the containers use pool and the API for allocate and release had changed.
See http://www.etlcpp.com/benchmarks.html for some initial benchmarks.

closed 2020-02-25 by jwellbelove