ETLCPP/etl #14: vector of pointer to a class fails to compile in Visual Studio 2015

AshleyRoll, 2016-12-29:
Hi,

I was trying to write a vector of pointers to a class but it fails to compile under VS2015. I'm using this to build my unit  tests before putting the code in to XC32 for the PIC.

I have updated my repository to the latest code (`5c42b7a`), but this also exhibits the error.

```c++
class AClass 
{
};

AClass a;
AClass b;

etl::vector<AClass*, 5> v = {&a, &b};
```

results in 

```
1>c:\ash\git\r4\externalrepos\etl\src\private\ivectorpointer.h(315): error C2440: 'type cast': cannot convert from 'ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass' to 'void *'
1>  c:\ash\git\r4\externalrepos\etl\src\private\ivectorpointer.h(315): note: No user-defined-conversion operator available that can perform this conversion, or the operator cannot be called
1>  c:\ash\git\r4\externalrepos\etl\src\vector.h(200): note: see reference to function template instantiation 'void etl::ivector<T *>::assign<TIterator>(TIterator,TIterator)' being compiled
1>          with
1>          [
1>              T=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass,
1>              TIterator=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass *
1>          ]
1>  c:\ash\git\r4\externalrepos\etl\src\vector.h(200): note: see reference to function template instantiation 'void etl::ivector<T *>::assign<TIterator>(TIterator,TIterator)' being compiled
1>          with
1>          [
1>              T=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass,
1>              TIterator=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass *
1>          ]
1>  c:\ash\git\r4\firmware\pic32\lib\_unittests\configtests\configloadtests.cpp(111): note: see reference to function template instantiation 'etl::vector<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass *,5>::vector<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass*>(TIterator,TIterator)' being compiled
1>          with
1>          [
1>              TIterator=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass *
1>          ]
1>  c:\ash\git\r4\firmware\pic32\lib\_unittests\configtests\configloadtests.cpp(111): note: see reference to function template instantiation 'etl::vector<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass *,5>::vector<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass*>(TIterator,TIterator)' being compiled
1>          with
1>          [
1>              TIterator=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass *
1>          ]
```

I get a similar error with XC32 (PIC32 compiler based on GCC). 

```
Config/../../../../ExternalRepos/etl/src/private/ivectorpointer.h: In instantiation of 'void etl::ivector<T*>::assign(TIterator, TIterator) [with TIterator = Config::foo()::AClass*; T = Config::foo()::AClass]':
In file included from Config/../../../../ExternalRepos/etl/src/ivector.h:904:0,
                 from Config/../../../../ExternalRepos/etl/src/vector.h:38,
                 from Config/ConfigurationManager.cpp:6:
Config/../../../../ExternalRepos/etl/src/vector.h:200:38:   required from 'etl::vector<T*, MAX_SIZE_>::vector(TIterator, TIterator) [with TIterator = Config::foo()::AClass*; T = Config::foo()::AClass; long unsigned int MAX_SIZE_ = 5ul]'
Config/ConfigurationManager.cpp:52:38:   required from here
Config/../../../../ExternalRepos/etl/src/private/ivectorpointer.h:315:34: error: invalid cast from type 'Config::foo()::AClass' to type 'void*'
         p_buffer[current_size++] = (void*)*first++;
```

jwellbelove, 2016-12-29:
Hi Ashley,

You're trying to use C++11 syntax to initialise the etl::vector.
The ETL was designed to support C++03 compliant compilers, which is as 
high as many embedded compilers go. (Keil, IAR etc)

John.

AshleyRoll, 2016-12-29:
Hi John,

Ah, OK. for once I'm running on embedded compiler that doesn't completely suck then :)

```c++
class AClass
{
};
AClass a;
AClass b;

AClass *t[] = { &a, &b };

etl::vector<AClass*, 5> v;
v.assign(std::begin(t), std::end(t));
```

works. Thanks for that.

I was hoping to avoid the intermediate array, or a set of `v.push_back()` calls to make my static initialisation easier..

closed 2016-12-29 by AshleyRoll