ETLCPP/etl #15: vector of reference to class fails to compile

AshleyRoll, 2016-12-29:
Hi,

When trying to create and initialise a vector to a reference of a class, the compilation fails.

```c++
class AClass 
{
};

AClass a;
AClass b;

etl::vector<AClass&, 5> v = {a, b};		
```

when compiled in VS2015, I get the following error:
```
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(66): error C2528: 'pointer': pointer to reference is illegal
1>  c:\ash\git\r4\externalrepos\etl\src\vector.h(59): note: see reference to class template instantiation 'etl::ivector<T>' being compiled
1>          with
1>          [
1>              T=ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass &
1>          ]
1>  c:\ash\git\r4\firmware\pic32\lib\_unittests\configtests\configloadtests.cpp(112): note: see reference to class template instantiation 'etl::vector<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass &,5>' being compiled
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(67): error C2528: 'const_pointer': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(68): error C2528: 'iterator': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(69): error C2528: 'const_iterator': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(70): error C2528: '<template-parameter>': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(71): error C2528: '<template-parameter>': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C2528: '<template-parameter>': pointer to reference is illegal
1>c:\program files (x86)\microsoft visual studio 14.0\vc\include\xutility(619): error C2528: '<template-parameter>': pointer to reference is illegal
1>  c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): note: see reference to class template instantiation 'std::iterator_traits<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass &(*)>' being compiled
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C2039: 'difference_type': is not a member of 'std::iterator_traits<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass &(*)>'
1>  c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): note: see declaration of 'std::iterator_traits<ConfigTests::ConfigLoadTests::TestETLVectorOfPointers::AClass &(*)>'
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C3646: 'difference_type': unknown override specifier
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(759): error C2528: 'p_buffer': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(785): error C2528: 'p_buffer': pointer to reference is illegal
```

in XC32 (PIC32 compiler based in GCC) I get:

```
Config/../../../../ExternalRepos/etl/src/ivector.h: In instantiation of 'class etl::ivector<Config::foo()::AClass&>':
In file included from Config/../../../../ExternalRepos/etl/src/vector.h:38:0,
                 from Config/ConfigurationManager.cpp:6:
Config/../../../../ExternalRepos/etl/src/vector.h:58:9:   required from 'class etl::vector<Config::foo()::AClass&, 5ul>'
Config/ConfigurationManager.cpp:52:27:   required from here
Config/../../../../ExternalRepos/etl/src/ivector.h:66:16: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef T*                                    pointer;
                ^
Config/../../../../ExternalRepos/etl/src/ivector.h:67:22: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef const T*                              const_pointer;
                      ^
Config/../../../../ExternalRepos/etl/src/ivector.h:68:16: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef T*                                    iterator;
                ^
Config/../../../../ExternalRepos/etl/src/ivector.h:69:22: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef const T*                              const_iterator;
                      ^
Config/../../../../ExternalRepos/etl/src/ivector.h:70:45: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef std::reverse_iterator<iterator>       reverse_iterator;
                                             ^
Config/../../../../ExternalRepos/etl/src/ivector.h:71:51: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef std::reverse_iterator<const_iterator> const_reverse_iterator;
                                                   ^
Config/../../../../ExternalRepos/etl/src/ivector.h:73:70: error: forming pointer to reference type 'Config::foo()::AClass&'
     typedef typename std::iterator_traits<iterator>::difference_type difference_type;
                                                                      ^
In file included from Config/../../../../ExternalRepos/etl/src/vector.h:38:0,
                 from Config/ConfigurationManager.cpp:6:
Config/../../../../ExternalRepos/etl/src/ivector.h:759:5: error: forming pointer to reference type 'Config::foo()::AClass&'
     ivector(T* p_buffer, size_t MAX_SIZE)
     ^
Config/../../../../ExternalRepos/etl/src/ivector.h:785:8: error: forming pointer to reference type 'Config::foo()::AClass&'
     T* p_buffer;
        ^
Config/../../../../ExternalRepos/etl/src/vector.h: In instantiation of 'etl::vector<T, MAX_SIZE_>::vector(TIterator, TIterator) [with TIterator = Config::foo()::AClass; T = Config::foo()::AClass&; long unsigned int MAX_SIZE_ = 5ul]':
In file included from Config/ConfigurationManager.cpp:6:0:
Config/ConfigurationManager.cpp:52:36:   required from here
Config/../../../../ExternalRepos/etl/src/vector.h:104:59: error: forming pointer to reference type 'Config::foo()::AClass&'
       : ivector<T>(reinterpret_cast<T*>(&buffer), MAX_SIZE)
                                                           ^
Config/../../../../ExternalRepos/etl/src/ivector.h: In instantiation of 'void etl::ivector<T>::destroy_element() [with T = Config::foo()::AClass&]':
In file included from Config/../../../../ExternalRepos/etl/src/vector.h:38:0,
                 from Config/ConfigurationManager.cpp:6:
Config/../../../../ExternalRepos/etl/src/ivector.h:780:27:   required from 'void etl::ivector<T>::initialise() [with T = Config::foo()::AClass&]'
Config/../../../../ExternalRepos/etl/src/ivector.h:448:18:   required from 'void etl::ivector<T>::clear() [with T = Config::foo()::AClass&]'
Config/ConfigurationManager.cpp:54:11:   required from here
Config/../../../../ExternalRepos/etl/src/ivector.h:818:15: error: using invalid field 'etl::ivector<T>::p_buffer'
       p_buffer[--current_size].~T();
               ^
In file included from Config/../../../../ExternalRepos/etl/src/vector.h:38:0,
                 from Config/ConfigurationManager.cpp:6:
Config/../../../../ExternalRepos/etl/src/ivector.h:383:10: error: 'void etl::ivector<T>::assign(TIterator, TIterator) [with TIterator = Config::foo()::AClass; T = Config::foo()::AClass&]', declared using local type 'Config::foo()::AClass', is used but never defined [-fpermissive]
     void assign(TIterator first, TIterator last)
          ^
```

AshleyRoll, 2016-12-29:
As per the pointer to class issue with C++11, I rewrote this test and I'm still getting the errors.

```c++
class AClass
{
public: int val;
};
			
AClass t[] = { 1, 2 };
etl::vector<AClass&, 5> v;
v.assign(std::begin(t), std::end(t));
```

results in

```
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(66): error C2528: 'pointer': pointer to reference is illegal
1>  c:\ash\git\r4\externalrepos\etl\src\vector.h(59): note: see reference to class template instantiation 'etl::ivector<T>' being compiled
1>          with
1>          [
1>              T=ConfigTests::ConfigLoadTests::bar::AClass &
1>          ]
1>  c:\ash\git\r4\firmware\pic32\lib\_unittests\configtests\configloadtests.cpp(130): note: see reference to class template instantiation 'etl::vector<ConfigTests::ConfigLoadTests::bar::AClass &,5>' being compiled
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(67): error C2528: 'const_pointer': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(68): error C2528: 'iterator': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(69): error C2528: 'const_iterator': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(70): error C2528: '<template-parameter>': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(71): error C2528: '<template-parameter>': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C2528: '<template-parameter>': pointer to reference is illegal
1>c:\program files (x86)\microsoft visual studio 14.0\vc\include\xutility(619): error C2528: '<template-parameter>': pointer to reference is illegal
1>  c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): note: see reference to class template instantiation 'std::iterator_traits<ConfigTests::ConfigLoadTests::bar::AClass &(*)>' being compiled
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C2039: 'difference_type': is not a member of 'std::iterator_traits<ConfigTests::ConfigLoadTests::bar::AClass &(*)>'
1>  c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): note: see declaration of 'std::iterator_traits<ConfigTests::ConfigLoadTests::bar::AClass &(*)>'
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C3646: 'difference_type': unknown override specifier
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(73): error C4430: missing type specifier - int assumed. Note: C++ does not support default-int
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(759): error C2528: 'p_buffer': pointer to reference is illegal
1>c:\ash\git\r4\externalrepos\etl\src\ivector.h(785): error C2528: 'p_buffer': pointer to reference is illegal
```

jwellbelove, 2016-12-29:
Hi Ashley,

You can't have a vector of references in C++.
If having references is important then take a look at 
std::reference_wrapper or etl::reference_wrapper.

John

On 29/12/2016 10:14, Ashley Roll wrote:
> etl::vector<AClass&,5> v;

AshleyRoll, 2016-12-29:
Hi John, 

Wow, sorry. How did I miss that.. that's embarrassing. Of course as there is no sensible initialisation.. doh!

Maybe I should not be reporting bugs until I get a clear head :)

My bad!
Ash.

closed 2016-12-29 by AshleyRoll

jwellbelove, 2016-12-30:
Don't worry it happens to all of us from time to time ;-)