ETLCPP/etl #16: type_def: Binding a reference to underlying type as TValue &

AshleyRoll, 2017-01-17:
I have a need to bind a reference to the underlying ```TValue``` type of an ```etl::type_def<>``` to pass it off to other code. This code is handling serialisation, but that is not important. The binding takes place at construction time of the consumer and the value that I wish to serialise changes (in my case it is a timestamp on a file).

```c++
// a dummy example, its a lot more complex
class Consumer
{
public:
    Consumer(int64_t const &val) : m_Value(val) {}

    void SerialiseNow(SomeStream s) 
    {
        // write the current value
        s.Write(m_Value);
    }
private:
    int64_t const &m_Value;
};

// --- later in program
// this is what I want to bind to the consumer

ETL_TYPEDEF(Timestamp, int64_t);

struct SomeDataThatChangesOverTime
{
    Timestamp time;
};

SomeDataThatChangesOverTime someData;

Consumer mySeraliser(someData.time);

// etc...
```

I can not create my consumer each time I need it (no dynamic memory allocation and other constraints in the design). And I can't implement a direct binding on the ```type_def<>``` without a lot of plumbing changes and probably circular dependencies.. 

Boost provides  reference and const reference operators on its equivalent strong typedef, which works for me (used in other places in my code currently as I'm in the middle of transitioning)

```c++
// template parameter T is the same as TValue
operator const T & () const {return t; }
operator T & () { return t; }
```

Can these be added to ```etl::type_def<>```?

Any thoughts?
Thanks,
Ash.

jwellbelove, 2017-01-17:
Seems like a reasonable suggestion.
I can't see any downsides at the moment.

It already has an convertion operator. They would replace it.

operator TValue() const
{
     return value;
}

jwellbelove, 2017-01-24:
Hi Ashley,I've put together an initial implementation of a 'user type', which seems to cover the requirements.ETL_DECLARE_USER_TYPE(MyType, uint32_t)ETL_USER_TYPE(First, 0)ETL_USER_TYPE(Step,  128)ETL_USER_TYPE(Last,  0xFFFF)ETL_END_USER_TYPE(MyType)ETL_DECLARE_USER_TYPE(MyType2, uint32_t)ETL_USER_TYPE(First, 0)ETL_USER_TYPE(Step,  256)ETL_USER_TYPE(Last,  0xFFFFFFFF)ETL_END_USER_TYPE(MyType2)MyType t;MyType2 t2;t = MyType::Last;    // OKt = 1;               // Errort = MyType(1);       // OK++t;                 // OKt += 2               // Errort += MyType(2)       // OKt += MyType::Step;   // OKt = t2;              // Errort += MyType2::Step;  // Erroruint32_t i = t.get();          // Copy value.uint32_t& ri = t.get();        // Bind internal value to reference.const uint32_t& cri = t.get(); // Bind internal value to const reference.//*****************************************************************************// The declaration of the structure.//*****************************************************************************#define ETL_DECLARE_USER_TYPE(TypeName, ValueType) \  struct TypeName \  { \    typedef ValueType value_type; \    TypeName() {} \    TypeName(const TypeName &other) : value(other.value) {} \    TypeName& operator=(const TypeName &other) { value = other.value; return *this; } \    explicit TypeName(ValueType value) : value(value) {} \    explicit operator ValueType() const { return value; } \    ValueType& get() { return value; } \    const ValueType& get() const { return value; } \    TypeName& operator ++() { ++value; return *this; } \    TypeName operator ++(int) { TypeName temp(*this); TypeName::operator ++(); return temp; } \    TypeName& operator --() { --value; return *this; } \    TypeName operator --(int) { TypeName temp(*this); TypeName::operator --(); return temp; } \    TypeName& operator +=(const TypeName& rhs) { value += rhs.value; return *this; } \    TypeName& operator -=(const TypeName& rhs) { value -= rhs.value; return *this; } \    TypeName& operator *=(const TypeName& rhs) { value *= rhs.value; return *this; } \    TypeName& operator /=(const TypeName& rhs) { value /= rhs.value; return *this; } \    TypeName& operator %=(const TypeName& rhs) { value %= rhs.value; return *this; } \    TypeName& operator &=(const TypeName& rhs) { value &= rhs.value; return *this; } \    TypeName& operator |=(const TypeName& rhs) { value |= rhs.value; return *this; } \    TypeName& operator ^=(const TypeName& rhs) { value ^= rhs.value; return *this; } \    TypeName& operator <<=(ValueType rhs) { value <<= rhs; return *this; } \    TypeName& operator >>=(ValueType rhs) { value >>= rhs; return *this; } \  private: \    ValueType value; \  public: \    enum enum_type \    {//*****************************************************************************// The predefined constants.//*****************************************************************************#define ETL_USER_TYPE(enum_name, value) \      enum_name = value,//*****************************************************************************// The final section of the structure.//*****************************************************************************#define ETL_END_USER_TYPE(TypeName) \    }; \    TypeName(enum_type value) : value(static_cast<value_type>(value)) {} \  };

AshleyRoll, 2017-01-24:
Hi John, 

Thanks, this looks good. I have a few minor changes below...

To bring other readers up to speed, John and I were discussing offline. Basically we want to be able to create a "User defined integer type" that is distinct from all other user defined types. In my case, one use is for a file system where I have ```Segment```s and ```Block```s for instance. These are numbers, but there are special "well known" values (constants) like ```INVALID```. They happen to both be ```uint32_t```, but it is important that the code doesn't get them mixed up. We also don't want the constant names to collide with other type constants (I want ```Block::INVALID``` and ```Segement::INVALID``` for instance) and also prevent the use of constants from other types (can't assign ```Block::INVALID``` to a ```Segment```).

The current ```type_def``` allows the compiler to implicitly convert to the value type and so there are some holes (like being able to compare values of different types). For implementation reasons I also needed a reference to the value type to be available that I can bind too which is not supported in the current ```type_def```. However we need to be explicit about getting that reference so we don't allow the accidental conversions to occur, this is done using the ```get()``` method.

### My Suggested Changes

- Rename ```enum_type``` to ```constants_type``` to make it a bit more obvious what it is doing, especially when using code completion in a modern editor
- Explicitly make the enum sized to the ```ValueType``` because it was possible, for instance, to declare a ```uint16_t``` based user type, and set a constant of ```0xFFFFFFFF``` and it would silently truncate and compile.

I complied and tested below in Visual Studio 2015 community.

```c++
#include <cstdint>

//*****************************************************************************
// The declaration of the structure.
//*****************************************************************************
#define ETL_DECLARE_USER_TYPE(TypeName, ValueType) \
  struct TypeName \
  { \
  typedef ValueType value_type; \
  TypeName() {} \
  TypeName(const TypeName &other) : value(other.value) {} \
  TypeName& operator=(const TypeName &other) { value = other.value; return *this; } \
  explicit TypeName(ValueType value) : value(value) {} \
  explicit operator ValueType() const { return value; } \
  ValueType& get() { return value; } \
  const ValueType& get() const { return value; } \
  TypeName& operator ++() { ++value; return *this; } \
  TypeName operator ++(int) { TypeName temp(*this); TypeName::operator ++(); return temp; } \
  TypeName& operator --() { --value; return *this; } \
  TypeName operator --(int) { TypeName temp(*this); TypeName::operator --(); return temp; } \
  TypeName& operator +=(const TypeName& rhs) { value += rhs.value; return *this; } \
  TypeName& operator -=(const TypeName& rhs) { value -= rhs.value; return *this; } \
  TypeName& operator *=(const TypeName& rhs) { value *= rhs.value; return *this; } \
  TypeName& operator /=(const TypeName& rhs) { value /= rhs.value; return *this; } \
  TypeName& operator %=(const TypeName& rhs) { value %= rhs.value; return *this; } \
  TypeName& operator &=(const TypeName& rhs) { value &= rhs.value; return *this; } \
  TypeName& operator |=(const TypeName& rhs) { value |= rhs.value; return *this; } \
  TypeName& operator ^=(const TypeName& rhs) { value ^= rhs.value; return *this; } \
  TypeName& operator <<=(ValueType rhs) { value <<= rhs; return *this; } \
  TypeName& operator >>=(ValueType rhs) { value >>= rhs; return *this; } \
  private: \
  ValueType value; \
  public: \
  enum constants_type : ValueType \
  {
//*****************************************************************************
// The predefined constants.
//*****************************************************************************
#define ETL_USER_TYPE(enum_name, value) \
      enum_name = value,

//*****************************************************************************
// The final section of the structure.
//*****************************************************************************
#define ETL_END_USER_TYPE(TypeName) \
    }; \
    TypeName(constants_type value) : value(static_cast<value_type>(value)) {} \
};

ETL_DECLARE_USER_TYPE(MyType, uint32_t)
ETL_USER_TYPE(First, 0)
ETL_USER_TYPE(Step, 128)
ETL_USER_TYPE(Last, 0xFFFF)
ETL_END_USER_TYPE(MyType)

ETL_DECLARE_USER_TYPE(MyType2, uint16_t)
ETL_USER_TYPE(First, 0)
ETL_USER_TYPE(Step, 256)
ETL_USER_TYPE(Last, 0xFFFF)
//ETL_USER_TYPE(TooBig, 0xFFFFFFFF)	// Error - truncation warning
ETL_END_USER_TYPE(MyType2)

int main(int argc, char** argv)
{
	MyType t;
	MyType2 t2;

	t = MyType::Last;    // OK
//	t = 1;               // Error
	t = MyType(1);       // OK
	++t;                 // OK
//	t += 2;              // Error
	t += MyType(2);      // OK
	t += MyType::Step;   // OK
//	t = t2;              // Error
//	t += MyType2::Step;  // Error
	uint32_t i = t.get();          // Copy value.
	uint32_t& ri = t.get();        // Bind internal value to reference.
	const uint32_t& cri = t.get(); // Bind internal value to const reference.

//	bool b = t == t2;	// Error - can't compare different types

	return 0;
}

```

AshleyRoll, 2017-01-25:
One slight wrinkle I just found..

It would be nice to be able to use a ```volatile``` version of the type for interrupts etc, so the type is a direct replacement for an integer. 

```c++
// MyType From above
volatile MyType vt;

vt = MyType::First;	// error
++vt;	// Error
// etc... I assume it applies to all the other operators
```
the errors are:
```
1>c:\temp\test\test.cpp(71): error C2678: binary '=': no operator found which takes a left-hand operand of type 'volatile MyType' (or there is no acceptable conversion)
1>  c:\temp\test\test.cpp(50): note: could be 'MyType &MyType::operator =(const MyType &)'
1>  c:\temp\test\test.cpp(71): note: while trying to match the argument list '(volatile MyType, MyType::constants_type)'
1>c:\temp\test\test.cpp(72): error C2678: binary '++': no operator found which takes a left-hand operand of type 'volatile MyType' (or there is no acceptable conversion)
1>  c:\temp\test\test.cpp(50): note: could be 'MyType &MyType::operator ++(void)'
1>  c:\temp\test\test.cpp(72): note: while trying to match the argument list '(volatile MyType)'
```

I don't know enough about the cv-qualifiers to know if this is a good or bad idea though. I quickly added some additional overloads for these two cases and discovered that you can't do duplicate assignment operators (one with volatile, one without) but you can with the increment operator.. I also don't know what the volatile assignment operator would do to optimisation.. Any thoughts?

```c++
volatile TypeName& operator ++() volatile { ++value; return *this; }
volatile TypeName& operator=(const TypeName &other) volatile { value = other.value; return *this; } 
```

I begin to feel for the std library developers :)

Ash.

jwellbelove, 2017-01-25:
>> Explicitly make the enum sized to the ValueType because it was possible, for instance, to declare a uint16_t based user type, and >> set a constant of 0xFFFFFFFF and it would silently truncate and compile.Unfortunately, adding a base type for the enum declaration is a C++11 addition and I want to keep the library C++03 backward compatible for now. I was thinking of adding some more ETL_XXX_SUPPORTED macros, similar to what the EASTL library does, to allow optimisations and additions supported by later versions of the compiler, so it is still an option. I would need to come up with a definitive table of what features are supported by which versions of compiler.I'll take look at the 'volatile' issue.I'm currently busy resolving changes due to my re-write of the etl::pool memory manager. It has required changes to most of the containers. It will require some thorough testing and review of the code mods.The benchmarks are looking good for what I've tested so far.http://www.etlcpp.com/benchmarks.htmlOn 24 January 2017 at 23:29 Ashley Roll <notifications@github.com> wrote:Hi John,Thanks, this looks good. I have a few minor changes below...To bring other readers up to speed, John and I were discussing offline. Basically we want to be able to create a "User defined integer type" that is distinct from all other user defined types. In my case, one use is for a file system where I have Segments and Blocks for instance. These are numbers, but there are special "well known" values (constants) like INVALID. They happen to both be uint32_t, but it is important that the code doesn't get them mixed up. We also don't want the constant names to collide with other type constants (I want Block::INVALID and Segement::INVALID for instance) and also prevent the use of constants from other types (can't assign Block::INVALID to a Segment).The current type_def allows the compiler to implicitly convert to the value type and so there are some holes (like being able to compare values of different types). For implementation reasons I also needed a reference to the value type to be available that I can bind too which is not supported in the current type_def. However we need to be explicit about getting that reference so we don't allow the accidental conversions to occur, this is done using the get() method.My Suggested ChangesRename enum_type to constants_type to make it a bit more obvious what it is doing, especially when using code completion in a modern editorExplicitly make the enum sized to the ValueType because it was possible, for instance, to declare a uint16_t based user type, and set a constant of 0xFFFFFFFF and it would silently truncate and compile.I complied and tested below in Visual Studio 2015 community.#include <cstdint>

//*****************************************************************************
// The declaration of the structure.
//*****************************************************************************
#define ETL_DECLARE_USER_TYPE(TypeName, ValueType) \
  struct TypeName \
  { \
  typedef ValueType value_type; \
  TypeName() {} \
  TypeName(const TypeName &other) : value(other.value) {} \
  TypeName& operator=(const TypeName &other) { value = other.value; return *this; } \
  explicit TypeName(ValueType value) : value(value) {} \
  explicit operator ValueType() const { return value; } \
  ValueType& get() { return value; } \
  const ValueType& get() const { return value; } \
  TypeName& operator ++() { ++value; return *this; } \
  TypeName operator ++(int) { TypeName temp(*this); TypeName::operator ++(); return temp; } \
  TypeName& operator --() { --value; return *this; } \
  TypeName operator --(int) { TypeName temp(*this); TypeName::operator --(); return temp; } \
  TypeName& operator +=(const TypeName& rhs) { value += rhs.value; return *this; } \
  TypeName& operator -=(const TypeName& rhs) { value -= rhs.value; return *this; } \
  TypeName& operator *=(const TypeName& rhs) { value *= rhs.value; return *this; } \
  TypeName& operator /=(const TypeName& rhs) { value /= rhs.value; return *this; } \
  TypeName& operator %=(const TypeName& rhs) { value %= rhs.value; return *this; } \
  TypeName& operator &=(const TypeName& rhs) { value &= rhs.value; return *this; } \
  TypeName& operator |=(const TypeName& rhs) { value |= rhs.value; return *this; } \
  TypeName& operator ^=(const TypeName& rhs) { value ^= rhs.value; return *this; } \
  TypeName& operator <<=(ValueType rhs) { value <<= rhs; return *this; } \
  TypeName& operator >>=(ValueType rhs) { value >>= rhs; return *this; } \
  private: \
  ValueType value; \
  public: \
  enum constants_type : ValueType \
  {
//*****************************************************************************
// The predefined constants.
//*****************************************************************************
#define ETL_USER_TYPE(enum_name, value) \
      enum_name = value,

//*****************************************************************************
// The final section of the structure.
//*****************************************************************************
#define ETL_END_USER_TYPE(TypeName) \
    }; \
    TypeName(constants_type value) : value(static_cast<value_type>(value)) {} \
};

ETL_DECLARE_USER_TYPE(MyType, uint32_t)
ETL_USER_TYPE(First, 0)
ETL_USER_TYPE(Step, 128)
ETL_USER_TYPE(Last, 0xFFFF)
ETL_END_USER_TYPE(MyType)

ETL_DECLARE_USER_TYPE(MyType2, uint16_t)
ETL_USER_TYPE(First, 0)
ETL_USER_TYPE(Step, 256)
[... cut ...]
// t = 1;               // Error
 t = MyType(1);       // OK
 ++t;                 // OK
// t += 2;              // Error
 t += MyType(2);      // OK
 t += MyType::Step;   // OK
// t = t2;              // Error
// t += MyType2::Step;  // Error
 uint32_t i = t.get();          // Copy value.
 uint32_t& ri = t.get();        // Bind internal value to reference.
 const uint32_t& cri = t.get(); // Bind internal value to const reference.

// bool b = t == t2; // Error - can't compare different types

 return 0;
}

AshleyRoll, 2017-01-25:
Hi John,

Ah.. yes that C++11 thing again.. sorry. I jumped from C++98 to C++11 after many years of C# so I'm a bit confused on what is in which standard. 

what about this (or similar) to enforce value ranges.. 

```c++
// note macro parameter value renamed to val to avoid confusion with ::value tokens
#define ETL_USER_TYPE(enum_name, val) \
      enum_name = etl::integral_constant<value_type, val>::value,
```

If we do that, the microsoft compiler at least issues a warning about truncation because we are forcing it to convert rather than silently changing the enum "size" under the hood.

The changes needed to support volatile would be a nice to have for the future (certainly not a show stopper) and would nicely complete the user type. The work you are putting into this library is great. 

Thanks. 
Ash.

jwellbelove, 2017-01-25:
Hi Ashley,>> enum_name = etl::integral_constant<value_type, val>::value,That's a neat idea!On the volatile front, I've done a small experiment to see what sort of changes the compiler requires to be happy.It seems that you must set the return type to void for volatile functions that would normally return a reference otherwise you will get the warning: implicit dereference will not access object of type 'volatile MyType' in statement ETL_DECLARE_USER_TYPE(MyType, int)Here's a flavour of what additions are needed. This is not a complete list.TypeName(const volatile TypeName &other) : value(other.value) {} \explicit operator ValueType() volatile const { return value; } \void operator =(const volatile TypeName &other) volatile { value = other.value; } \void operator ++() volatile { ++value; } \volatile TypeName operator ++(int) volatile { volatile TypeName temp(*this); TypeName::operator ++(); return temp; } \I'm a little unsure about whether the underlying type should be volatile or not, or whether making the user type volatile is good enough on its own.John.

closed 2018-05-30 by jwellbelove