ETLCPP/etl #19: master branch doesn't compile

AshleyRoll, 2017-02-27:
Hi John,

I've just updated to the latest master and I'm getting errors on Microchip XC32 1.42 (basically GCC). Not sure if it is an issue with my standard library, or something funky with the merge you did.

There seems to be a file merge marker at the end of ```src\algorithm.h```.

After I edited that out, then I get a series of errors on the ```is_input_iterator```, ```is_output_iterator```, ```is_bidirectional_iterator``` and ```is_random_iterator``` templates.

specifically I'm seeing: 

```
Radio/../../../../ExternalRepos/etl/src/iterator.h:46:113: error: type/value mismatch at argument 1 in template parameter list for 'template<class T1, class T2> struct etl::is_same'
     static const bool value = etl::is_same<::std::iterator_traits<T>::iterator_category, std::input_iterator_tag>::value;
                                                                                                                 ^
Radio/../../../../ExternalRepos/etl/src/iterator.h:46:113: error:   expected a type, got 'std::iterator_traits<_Iter>::iterator_category'
Radio/../../../../ExternalRepos/etl/src/iterator.h:52:114: error: type/value mismatch at argument 1 in template parameter list for 'template<class T1, class T2> struct etl::is_same'
     static const bool value = etl::is_same<::std::iterator_traits<T>::iterator_category, std::output_iterator_tag>::value;
                                                                                                                  ^
Radio/../../../../ExternalRepos/etl/src/iterator.h:52:114: error:   expected a type, got 'std::iterator_traits<_Iter>::iterator_category'
Radio/../../../../ExternalRepos/etl/src/iterator.h:58:115: error: type/value mismatch at argument 1 in template parameter list for 'template<class T1, class T2> struct etl::is_same'
     static const bool value = etl::is_same<::std::iterator_traits<T>::iterator_category, std::forward_iterator_tag>::value;
                                                                                                                   ^
Radio/../../../../ExternalRepos/etl/src/iterator.h:58:115: error:   expected a type, got 'std::iterator_traits<_Iter>::iterator_category'
Radio/../../../../ExternalRepos/etl/src/iterator.h:64:121: error: type/value mismatch at argument 1 in template parameter list for 'template<class T1, class T2> struct etl::is_same'
     static const bool value = etl::is_same<::std::iterator_traits<T>::iterator_category, std::bidirectional_iterator_tag>::value;
```

jwellbelove, 2017-03-02:
I think it may have been due to a missing 'typename' before the 'std::iterator_traits<T>::iterator_category'.
I've updated the master branch. Can you try again and let me know?

AshleyRoll, 2017-03-02:
Hi John,

That has fixed the iterator errors, thanks!

There is still an issue in ```src\algorithm.h``` Line 698 is a git merge marker. I have to comment this out to get it to build. 

Thanks heaps,
Ash.

AshleyRoll, 2017-03-08:
Thanks for the fixes in the update John, but I'm not getting another error in ```algorithm.h```

```
../Lib/BSGStream/../../../../ExternalRepos/etl/src/algorithm.h: In function 'typename etl::enable_if<(etl::is_random_iterator<T>::value && etl::is_random_iterator<TOutputIterator>::value), void>::type etl::transform_n(TInputIterator1, TInputIterator2, TSize, TOutputIterator, TBinaryFunction)':
In file included from ../Lib/BSGStream/../../../../ExternalRepos/etl/src/array.h:44:0,
                 from ../Lib/BSGStream/../../../../ExternalRepos/etl/src/queue.h:40,
                 from ../Lib/BSGStream/Encoder.h:10,
                 from ../Lib/BSGStream/Encoder.cpp:1:
../Lib/BSGStream/../../../../ExternalRepos/etl/src/algorithm.h:714:20: error: 'i_begin' was not declared in this scope
     std::transform(i_begin, i_begin + n, i_begin2, o_begin, function);
                    ^
```

If I change both instances of ```i_begin``` to ```i_begin1``` it is all good.

Thanks,
Ash.

closed 2017-03-10 by AshleyRoll