ETLCPP/etl #21: build problems using IAR for ARM

RallyTronics, 2017-03-08:
I have run into a few issues:

as previously mentioned identifier i_begin is undefined in transform_n in algorithm.h

I also get undefined identifiers for: 
Error[Pe020]: identifier "UINT8_MAX" is undefined etl\smallest.h 255 
Error[Pe020]: identifier "UINT16_MAX" is undefined etl\smallest.h 256 
Error[Pe020]: identifier "UINT32_MAX" is undefined etl\smallest.h 257 
Error[Pe020]: identifier "INT8_MAX" is undefined etl\smallest.h 276 
Error[Pe020]: identifier "INT8_MIN" is undefined etl\smallest.h 276 
Error[Pe020]: identifier "INT16_MAX" is undefined etl\smallest.h 277 
Error[Pe020]: identifier "INT16_MIN" is undefined etl\smallest.h 277 
Error[Pe020]: identifier "INT32_MAX" is undefined etl\smallest.h 278 
Error[Pe020]: identifier "INT32_MIN" is undefined etl\smallest.h 278

jwellbelove, 2017-03-10:
The error in 'transform_n' has been corrected.

UINT8_MAX etc.
That's odd as all of those definitions should be in <stdint.h>, which 'smallest.h' includes.
See http://www.cplusplus.com/reference/cstdint/

jwellbelove, 2017-03-10:
Could this help?
http://stackoverflow.com/questions/16748392/cant-find-symbols-in-stdint-h

RallyTronics, 2017-03-10:
That took care of the issue.  The last error I am getting is

Error[Pe937]: a class or namespace qualified name is required iflat_map.h 466 

which is on this statement: _typename const_iterator i_element = lower_bound(key)_; in the function below:

    //*********************************************************************
    /// Returns a const reference to the value at index 'key'
    /// If asserts or exceptions are enabled, emits an etl::flat_map_out_of_bounds if the key is not in the range.
    ///\param i The index.
    ///\return A const reference to the value at index 'key'
    //*********************************************************************
    const mapped_type& at(key_value_parameter_t key) const
    {
      typename const_iterator i_element = lower_bound(key);

      ETL_ASSERT(i_element != end(), ETL_ERROR(flat_map_out_of_bounds));

      return i_element->second;
    }

jwellbelove, 2017-03-11:
OK, removed the surplus 'typename'.

RallyTronics, 2017-03-13:
That did it, thanks!

closed 2017-03-13 by RallyTronics

ben-edna, 2019-03-19:
For my IAR based project, I had to add `__STDC_LIMIT_MACROS` to the project as well when I used `etl/largest.h`. It might help other people to add this to https://www.etlcpp.com/compilers.html?