ETLCPP/etl #27: Usage of c++11 std::nullptr_t and std::alignment_of

AODtorusan, 2017-05-02:
Hi,

I'm trying to compile etl as a static library in Ti Code Composer Studio (Ti compiler v16.9.2.LTS), but I get the following errors in the src tree;

```
Description                                       Resource            Path       Location   Type
#136 namespace "std" has no member "nullptr_t"    type_traits.h       /etl/src   line 212   C/C++ Problem
#20 identifier "nullptr" is undefined             error_handler.cpp   /etl/src   line 37    C/C++ Problem
#918 alignment_of is not a template               type_traits.h       /etl/src   line 388   C/C++ Problem
```

Both the std::nullptr_t and std::alignment_of seem to be newly introduced in c++11 (which is not yet supported by this compiler flavor, only c++03).

jwellbelove, 2017-05-02:
Hi Simon,

The ETL checks the compiler version in platform.h to see if it natively supports nullptr.
type_traits.h also checks the compiler type to determine the correct syntax for alignment.

I am on holiday in Spain at the moment and don’t have access to a decent computer with the appropriate tools to modify and check the code. I will be back in the UK next week.

If you want to try to get this working now, then here is what you need to do...

There are two places that will need modifications to support your compiler.

platform.h

Modify this line, or add another.
#elif defined(__TI_COMPILER_VERSION__) && defined(__MSP430__)
  #define ETL_COMPILER_TI_MSP430

Modify this too so that your compiler is included.
// Check to see if the compiler supports nullptr and large character types.
#if (defined(ETL_COMPILER_MICROSOFT) && (_MSC_VER < 1600)) || \
     defined(ETL_COMPILER_KEIL) || \
     defined(ETL_COMPILER_TI_MSP430) || \
     defined(ETL_COMPILER_IAR) || \
     (defined(ETL_COMPILER_GCC) && (__cplusplus < 201103L))
  #define ETL_NO_NULLPTR_SUPPORT
  #define ETL_NO_LARGE_CHAR_SUPPORT
#endif

type_traits.h

Modify this line, or add another. __ALIGNOF__ is probably good for your compiler.
#ifdef ETL_COMPILER_TI_MSP430
  template <typename T> struct alignment_of : integral_constant<size_t, size_t(__ALIGNOF__(T))> {};
#endif

Let me know if you have any questions regarding this. 
It maybe that a general TI compiler entry could be used rather than a target specific one.

John.

From: Simon Billemont
Sent: 02 May 2017 15:04
To: ETLCPP/etl
Cc: Subscribed
Subject: [ETLCPP/etl] Usage of c++11 std::nullptr_t and std::alignment_of(#27)

Hi,
I'm trying to compile etl as a static library in Ti Code Composer Studio (Ti compiler v16.9.2.LTS), but I get the following errors in the src tree;
Description                                       Resource            Path       Location   Type
#136 namespace "std" has no member "nullptr_t"    type_traits.h       /etl/src   line 212   C/C++ Problem
#20 identifier "nullptr" is undefined             error_handler.cpp   /etl/src   line 37    C/C++ Problem
#918 alignment_of is not a template               type_traits.h       /etl/src   line 388   C/C++ Problem
Both the std::nullptr_t and std::alignment_of seem to be newly introduced in c++11 (which is not yet supported by this compiler flavor, only c++03).

AODtorusan, 2017-05-03:
Hello,

Thanks for the tip, I'm using the MSP432 platform so that explains the issue I'm seeing. Looking at my TI CCS project, it has the following defines; 

- ```__MSP432P401R__```
- ```TARGET_IS_MSP432P4XX```

After changing platform.h by adding a check for one of these defines, to force the same behavior as the msp430, it compiles cleanly.

However since its fundamentally a different core architecture, perhaps a dedicated define would be better.

Simon

jwellbelove, 2017-05-03:
That’s great to hear you’ve fixed it. I may have to investigate how many other TI targets there are.
Any chance you could push your changes back to Git?

John.

From: Simon Billemont
Sent: 03 May 2017 10:30
To: ETLCPP/etl
Cc: John Wellbelove; Comment
Subject: Re: [ETLCPP/etl] Usage of c++11 std::nullptr_t and std::alignment_of(#27)

Hello,
Thanks for the tip, I'm using the MSP432 platform so that explains the issue I'm seeing. Looking at my TI CCS project, it has the following defines;
• __MSP432P401R__
• TARGET_IS_MSP432P4XX
After changing platform.h by adding a check for one of these defines, to force the same behavior as the msp430, it compiles cleanly.
However since its fundamentally a different core architecture, perhaps a dedicated define would be better.
Simon

AODtorusan, 2017-05-09:
Hello,

I was about to create a pull request, but I see you already removed the MSP430 check from the source code.
It took a while because I was doing some porting some of the code to work with etl. So far did not see any other msp432 specific issues. so I'll close this issue.

closed 2017-05-09 by AODtorusan