ETLCPP/etl #30: Ambiguous string, vector, etc. realizations for byte coping purpose

mrdimfox, 2017-06-05:
If you have some object of type string<N> and you try to make memcpy of it, you find out that method data() of copied obj returns an ambiguous pointer to the buffer. It happens because of the string<N> class stores the buffer's pointer separately in the base class but while copying the real address of the buffer is changed. It can be critical for some embedded purposes.

jwellbelove, 2017-06-07:
Using memcpy on C++ objects is usually a very bad idea. It can be easily broken by changes to the class or its bases, as you have found.I've had to spend ages in the past where a developer memcpy'd a class in a number of places, which caused issues when a new member was added to it. The bug was subtle and time consuming to track down. Don't do it.

mrdimfox, 2017-06-11:
> Using memcpy on C++ objects is usually a very bad idea

Yes, I fully agree with you. But we have deal with implementations of std containers for embedded proposes so it will be very useful to resolve this issue. 
For example, if I use queues in the FreeRTOS OS, I can't avoid memcpy, cause FreeRTOS use it for all stuff placed in queues.

jwellbelove, 2017-06-15:
Any view on my suggestion?The 'fixup' option would be relatively simple to implement for strings, vector and possibly deque.A quick test shows it to be a viable, low-level option.

mrdimfox, 2017-06-15:
> Any view on my suggestion?

@jwellbelove, I've already answered you above, that I agree with you. But there are some cases, for example when we use some C libs, we can't avoid a memcpy cause we can't change the implementation.

jwellbelove, 2017-06-15:
I meant the suggestion that I add a fixup() function to the top level of the container classes to repair the internal pointer after a memcpy to a raw buffer, and a reinterpret cast back from a raw pointer to the original container reference.

mrdimfox, 2017-06-15:
Oh, okay. I think it would be great.

closed 2018-02-25 by jwellbelove