ETLCPP/etl #31: problem with alignment on platforms that don't support int8_t

RallyTronics, 2017-06-06:
working on a TI platform that does not support an 8-bit data type I get errors such as:

"alignment.h", line 99: error: 
          identifier "int8_t" is undefined
      typedef typename __private_alignment__::type_with_alignment_helper<ALIGNMENT, int8_t, int16_t, int32_t, int64_t, float, double, void*>::type type;

this is the offending template:

  ```
template <const size_t ALIGNMENT>
  class type_with_alignment
  {
  public:

    typedef typename __private_alignment__::type_with_alignment_helper<ALIGNMENT, int8_t, int16_t, int32_t, int64_t, float, double, void*>::type type;
  };

```

for now I have defined uint8_t=uint16_t

there should probably be a check based on your ETL_8BIT_SUPPORT macro

jwellbelove, 2017-06-07:
Thanks, I'll either add an #ifdef to the code or look to see if changing the uint8_t to uint_least8_t would work.

jwellbelove, 2017-06-10:
I've modified alignment.h to fix the issue.

closed 2018-02-25 by jwellbelove