ETLCPP/etl #32: Problem with unsorted_map using with etl::string

mrdimfox, 2017-06-11:
I've tried to create this type of unsorted_map, but it didn't work.

```
struct Credentials {
    using Key = etl::string<32>;
    using Value = Result::Token;

    Credentials() {
        map.insert(std::pair<Key, Value>("USERNAME", Value::USERNAME));
        map.insert(std::pair<Key, Value>("PASSWORD", Value::PASSWORD));
    }

    etl::flat_map<Key, Value, 2> map;
} credentials;
```

I get an error during compilation:

```
In file included from ../lib/etl/iunordered_map.h:47:0,
                 from ../lib/etl/unordered_map.h:38,
                 from ../utils/keepass/AutotypeReader.cpp:6:
../lib/etl/intrusive_forward_list.h: In instantiation of 'class etl::intrusive_forward_list<etl::iunordered_map<etl::string<32>, utils::keepass::AutotypeReader::Result::Data::Key, etl::hash<etl::string<32> >, std::equal_to<etl::string<32> > >::node_t, etl::forward_link<0> >':
../lib/etl/iunordered_map.h:160:47:   required from 'class etl::iunordered_map<etl::string<32>, utils::keepass::AutotypeReader::Result::Data::Key, etl::hash<etl::string<32> >, std::equal_to<etl::string<32> > >'
../lib/etl/unordered_map.h:57:9:   required from 'class etl::unordered_map<etl::string<32>, utils::keepass::AutotypeReader::Result::Data::Key, 2>'
../utils/keepass/AutotypeReader.cpp:24:39:   required from here
../lib/etl/intrusive_forward_list.h:151:38: warning: comparison between 'enum etl::__private_intrusive_links__::forward_link_base<etl::forward_link<0>, 0, 0>::<unnamed>' and 'enum etl::link_option::<unnamed>' [-Wenum-compare]
       COUNT_OPTION = ((TLink::OPTION == etl::link_option::AUTO) ||
                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~
../lib/etl/intrusive_forward_list.h:152:38: warning: comparison between 'enum etl::__private_intrusive_links__::forward_link_base<etl::forward_link<0>, 0, 0>::<unnamed>' and 'enum etl::link_option::<unnamed>' [-Wenum-compare]
                       (TLink::OPTION == etl::link_option::CHECKED)) ? etl::count_option::SLOW_COUNT : etl::count_option::FAST_COUNT
                       ~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from ../lib/etl/unordered_map.h:38:0,
                 from ../utils/keepass/AutotypeReader.cpp:6:
../lib/etl/iunordered_map.h: In instantiation of 'class etl::iunordered_map<etl::string<32>, utils::keepass::AutotypeReader::Result::Data::Key, etl::hash<etl::string<32> >, std::equal_to<etl::string<32> > >':
../lib/etl/unordered_map.h:57:9:   required from 'class etl::unordered_map<etl::string<32>, utils::keepass::AutotypeReader::Result::Data::Key, 2>'
../utils/keepass/AutotypeReader.cpp:24:39:   required from here
../lib/etl/iunordered_map.h:1237:12: error: 'etl::iunordered_map<TKey, T, THash, TKeyEqual>::key_hash_function' has incomplete type
     hasher key_hash_function;
            ^~~~~~~~~~~~~~~~~
```

mrdimfox, 2017-06-12:
Are there any chances to resolve this? Or I don't understand something?
This issue is actual for the last commit and for ver. 9.1.3

jwellbelove, 2017-06-13:
I've created a new release 9.1.4 that contains all of the current changes.The example code you show is using etl::flat_map, but the errors are indicating that the problem is with the use of etl::unordered_map.

mrdimfox, 2017-06-13:
> The example code you show is using etl::flat_map, but the errors are indicating that the problem is with the use of etl::unordered_map.

Yes, I know. But it is just a typo. I have already pointed it out in my issue. Here the right example:
```
struct Credentials {
    using Key = etl::string<32>;
    using Value = Result::Token;

    Credentials() {
        map.insert(std::pair<Key, Value>("USERNAME", Value::USERNAME));
        map.insert(std::pair<Key, Value>("PASSWORD", Value::PASSWORD));
    }

    etl::unsorted_map<Key, Value, 2> map;
} credentials;
```
And I get an error form original post.

mrdimfox, 2017-06-13:
Omg, I am always writing «unsorted» instead of «unordered» -_-
Okey, everywhere I mean only «unordered_map» here above.

jwellbelove, 2017-06-13:
OK, I'll check this example out.

mrdimfox, 2017-06-13:
I have the better example for illustrating the problem. By the way, the issue is actual for ver. 9.1.4

```
template<typename T, typename U, std::size_t N>
class Dict {
public:
	using Key = T;
	using Value = U;
	static constexpr std::size_t SIZE = N;

	using DictType = etl::unordered_map<T, U, N>;
	using Pair = std::pair<Key, Value>;

	using iterator = typename DictType::iterator;
	using const_iterator = typename DictType::const_iterator;

	Dict() = default;
	~Dict() = default;

	const_iterator get(const Key& key) const {
		return _map.find(key);
	}

	iterator get(const Key& key) {
		return _map.find(key);
	}

	Value get_value(const Key& key) {
		auto res = _map.find(key);
		if (res != not_a_value()) {
			return res->second;
		}
		return _default_value;
	}

	const_iterator not_a_value() {
		return _map.end();
	}

	void set_default(const Value& value) {
		_default_value = value;
	}

protected:
	void _add(const Key& key, const Value& value) {
		_map.insert(Pair(key, value));
	}

	DictType _map;
	Value _default_value;
};

enum class Tag {
	NO_TAG,
	ENTRY,
	GROUP,
	VALUE,
	KEY,
	NAME,
	HISTORY,
	KEEPASS,
	DEFAULT_SEQUENCE,
};

struct _TAG_DICT : public utils::memory::Dict<etl::cstring<64>, Tag, 9> {
	_TAG_DICT() {
		_add(Key("Group"), Tag::GROUP);
		_add(Key("Entry"), Tag::ENTRY);
		_add(Key("Name"), Tag::NAME);
		_add(Key("Key"), Tag::KEY);
		_add(Key("Value"), Tag::VALUE);
		_add(Key("History"), Tag::HISTORY);
		_add(Key("KeePassFile"), Tag::KEEPASS);
		_add(Key("DefaultSequence"), Tag::DEFAULT_SEQUENCE);

		set_default(Tag::NO_TAG);
	}
} static TAG_DICT;
```

jwellbelove, 2017-06-14:
Correction:One of the lines added in the new hash template should be...reinterpret_cast<const uint8_t*>(&text[text.size()]))NOTreinterpret_cast<const uint8_t*>(&text[SIZE]))

jwellbelove, 2017-06-15:
Hopefully all sorted with the latest release.Can you confirm?

mrdimfox, 2017-06-15:
Yes, everything is alright, thanks!

closed 2019-02-11 by jwellbelove