ETLCPP/etl #34: etl::set : lower_bound does not behave as stl::set::lower_bound

NeimadG, 2017-06-12:
In test_set, create a test that use (in pseudocode) : 
etl::set<int, 10> etlset;
and
stl::set<int> stlset;
both with values {0, 2, 4, 6, 8, 10, 12, 14, 16, 18}

with i in {1, 3, 5, 7, 9, 11, 13, 15, 17} : 
stlset.lower_bound(i) != etlset.lower_bound(i)

As result, etl::set::equal_range != stl::set::equal_range

Damien

jwellbelove, 2017-06-12:
OK, I'll check that out and, if necessary, add to the unit tests.John.

jwellbelove, 2019-10-12:
Do you know if this is still an issue?

NeimadG, 2019-10-31:
Hi,
Yes the issue still exists. I've just cloned the repo and added a test fixture in test_set.cpp
Do you want me to push my new version ?

jwellbelove, 2019-10-31:
You've fixed it, or written a test just to highlight it?

NeimadG, 2019-10-31:
only written the test for now... I'm installing IDE for debugging, no fix yet

jwellbelove, 2019-11-01:
If you could do a pull request for the test fixture to a 'hotfix' branch on Github, that would be great.

NeimadG, 2019-11-03:
I have a fix, but I don't know how to use github... Am I allowed to create a branch for pull request ?

jwellbelove, 2019-11-03:
Create a local hotfix branch on your computer and then do a pull request to the ETL project on github
Have you just done the test and fix for etl::set? I imagine the fix will be applicable to map, multimap & multiset too.

https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request

NeimadG, 2019-11-03:
I can have a look to the other containers. But I cannot push anything (or I don't know how to proceed). Don't I need to be a contributor ? Do I need some kind of privilege ?

jwellbelove, 2019-11-04:
https://help.github.com/en/github/collaborating-with-issues-and-pull-requests/creating-a-pull-request-from-a-fork

Or if you are really stuck, then send me a zip of your branch.

NeimadG, 2019-11-04:
pull request open (from fork). I found issues on set & map only.

jwellbelove, 2019-11-05:
Awesome! Thanks.

closed 2020-03-28 by jwellbelove