ETLCPP/etl #37: etl::ilist::sort does not preserve equivalent nodes order

NeimadG, 2017-06-13:
when sorting a list with some equal element, e.g. (list of ints, with operator<)
5, 15, 10 _(a)_, 10 _(b)_, 10 _(c)_, 20
the result is
5, 10 _(c)_, 10 _(b)_, 10 _(a)_, 15, 20
instead of
5, 10 _(a)_, 10 _(b)_, 10 _(c)_, 15, 20

where 10 _(a)_ denotes the int value 10 with address _a_

in ilist.h (on older version, I'm currently updating) replacing
else if (compare(*i_left, *i_right)
by
else if (!compare(*i_right, *i_left)

solves the problem

Cheers

jwellbelove, 2017-06-13:
Thanks, I'll look into that. The sort is supposed to be stable.

jwellbelove, 2017-06-13:
I shall also add to the unit tests to check that this will always be correct.

jwellbelove, 2017-06-15:
Fixed in the latest release.

closed 2019-11-05 by NeimadG