ETLCPP/etl #43: etl::string(const char* origin, size_t sz) bad behavior

tntodorov, 2017-07-11:
The above mentioned constructor is IMO buggy. I am using an `etl::string<MSG_SZ>` as a holder for messages to show on a screen, which are coming from arbitrary length sources. When the source is shorter than `MSG_SZ`, copying the contents in this manner results in garbage at the end of the `istring `buffer, which in turn gives the wrong value for `length()` and `size()`.

In some cases (when the original buffer may not be properly terminated) the other constructor `etl::string(const char*)` is not usable, so I have to use this one.

If the actual origin `cstring` is shorter, the resulting object is filled with the random crap that is in the SRAM after the `cstring` end, whatever is needed to satisfy the given `sz` requirement. But we are dealing with c-style strings after all. I believe that duplicating one of these should always end at the \0 terminator, even when a longer size is indicated.

My suggested fix is to replace line 585 in basic_string.h from

`while ((first != last) && (current_size != CAPACITY))`

to

`while ((first != last) `**`&& (first != '\0')`**`  && (current_size != CAPACITY))`

tntodorov, 2017-07-11:
Sorry, the actual fix should read:

while ((first != last)&& (*first != '\0')&& (current_size != CAPACITY))

jwellbelove, 2017-07-13:
You could use this method of construction.const size_t MAX = 100;etl::vector<char, MAX> buffer;// Add characters to buffer. May or may not be null terminated.// Construct the string from the buffer, to first null or the last character.etl::string<MAX> text(buffer.begin(), std::find(buffer.begin(), buffer.end(), '\0'));

closed 2018-05-30 by jwellbelove