ETLCPP/etl #47: std::string_view support

mrdimfox, 2017-10-19:
In c++17 you can use awesome class std::string_view. See more here: http://en.cppreference.com/w/cpp/string/basic_string_view
Can you add a support of a casting to std::string_view from etl::string and back? It will be extremely useful for embedded software purposes.

jwellbelove, 2017-12-07:
std::string_view should be usable as-is, just like std::string.

etl::string text("Hello World");
std::string_view view(text.data(), text.size());

mrdimfox, 2017-12-08:
Yes, sure. But it could be a string_view operator there in the etl::string class. string_view is a part of the standard lib (since C++17) now, so it can be useful to have one, IMHO. 😄

jwellbelove, 2017-12-09:
OK, I will look into it.

closed 2018-05-30 by jwellbelove