ETLCPP/etl #49: Created Slack group

jwellbelove, 2017-11-01:
I have created a Slack group to better support the ETL.
You can find it [here](https://join.slack.com/t/etlcpp/shared_invite/enQtMjY1MjcyNDAzNDYxLTU0ODY4N2M3MGMwNGQ3YmVlNmI1NzY4MGViNmMxNDgxMjhkOTU3MzE0MzBlMGI5M2E1YjlkYzFjZGJjYTQ2ZDU ).

apmorton, 2018-01-02:
>This invite link is no longer active.

According to slack: 

> For security, invite links will expire and can be used by up to 1000 people. If you need to, you can get a new link.

I suggest you list the slack team on [slofile](https://slofile.com/) - which is a directory of public slack teams, but also has a built in invite feature that won't self destruct after some period of time like slacks shared invite feature.

jwellbelove, 2018-01-02:
Thanks. I've updated the link to take people to slofile.

closed 2018-05-30 by jwellbelove