ETLCPP/etl #51: Google mock support?

jeremyherbert, 2017-11-09:
Hi,

I'm trying to use gmock with ETL. Has anyone got this working? 

The following code works for me:

```
#include "gtest.h"
#include "gmock.h"

#include "etl/src/vector.h"

class GetSomething {
public:
    virtual int get_something(etl::ivector<int> *input) { return (*input)[0]; };
};

class GetSomethingMock : public GetSomething {
public:
    MOCK_METHOD1(get_something, int(etl::ivector<int> *input));
};

using testing::Return;

TEST(SimpleTest, CallWithVector) {
    GetSomethingMock mock;

    etl::vector<int, 5> v;
    v.push_back(10);

    EXPECT_CALL(mock, get_something(&v))
                .Times(1)
                .WillRepeatedly(Return(12));

    EXPECT_EQ(mock.get_something(&v), 12);
}
```

However, if I change the signature of the `get_something` function to take an argument by reference, I receive compiler errors:

```
#include "gtest.h"
#include "gmock.h"

#include "etl/src/vector.h"

class GetSomething {
public:
    virtual int get_something(etl::ivector<int> &input) { return input[0]; };
};

class GetSomethingMock : public GetSomething {
public:
    MOCK_METHOD1(get_something, int(etl::ivector<int> &input));
};

using testing::Return;

TEST(SimpleTest, CallWithVector) {
    GetSomethingMock mock;

    etl::vector<int, 5> v;
    v.push_back(10);

    EXPECT_CALL(mock, get_something(v))
                .Times(1)
                .WillRepeatedly(Return(12));

    EXPECT_EQ(mock.get_something(v), 12);
}
```

and the output:
```
In file included from /Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:6:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock.h:61:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-generated-function-mockers.h:43:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-spec-builders.h:75:
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:3748:43: error: calling a private constructor of class 'etl::ivector<int>'
Matcher<T>::Matcher(T value) { *this = Eq(value); }
                                          ^
/Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:28:37: note: in instantiation of member function 'testing::Matcher<etl::ivector<int> &>::Matcher' requested here
    EXPECT_CALL(mock, get_something(v))
                                    ^
/Users/jeremy/Dropbox/code/fw_test/submodules/etl/src/vector.h:959:5: note: declared private here
    ivector(const ivector&);
    ^
In file included from /Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:6:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock.h:61:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-generated-function-mockers.h:43:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-spec-builders.h:75:
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:3743:48: error: call to implicitly-deleted copy constructor of 'internal::EqMatcher<ivector<int> >'
inline internal::EqMatcher<T> Eq(T x) { return internal::EqMatcher<T>(x); }
                                               ^~~~~~~~~~~~~~~~~~~~~~~~~
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:3748:40: note: in instantiation of function template specialization 'testing::Eq<etl::ivector<int> >' requested here
Matcher<T>::Matcher(T value) { *this = Eq(value); }
                                       ^
/Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:28:37: note: in instantiation of member function 'testing::Matcher<etl::ivector<int> &>::Matcher' requested here
    EXPECT_CALL(mock, get_something(v))
                                    ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:927:19: note: copy constructor of 'EqMatcher<etl::ivector<int> >' is implicitly deleted because base class 'ComparisonBase<EqMatcher<etl::ivector<int> >, etl::ivector<int>, testing::internal::AnyEq>' has a deleted copy constructor
class EqMatcher : public ComparisonBase<EqMatcher<Rhs>, Rhs, AnyEq> {
                  ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:922:7: note: copy constructor of 'ComparisonBase<testing::internal::EqMatcher<etl::ivector<int> >, etl::ivector<int>, testing::internal::AnyEq>' is implicitly deleted because field 'rhs_' has an inaccessible copy constructor
  Rhs rhs_;
      ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:895:45: error: field of type 'etl::ivector<int>' has private copy constructor
  explicit ComparisonBase(const Rhs& rhs) : rhs_(rhs) {}
                                            ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:930:9: note: in instantiation of member function 'testing::internal::ComparisonBase<testing::internal::EqMatcher<etl::ivector<int> >, etl::ivector<int>, testing::internal::AnyEq>::ComparisonBase' requested here
      : ComparisonBase<EqMatcher<Rhs>, Rhs, AnyEq>(rhs) { }
        ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:3743:48: note: in instantiation of member function 'testing::internal::EqMatcher<etl::ivector<int> >::EqMatcher' requested here
inline internal::EqMatcher<T> Eq(T x) { return internal::EqMatcher<T>(x); }
                                               ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:3748:40: note: in instantiation of function template specialization 'testing::Eq<etl::ivector<int> >' requested here
Matcher<T>::Matcher(T value) { *this = Eq(value); }
                                       ^
/Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:28:37: note: in instantiation of member function 'testing::Matcher<etl::ivector<int> &>::Matcher' requested here
    EXPECT_CALL(mock, get_something(v))
                                    ^
/Users/jeremy/Dropbox/code/fw_test/submodules/etl/src/vector.h:959:5: note: declared private here
    ivector(const ivector&);
    ^
In file included from /Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:6:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock.h:61:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-generated-function-mockers.h:43:
In file included from /Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-spec-builders.h:75:
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:905:37: error: field of type 'etl::ivector<int>' has private copy constructor
    explicit Impl(const Rhs& rhs) : rhs_(rhs) {}
                                    ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:898:28: note: in instantiation of member function 'testing::internal::ComparisonBase<testing::internal::EqMatcher<etl::ivector<int> >, etl::ivector<int>, testing::internal::AnyEq>::Impl<etl::ivector<int> &>::Impl' requested here
    return MakeMatcher(new Impl<Lhs>(rhs_));
                           ^
/Users/jeremy/Dropbox/code/fw_test/submodules/googletest/googlemock/include/gmock/gmock-matchers.h:3748:40: note: in instantiation of function template specialization 'testing::internal::ComparisonBase<testing::internal::EqMatcher<etl::ivector<int> >, etl::ivector<int>, testing::internal::AnyEq>::operator Matcher<etl::ivector<int> &>' requested here
Matcher<T>::Matcher(T value) { *this = Eq(value); }
                                       ^
/Users/jeremy/Dropbox/code/fw_test/test/simple_test.cpp:28:37: note: in instantiation of member function 'testing::Matcher<etl::ivector<int> &>::Matcher' requested here
    EXPECT_CALL(mock, get_something(v))
                                    ^
/Users/jeremy/Dropbox/code/fw_test/submodules/etl/src/vector.h:959:5: note: declared private here
    ivector(const ivector&);
    ^
4 errors generated.
```

Is there any way to get this going? I suppose in the meantime I will just have to only use pass-by-pointer.

jeremyherbert, 2017-11-09:
I should also note, if one uses a proper fixed-size vector object in the mock class, it works correctly:

```
#include "gtest.h"
#include "gmock.h"

#include "etl/src/vector.h"

class GetSomething {
public:
    virtual int get_something(etl::ivector<int> &input) { return input[0]; };
};

class GetSomethingMock : public GetSomething {
public:
    MOCK_METHOD1(get_something, int(etl::vector<int, 5> &input));
};

using testing::Return;

TEST(SimpleTest, CallWithVector) {
    GetSomethingMock mock;

    etl::vector<int, 5> v;
    v.push_back(10);

    EXPECT_CALL(mock, get_something(v))
                .Times(1)
                .WillRepeatedly(Return(12));

    EXPECT_EQ(mock.get_something(v), 12);
}
```

jwellbelove, 2019-10-12:
It appears that Google mock is trying to copy construct the ivector, which is not allowed.

closed 2019-10-12 by jwellbelove