ETLCPP/etl #52: etl don't build under platformio for Arduino

ederle, 2017-11-14:
I have upgraded from version **9.6.1** to **10.3.2** today and i've got some compiler errors:

In **etl/src/platform.h**,  error on #include "etl_profile.h",
i replace it by  #include "etl_arduino.h".

In **etl/src/c/ecl_timer.h**, error on #include "ecl_user.h"

Is this a problem or did I make a mistake?
Thanks for help.

Cheers

jwellbelove, 2017-12-07:
I've changed the way that the ETL setup macros are defined. Previously, some were set in the project settings and others were set in platform.h which checked the compiler and tried to set all of the macro values. This started to become difficult to get right for all of the different compilers and versions. Now it is up to the user to provide an etl_profile.h that defines them. There are several predefined headers that can be included from there in the src/profiles directory or use them to define your own.

See
https://www.etlcpp.com/macros.html

https://www.etlcpp.com/setup.html

https://www.etlcpp.com/ecl_timer.html
You can find examples of etl_profile.h and ecl_user.h in the /test directory

John.

jwellbelove, 2017-12-07:
Sorry about the delay in replying. My replies through my email client were not being recognised by Github for a while.

ederle, 2017-12-07:
No problem ;-)

Thank you for answer.

Cheers.

SurfingDude, 2018-02-27:
This is an excellent library, it provides an elegant alternative to hand-coded linked lists on the ATSAMD21 Arduinos. However, the web site https://www.etlcpp.com/home.html would be improved if it described the ecl_user.h file in the setup documentation.

puzrin, 2019-09-22:
You should add `-I include` to build_flags, and place `ecl_user.h` into appropriate folder.

**platform.ini**:

```
[platformio]
default_envs = native

[env:native]
platform = native
build_flags =
  -D PROFILE_GCC_GENERIC
  -I include
lib_deps =
  ; Set proper url here, or comment and place etl to `/lib` folder of PIO project.
  etl=https://github.com/puzrin/etl/archive/master.zip
  ;etl@^14.30.0
```

**include/etl_profile.h**:

```c
#include "etl/profiles/etl_profile.h"
```

**include/ecl_user.h**:

```c
#include "etl/../../test/ecl_user.h"
```

Probably, this can be simplified (to work without `-I include`), but i did not investigated details. Without that option, configs a found when included from `.h`, but not found when included from `.c`.

https://github.com/puzrin/etl_check - here is my temporary demo repo to test & fix pio integration.

@jwellbelove 

I agree with note about undocumented `ecl_user.h`. This should be fixed. in "Setup" section of website. Probably, there also should be more verbose error with advice where to take file from.

puzrin, 2019-09-26:
Tested 14.31.2.

I think this issue can be closed as solved.

closed 2020-03-28 by jwellbelove