ETLCPP/etl #54: [enhancement] : running tests on target platform (i.e. up to C++03 compatible)

NeimadG, 2017-11-20:
Hi
As a paranoid developer, I need to compile and run unit tests on my target to ensure compatibility with OS, tool suite, etc. . Unfortunately, tests are not C++03 compatible. So I have to modify all required tests to do so.
It would be great to update tests so that they can be run on target as well.
Regards

jwellbelove, 2017-12-05:
That's odd. I replied via the Github email notification, but it doesn't seem to have appeared here.
I'll repeat the message again.

Hi,

The issue with changing all of the tests to be C++03 compatible is that the reverse engineered C++11 features could no longer be compared directly against the STL originals. To change this I would have to log all of the output from the C++11 features and hardcode the results for the unit test checks.

I can see where you're coming from and why it could be useful, but at the moment I don't have the spare time to do what is likely to be a large amount of work. (I'm currently working on a programming blog and a "C++ templates in embedded" ebook.)
(I'll also have to go back to do some more contracting work soon, before the money runs out!)

If you're up for continuing to do the necessary changes I would be happy to merge them back to the Git repository when you're finished. It does look like it could be a lot of work though.

Regards,
John.

NeimadG, 2017-12-07:
Hi,
Of course I understand that porting all tests is a very big piece of work. Unfortunately, I can't afford at work to port all tests, and I simply can't do it at home as I don't have enough hardware. For the project I work on at work, I only use a very small subset of the library (array/vector/set/map/list), and I ported only the corresponding tests. Furthermore I'm still on a 9.X version (sorry I forgot the exact one), and I'm not planning to update to the latest one in the short term.
For all those reasons, the only contribution I can afford today is posting issues or enhancement requests as I come up with. Sorry about that.
Although thanks for sharing your work with the community !
Regards
Damien

jwellbelove, 2017-12-07:
No worries. Hopefully when I've got this lastest work out of the way I'll be able to look at the issue. One other issue that someone has recently brought up is that clear() for some containers (set/map and some others) is slower that it really needs to be when the contained type has a trivial destructor. This is not too hard a task to do, but I'm desperately trying not to get diverted by 'interesting' issues!

Have you also considered the 'flat' map/set containers? They may possibly be performant enough for your application whilst having considerably less code and allocation overhead.

Are your current unit test changes in a suitable form for a pull request? I'd be happy to look at them to see if they can be merged into the current release.

NeimadG, 2017-12-16:
I have not spent time to check compatibility of my tests modifications with latest version yet. I'm not sure I can do it in the short term. 
In issue #36 I requested support for non-copyable objects. As I needed a quick solution, I implemented macros as described in issue. So tests use those macros. I noticed that you added this support (with the use of as many constructors and others functions as required for parameter number), but I have not switched to your method yet.
I also set an automatic test code generation to test operators less/greater (no need to choose), based on tags in existing test sources, on sed for text substitution and makefile (linux).
I did my best to preserve existing philosophy, but I'm not sure compatibility is still ensured...
If you wish I can request a pull, but I'm not sure you have spare time to review my commits...

I did not consider 'flat' containers. I can have a look at them ...

I won't be able to push anything till January, so don't be surprised if you don't see anything coming from me !
 PS. I'm not sure, but you refer to  "someone" who recently suggested 'fast' clear() for some containers... If you think about #55, so "someone" is me ;)

jwellbelove, 2017-12-16:
Ah yes, that 'someone' was you.

Re: Non-copyable objects. 
Yes, I've implemented C++03 compatible C++11 'emplace' style functions for some of the containers (list, forward_list, vector, stack, queue, priority_queue) that accept up to 4 parameters. I'd like to add more containers in the future, as well as variadic versions for those that have compilers that support C++11.

I may start a 'ToDo' page on the website and Slack to see if I can encourage any others to take up some of the enhancements and new features that are in the pipeline.

If you want to do a pull request at some time in the future, can I request that you do it on a new feature branch, as I am unlikely to want to merely 'merge' as-is. I'll read through the changes and see what ideas I can replicate in the main branch. Either that or mail the changed files to me.

closed 2018-02-25 by jwellbelove