ETLCPP/etl #55: [enhancement] : set/map/??? fast clear

NeimadG, 2017-12-05:
Hi
I noticed on etl::set than calling clear() gently destroys all items one after another. For POD types, this is not necessary, a simple reset would be enough, and faster.
It applies to set at least, maybe to other containers?
Regards
Damien

jwellbelove, 2017-12-05:
Yes that's true. By default the containers delete the objects one by one as the underlying pool cannot easily tell which blocks have been used. It has a list of blocks that are *not* used and in theory it could use this to discover the used blocks and delete them. So far I haven't found a zero overhead and fast way of doing this. That said, the container will know what type it contains and could possibly invoke a pool 'reset' it the type is POD.

jwellbelove, 2017-12-07:
Actually it's not just for POD types. Any type with a trivial destructor would be eligible for 'fast' clear.

NeimadG, 2017-12-07:
Hi
I wrote the evolution for list/set/map/vector. I used enable_if with has_trivial_destructor to make the distinction for complete or fast clearing (for map, key AND value need to have trivial destructor to be eligible to 'fast' clearing.
If you're interested in, I can push that (If I found how to do so since I've never pushed anything to github ...)

jwellbelove, 2017-12-09:
If you push then it would be better to do it on a new feature branch rather than the master.
If you don't want to push then you could attach the files to a message and I'll review and merge them here to a local feature branch.

NeimadG, 2019-11-05:
Hi,
This issue appears to be fixed. Isn't it ?

closed 2020-03-28 by jwellbelove