ETLCPP/etl #58: ETL_DEBUG should not be defined in release builds

apmorton, 2018-01-07:
[`platform.h`](https://github.com/ETLCPP/etl/blob/fd3cb9d52d83e7dac3dcd467d4ee28c5429656d0/src/platform.h#L37) defines `ETL_DEBUG` to `1` or `0`
[all the places](https://github.com/ETLCPP/etl/search?utf8=%E2%9C%93&q=ETL_DEBUG+&type=) that check for `ETL_DEBUG` use `#if defined(ETL_DEBUG)`

The result is that assertions intended for debug builds only are always enabled.
Not sure what the original intent was, but its probably just easiest to change `platform.h` to something like the following.
```C
#if defined(_DEBUG) || defined(DEBUG)
  #define ETL_DEBUG
#endif
```

jwellbelove, 2018-01-07:
Thanks, I shall correct that immediately.

jwellbelove, 2018-01-07:
Done.

apmorton, 2018-01-07:
cheers - thanks for the great library! just getting started with it, but it has already been amazingly useful thus far.

closed 2018-01-07 by apmorton