ETLCPP/etl #65: Suggestion to modify location of header files

DrChr, 2018-03-18:
Issue to track work on a modified directory structure to e.g. reduce name collisions for header files. Previously discussed in #64, but better off as a separate issue.

Background: ETL currently has .h and .cpp files in etl/src/ and the current documentation says:

    A path to the ETL's src directory must either be defined or full
    paths to the ETL includes must be used.

If your project e.g. include other 3rd party libraries, containing e.g. queue.h, there's a risk of name collision with e.g. etl/src/queue.h.

Possible solution: Move ETL headers to:

    etl/include/etl

and let the project add etl/include/ to it's include paths. The source code would then include "queue.h" by doing:

    #include "etl/queue.h"

jwellbelove, 2018-03-18:
I've created a tag 10.21.2 as the last Arduino compatible version, due to Arduino's very specific directory layout specifications. The new layout will start from 11.0.0

jwellbelove, 2018-03-20:
The new layout has been pushed to master.

DrChr, 2018-03-20:
Thanks. I gave it a very quick look/test and it seems to work. :-)

Some notes follows now. I cloned master to `<repo-root>/lib/etl-cpp/` and added `<repo-root>/lib/etl-cpp/include/` to the include paths. Then I created a source file that used

    #include "etl/queue.h"

I got an compile error for my "etl_profile.h", as it since before said

    #include "profiles/gcc_generic.h"

but to make it work I just modified it to

    #include "etl/profiles/gcc_generic.h"

So existing "etl_profile.h":s will likely need a quick change, once per project.

As a minor issue, I looked in e.g. 'queue.h', which includes eg:

    #include "container.h"

I'm wondering if e.g. the include directive should instead read e.g.
 
    #include "etl/container.h"

This would make it clearer that it's an internal etl header that's used.
But I don't _think_ it's needed though, I just don't remember what the standard says. I did check GCC for which it's not needed. I.e., according to as https://gcc.gnu.org/onlinedocs/cpp/Search-Path.html, GCC will search for "container.h" first in the folder that contains queue.h.

jwellbelove, 2018-03-20:
Actually,n container.h is not an internal header.

DrChr, 2018-03-20:
You're right, "internal" is incorrect. What I meant was that it's just an ETL header. 
Note: I've updated my comment text above to avoid confusing the issue.

closed 2018-05-30 by jwellbelove