ETLCPP/etl #67: Add initializer-list constructor to `etl::set`

ArekSredzki, 2018-03-20:
Currently initializing an `etl::set` instance with a few values rather long-form. Having access to an initializer-list constructor would be very useful.

jwellbelove, 2018-03-20:
At the moment the ETL focuses on C++03 compatibility. There are already a few C++11 additions that can be enabled using a macro definition in the profile header and this may be another that could be added.
If I do add this functionality then I would like to make it applicable to all of the containers in the ETL. That's not to say it could not be done in stages rather than in one bit hit. I'll add it to the 'Jobs' list on Slack.

jwellbelove, 2018-03-22:
I've made a start and modified about half a dozen containers so far.
It seems to be an easy update to do, so I should have all of them done by the end of the weekend.

jwellbelove, 2018-03-25:
Just pushed the std::initializer_list constructor changes.

ArekSredzki, 2018-03-26:
Awesome, thank you so much!

closed 2018-03-26 by ArekSredzki