ETLCPP/etl #69: Unexpected(?) behavior in `etl::optional`

annius, 2018-04-10:
I was curious as to what the expected behavior for this snippet of code is:
```
#include "optional.h"
#include <iostream>

int main() {
  etl::optional<double> a;
  const double b = 0;
  const bool is_equal = b == a;
  std::cout << is_equal << std::endl;
}
```

It builds correctly with `g++ main.cpp -std=c++11` and upon executing it prints 0. Shouldn't it either
a. throw a compile-time error because `b` and `a` are different types, or at least
b. a run-time error because `a` has not been initialized? For instance, http://en.cppreference.com/w/cpp/experimental/optional suggests that it should throw a `bad_optional_access` exception.

jwellbelove, 2018-04-11:
For the comparison operators the following paragraph applies.

_Compares opt with a value. The values are compared (using the corresponding operator of T) only if opt contains a value. **Otherwise, opt is considered less than value**. If the corresponding comparison expression between *opt and value is not well-formed, or if its result is not convertible to bool, the behavior is undefined._

Therefore, comparing an uninitialised optional with a value should return _false_.

annius, 2018-04-11:
Good to know, thanks!

closed 2018-04-11 by annius