ETLCPP/etl #73: Modification / unregistration inactive timers

mchodzikiewicz, 2018-05-09:
There is no way to access message and callback single shot timer if it is inactive, specifically, you cannot change its period or unregister it.

See example below:

```
etl::message_timer<3> timer;
timer.enable(true);
etl::timer::id::type id1 = timer.register_timer(message,router,period,etl::timer::mode::SINGLE_SHOT);
timer.start(id1);

//Timer elapses due to some ticks elsewhere
timer.tick(period);

//this one fails because timer is inactive
bool result = timer.unregister_timer(id1);

//this also fails for the same reasons
result = timer.set_period(id1,new_period);

//this will launch the timer, but with old period value
timer.start(id1);
```

jwellbelove, 2018-05-09:
I noticed that I automatically start timers after the period or mode is set. I'm sure it seemed a good idea at the time, but I can't think of one at the moment. Do you think that timers should automatically stop after having a period or mode changed? Or only restart if they were currently active?

mchodzikiewicz, 2018-05-10:
According to least astonishment principle, I would either expect them to restart if they were active.

closed 2018-05-10 by jwellbelove