ETLCPP/etl #77: etl::observable's method should return bool

mchodzikiewicz, 2018-06-04:
I have encountered this problem, that I cannot reach information, if removing observer from list was successful
This is a problem, because in my case I am unable to maintain information about observers registered in each observable and I have to try each one and execute additional actions.
Current code:

```
        void removeBindingsFromInput(int input){
            unsigned int number_of_obs = Inputs[input]->number_of_observers();
            for(unsigned int i=0;i<Hub.size();i++){
                Inputs[input]->remove_observer(*Hub.at(i).Binding);
                if(Inputs[input]->number_of_observers() < number_of_obs){
                    Hub.at(i).Binding->unbind();
                    Hub.at(i).Val->setValue(-1);
                    return;
                }
            }
        }
```

It would be even more problematic if I would need to remove more than one observer with related additional actions

Desired code:
```
        void removeBindingsFromInput(int input){
            for(unsigned int i=0;i<Hub.size();i++){
                if(Inputs[input]->remove_observer(*Hub.at(i).Binding){
                    Hub.at(i).Binding->unbind();
                    Hub.at(i).Val->setValue(-1);
                }
            }
        }
```

mchodzikiewicz, 2018-06-04:
I have adjusted remove_observer locally and I can make PR when we (me and Adam) will solve some CMake related problems in our project config

jwellbelove, 2018-07-12:
Done

closed 2018-07-12 by jwellbelove