ETLCPP/etl #82: Debugging support

annius, 2018-07-20:
Is there any scope of inspecting elements of an ETL container e.g. `etl::vector` in a debugger? For instance, in CLion with an attached process I see something like

[image]

jwellbelove, 2018-07-20:
It looks like CLion uses GDB. The visualisers are written in Python, so custom ones can be created. I've rarely used GDB or Python, so I can't really be a lot of help here. There does seem to be a way of casting a pointer to be seen as an array of a different type.
This may help...
https://stackoverflow.com/questions/40327089/in-clions-debugger-how-do-i-show-the-entire-contents-of-an-int-array

annius, 2018-07-20:
That trick works! Thanks a lot for the information.

closed 2018-07-20 by annius

flaviut, 2022-11-13:
```
# Copyright (C) 2013-2022 Free Software Foundation, Inc.

# This program is free software; you can redistribute it and/or modify
# it under the terms of the GNU General Public License as published by
# the Free Software Foundation; either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU General Public License for more details.
#
# You should have received a copy of the GNU General Public License
# along with this program.  If not, see <http://www.gnu.org/licenses/>.

import gdb

class VectorPrinter:
    class Iterator:
        def __init__(self, start, finish):
            self.item = start
            self.finish = finish
            self.count = 0

        def __iter__(self):
            return self

        def __next__(self):
            count = self.count
            self.count = self.count + 1
            if self.item == self.finish:
                raise StopIteration
            elt = self.item.dereference()
            self.item = self.item + 1
            return f'[{count}]', elt

    def __init__(self, val):
        self.val = val

    def children(self):
        return self.Iterator(self.val['p_buffer'], self.val['p_end'])

    def to_string(self):
        start = self.val['p_buffer']
        finish = self.val['p_end']
        end = self.val['p_buffer'] + self.val['CAPACITY']
        return f'len={finish - start}, cap={end - start}'

    def display_hint(self):
        return 'array'

class DequePrinter:
    class Iterator:
        def __init__(self, start, finish, buffer, capacity):
            self.item = start
            self.finish = finish
            self.buf = buffer
            self.cap = capacity
            self.count = 0

        def __iter__(self):
            return self

        def __next__(self):
            count = self.count
            self.count = self.count + 1
            if self.item == self.finish:
                raise StopIteration
            elt = self.item.dereference()
            self.item = self.item + 1
            if self.item == (self.buf + self.cap):
                self.item = self.buf
            return f'[{count}]', elt

    def __init__(self, val):
        self.val = val

    def children(self):
        return self.Iterator(self.val['p_buffer'] + self.val['_begin']['index'],
                             self.val['p_buffer'] + self.val['_end']['index'],
                             self.val['p_buffer'], self.val['BUFFER_SIZE'])

    def to_string(self):
        size = self.val['current_size']
        cap = self.val['BUFFER_SIZE']
        return f'len={size}, cap={cap}'

def build_pretty_printer():
    pp = gdb.printing.RegexpCollectionPrettyPrinter("embedded-template-library")
    pp.add_printer('etl::vector<T>', 'etl::i?vector<(.*)>', VectorPrinter)
    pp.add_printer('etl::deque<T>', 'etl::i?deque<(.*)>', DequePrinter)
    return pp

gdb.printing.register_pretty_printer(
    gdb.current_objfile(),
    build_pretty_printer())
```

I've written a couple pretty-printers based on the [the `std::` printers](https://github.com/gcc-mirror/gcc/blob/91c632c88994dca583bcd94e39cd3eba1506ecfe/libstdc%2B%2B-v3/python/libstdcxx/v6/printers.py).

This works quite well, both in CLion:

[image]

and in standard GDB:

```
(gdb) source scripts/etl_pp.py
Registering global embedded-template-library pretty-printer ...
(gdb) p this->uart->rxBuffer
$3 = {<etl::deque<unsigned char, 255>> = len=25, cap=256 = {[0] = 126 '~', [1] = 0 '\000', [2] = 21 '\025', [3] = 144 '\220', [4] = 0 '\000', [5] = 0 '\000', [6] = 0 '\000', [7] = 0 '\000', [8] = 0 '\000', [9] = 0 '\000', [10] = 0 '\000', [11] = 0 '\000', [12] = 255 '\377', [13] = 254 '\376', [14] = 1 '\001', [15] = 222 '\336', [16] = 0 '\000', [17] = 1 '\001', [18] = 164 '\244', [19] = 116 't', [20] = 121 'y', [21] = 112 'p', [22] = 101 'e', [23] = 2 '\002', [24] = 42 '*'}, <No data fields>}
```

the `source scripts/etl_pp.py` can be added to your `.gdb_init` file to have it automatically loaded each time.