ETLCPP/etl #83: string insert differs from std::string when string is inserted into itself

benjamind2330, 2018-07-20:
I am not sure if the behaviour of this is defined this way to avoid a copy of the input string but this instance differs from the std::string in a bad way:

    {
        std::string sut = "abcdefgh";
        sut.insert(3, sut);
        std::cout << sut << "\n";
    }
    {
        etl::string<50> sut = "abcdefgh";
        sut.insert(3, sut);
        std::cout << sut.data() << "\n";
    }

Results in the output:

    abcabcdefghdefgh
    abcabcabcabdefgh

Which shows that the behaviour of the string is incorrect for this example.

benjamind2330, 2018-07-20:
I believe this fix to this is simple, but unfortunate. You need to copy the string passed before inserting. This will occur also for input iterators that reference the string you are inserting to, so you can't even only copy if the string references yourself (self reference check).

jwellbelove, 2018-07-20:
I haven't tried it out yet, but I think the issue may be able to be fixed without copying (which may be difficult to do in the ETL without using the heap).

It should be possible to detect that the input iterator is within the range of the current string. The input iterators could then be modified as characters are inserted, compensating for the changing string contents. As I said, I've not tried to code this yet so I don't know how convoluted the solution may turn out to be, but I think its possible to do.

jwellbelove, 2018-07-21:
I been looking at this problem this morning and although my initial idea would work in some instances, it would not work for everything. The issue occurs when the inserted size would cause the string to truncate, where part or all of the insert text gets pushed outside of the truncated range. The only other ways in which this can be accomplished would be to...
1. Use alloca() to allocate temporary string storage on the stack.
2. Use variable length arrays (VLA)
3. Create a virtual function that is implemented in the top level class that implements 'insert' (This is the only class that can know the string size at compile time).
4. Copy the string externally and insert from that. 'insert' from 'self' would be 'undefined' behaviour'.

1. alloca() is non-standard and may not be supported on some platforms.
2. VLAs are a C extension and are not supported by C++.
3. This will make the string classes virtual by default. All of the ETL containers are currently non-virtual by default, which I am loath to change.
4. This would have the least impact of the ETL (i.e. none), but requires the application code to be able to make a copy.

At the moment I have a preference for 4.
I would maybe consider 3. but the functionality would have to be enabled via a compile time macro flag (ETL_ALLOW_STRING_SELF_INSERT), to keep the current default behaviour.

What are your thoughts?

benjamind2330, 2018-07-23:
Yeah, I don't think 3 is a particularly good solution. You cannot cast or use RTTI either. 

So, the capacity and string type (char w_char) is known as part of the insert function. (CAPACITY). It is checked to clip the string if the insert would overrun it. For me the solution would be more like:

1. Check for self insert (iterator reference inside buffer reference, string reference the same). If this is possible it will make things easier.
2. If so create copy of the input using the size CAPACITY. We know that since it is us, we cannot be greater than size CAPACITY. If the user gave out of bound iterators or something, it is undefined behaviour anyway. I guess this part is tricky because you need to be able to build a string with the correct size and type. Is this possible?
3. Copy the string into our recent build.
4. Do the insert. 

This will also occur with insert of char*. Like this:

    const char* title = "Everything was beautiful, and nothing hurt";
    str01 = title;
    str01.insert(0, str01.c_str() + str01.size() - 4, 4);
    EXPECT_TRUE(str01 == "hurtEverything was beautiful, and nothing hurt");

I found this by running the unit tests from the GCC implementation of std::string over this. It will also happen in replace etc. Pretty much anywhere that modifies the string and accepts, String, iterators or char* input.

jwellbelove, 2018-07-23:
The issue is that the capacity is only a compile time constant for the top level class 'etl::string<N>'. The code that that implements 'insert' is in the base class that only knows the capacity at _runtime_. This would mean that insert would have to allocate temporary storage based on a runtime value. This is only possible by either allocating from the stack, via alloca() or VLAs or allocating from the heap, none of which are viable solutions. To create compile time temporary storage would require to functionality to be moved to etl::string<N>, requiring a virtual 'insert' function. 
I can't see an easy solution to this and may have to be a situation where the ETL functionality deviates from that of the STL.

jwellbelove, 2018-07-23:
Note that self insert does work as long as the source range is before the insert position .

closed 2019-02-11 by jwellbelove