ETLCPP/etl #91: deque.h makes 3 types of errors

uchan-nos, 2018-08-26:
Hello, I'm trying to include deque.h in my hobby baremetal program.
I found 3 types of errors while compiling it.

## Test environment

clang version:

```
$ clang++ --version
clang version 6.0.0-1ubuntu2 (tags/RELEASE_600/final)
Target: x86_64-pc-linux-gnu
Thread model: posix
InstalledDir: /usr/bin
```

test file

```
$ cat b.cpp
#include <deque.h>
```

etl_profile.h is copied from etl/include/etl/profiles/cpp14_no_stl.h.

## Type 1: namespace bug

`etlstd` seems the namespace for unit testing. I think `ETLSTD` macro should be defined as `etlstd` for testing and `std` for normal use, and use the macro instead of hard-coded `etlstd`.

```
$ clang++ -ffreestanding -std=c++17  -I$ETL_INCDIR -I. -c b.cpp
In file included from b.cpp:1:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/deque.h:39:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/algorithm.h:37:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/alternate/algorithm.h:41:
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/alternate/functional.h:75:28: error: use of undeclared identifier 'etlstd'
  class binder1st : public etlstd::unary_function<typename TFunction::second_argument_type, typename TFunction::result_type> 
                           ^
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/alternate/functional.h:109:28: error: use of undeclared identifier 'etlstd'
  class binder2nd : public etlstd::unary_function<typename TFunction::first_argument_type, typename TFunction::result_type> 
                           ^
```

## Type 2: missing include

alternate/algorithm.h should include alternate/utility.h, which includes `pair` definition.

```
In file included from b.cpp:1:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/deque.h:39:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/algorithm.h:37:
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/alternate/algorithm.h:265:8: error: no template named 'pair' in namespace 'std'
  std::pair<TIterator, TIterator> equal_range(TIterator first, TIterator last, const TValue& value, TCompare compare)
  ~~~~~^
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/stl/alternate/algorithm.h:267:17: error: no member named 'make_pair' in namespace 'std'
    return std::make_pair(ETLSTD::lower_bound(first, last, value, compare),
           ~~~~~^
```

## Type 3: undefined functions

algorithm.h uses functions which are undefined in the etl repository.

```
In file included from b.cpp:1:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/deque.h:44:
In file included from /home/uchan/workspace/github.com/ETLCPP/etl/include/etl/array.h:47:
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/algorithm.h:517:17: error: no member named 'find_if' in namespace 'std'
    return std::find_if(begin, end, predicate) != end;
           ~~~~~^
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/algorithm.h:531:17: error: no member named 'find_if' in namespace 'std'
    return std::find_if(begin, end, predicate) == end;
           ~~~~~^
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/algorithm.h:553:23: error: no member named 'find' in namespace 'std'
        if (i == std::find(begin1, i, *i))
                 ~~~~~^
/home/uchan/workspace/github.com/ETLCPP/etl/include/etl/algorithm.h:555:27: error: no member named 'count' in namespace 'std'
          size_t n = std::count(begin2, end2, *i);
                     ~~~~~^
```

jwellbelove, 2018-08-26:
Thanks for info. I'll get on to it as soon as I can. (no later than tomorrow)

jwellbelove, 2018-08-27:
Hopefully should all be fixed.
Tested with a mini Keil project using ARM compiler v5

The atomics in the ETL will not currently compile with v6 (with No STL option), but I should have that sorted soon once I investigate the LLVM atomic intrinsics.

Let me know if you find any other issues.

closed 2019-02-11 by jwellbelove