ETLCPP/etl #92: Keil ARM v5 Compiler

JohannesWoeber, 2018-08-31:
Hi,
when using the armv5.h profile and compiling the library the following line 

```
#if !defined(ETL_STLPORT)
const std::nullptr_t nullptr = {};
#endif
```
causes the
.\libs\ETL\include\etl/nullptr.h(77): error:  #40: expected an identifier
error

Compiler:
*** Using Compiler 'V5.06 update 5 (build 528)', folder: 'C:\Keil\ARM\ARMCC\Bin'

JohannesWoeber, 2018-08-31:
Hi,
Thing is it compiles fine when commenting out the nullptr redefinition. The keil arm v5 compiler seems to have nullptr support (http://www.keil.com/support/man/docs/armcc/armcc_chr1407404265784.htm). It doesnt compile with the ETL_NO_NULLPTR_SUPPORT 0  define. There seems to be no std::nullptr_t type definition.

I fixed it ghetto style by using to ETL_COMPILER_KEIL define to not redefine the nullptr while still having the std::nullptr_t (see Feature/keil_uvision5_nullptr_fix ).

+ some small fixes for annoying keil specific compiler warnings

jwellbelove, 2018-09-01:
I've pushed a new version to master.
I've renamed ETL_COMPILER_ARM and ETL_COMPILER_LLVM to ETL_COMPILER_ARM5 and ETL_COMPILER_ARM6 respectively.
I've also updated a couple of other related files.

JohannesWoeber, 2018-09-02:
Hi thanks,
it still threw the error message for me though.

`#if !defined(ETL_STLPORT) && !defined(ETL_COMPILER_ARM5)
`
instead of 
`#if !defined(ETL_STLPORT) || !defined(ETL_COMPILER_ARM5)
`
worked for me though

jwellbelove, 2018-09-02:
I've pushed a new version to master that compiles my Keil project correctly for ARM5 (with or without --cpp11) and ARM6 compilers

JohannesWoeber, 2018-09-03:
Hi, 
works great. Thanks.

closed 2018-09-03 by JohannesWoeber