ETLCPP/etl #93: map.h on release 11.15.0 does not compile with clang-6

annius, 2018-09-04:
Created a file called `main.cpp` with the following code:
```
#include "include/etl/map.h"

int main() {
  return 0;
}
```

Compiling with clang-6 results in 
```
In file included from main.cpp:1:
./include/etl/map.h:484:16: error: use of non-static data member 'compare' of 'imap' from nested type 'key_comp'
        return compare(key1, key2);
               ^~~~~~~
./include/etl/map.h:495:16: error: use of non-static data member 'compare' of 'imap' from nested type 'value_comp'
        return compare(value1.first, value2.first);
               ^~~~~~~
2 errors generated.
```

annius, 2018-09-04:
Also occurs in `set.h`, `multimap.h`,and `multiset.h`. The solution is to change public references to `compare` to `this->compare`.

jwellbelove, 2018-09-04:
Actually, the implementations of key_comp and value_comp are not currently compliant with the specification of std::map etc. They are member structures, when in fact they should be member function.

I will re-implement them.