ETLCPP/etl #101: Non-const overloads for emplace / emplace_back

jpan127, 2018-10-09:
There are certain cases where the template type of a container has constructors requiring non-const references.

```c++
class Dummy {
public:
    Dummy(Bar &bar);
};
```

Since the `emplace_back` method for `etl::vector` only accepts const reference parameters, it fails for classes like `Dummy`.

```c++
    template <typename T1>
    void emplace_back(const T1& value1)
    {
#if defined(ETL_CHECK_PUSH_POP)
      ETL_ASSERT(size() != CAPACITY, ETL_ERROR(vector_full));
#endif
      ::new (p_end) T(value1);
      ++p_end;
      ETL_INCREMENT_DEBUG_COUNT
    }
```

If there is an overload for non-const parameters, `emplace_back` will work for `Dummy`.

```c++
    template <typename T1>
    void emplace_back(T1& value1)
    {
#if defined(ETL_CHECK_PUSH_POP)
      ETL_ASSERT(size() != CAPACITY, ETL_ERROR(vector_full));
#endif
      ::new (p_end) T(value1);
      ++p_end;
      ETL_INCREMENT_DEBUG_COUNT
    }
```

If this makes sense for the ETL team, I can gladly make the appropriate changes across the repo.  I would like to get some consensus first, because it can be tedious 😄 .

jwellbelove, 2018-10-09:
The normal C++11 emplace methods use rv
alue references, but the C++03 versions only emulate this functionality for up to 4 parameters. I'll need to check, but it may be possible to remove the 'const' if the compiler substitutes the parameter type in full (including its constness).

jwellbelove, 2018-10-09:
I've only done a quick test but it looks like the answer is to remove 'const' from all of the parameter s for the C++03 style emplace functions.

dev-0x7C6, 2018-10-09:
This is probably slightly off-topic, but this bother me. You mention rvalue references in emplace method signature, but you mean universal reference yes? To my knowledge T&& (where T is used in template context) is syntax feature and allow to forward any kid of value category and generate strict function/method signature.

apmorton, 2018-10-10:
Not strictly related, but on the topic of the emplace functionality in ETL, would you be open to a PR to conditionally use a full variadic template version of emplace if compiling with a version of c++ that supports it?

It's not strictly only an issue for this kind of feature, but in general the more of these things that get implemented as compile time switches the more complex a full test suite becomes.

jwellbelove, 2018-10-16:
I can accept the C++11 variadic versions, but the C++03 non-const versions will break code that use literals as parameters.

closed 2019-02-11 by jwellbelove