ETLCPP/etl #103: Potential dereference of nullptr

annius, 2018-11-09:
Ran `clang-check` on an example with a set and got an error at https://github.com/ETLCPP/etl/blob/master/include/etl/set.h#L1575.

It seems like `critical_parent_node->children[critical_parent_node->dir]` could potentially attempt to dereference a `nullptr`. This is part of an else block; the if condition checks for
```
if (critical_parent_node == nullptr && critical_node == root_node)
```
So it's possible to reach the else block even if `critical_parent_node` is nullptr, if `critical_node` is not equal to `root_node`.

Is this every possible?

annius, 2018-11-09:
There's actually more than one such warning on this file. Happy to furnish the full report if you think this is actually an issue.

jwellbelove, 2018-11-09:
I'll have to take some time to analyse the code in question. map/set was added by another contributor, so I'm not overly familiar with the underlying algorithm.

annius, 2018-11-09:
OK, if you want to replicate the warnings, here's how to do it.

1. Create a file (`clang_check_example.cpp`) with the following contents:
```
#include "etl/vector.h"
#include "etl/map.h"
#include "etl/set.h"

int main() {
    etl::vector<int32_t, 5> my_vec;
    etl::map<int32_t, float, 5> my_map;
    etl::set<float, 5> my_set;
    return 0;
}
```
2. Create a file named `compile_commands.json` with the following contents:

```
[
  {
    "command": "clang++ clang_check_example.cpp -o clang_check_example -std=c++11 -Werror -Wall -Wextra -Wpedantic -Wfloat-equal -I<path-to-etl>/include -I<path-to-etl>/include/etl/profiles  -DPROFILE_CPP11_GENERIC -g",
    "file": "clang_check_example.cpp",
    "directory": "<path-to-source-folder>"
  }
]
```
(please note `<path-to-etl>` and `<path-to-source-folder>` in the above)

3. Compile with
```
clang++ clang_check_example.cpp -o clang_check_example -std=c++11 -Werror -Wall -Wextra -Wpedantic -Wfloat-equal -I<path-to-etl>/include -I<path-to-etl>/include/etl/profiles  -DPROFILE_CPP11_GENERIC -g
```

4. Run `clang-check` with
```
clang-check -analyze                                                                                  \
                    -extra-arg -Xclang -extra-arg -analyzer-output=html               \
                    -extra-arg -Xclang -extra-arg -analyzer-opt-analyze-headers \
                    -extra-arg -Qunused-arguments                                              \
                    -p ./ clang_check_example.cpp
```

annius, 2018-11-09:
Oh and this is with clang-6.

jwellbelove, 2020-03-28:
Updated in next release

closed 2020-03-28 by jwellbelove