ETLCPP/etl #105: `etl::deque::push_back` issues on Release 14.5.0

annius, 2018-12-17:
A file with the following contents (say `etl_test.cpp`)
```
#include "etl/deque.h"

int main() {
    etl::deque<int, 10> q{};
    q.push_back(20);
    return 0;
}
```

Fails upon compiling using
```
clang++ etl_test.cpp -o etl_test -std=c++11 -Werror -Wall -Wextra -Wpedantic -Wfloat-equal -Wno-unused-variable -I$ETL_ROOT -I$ETL_ROOT/etl/profiles  -DPROFILE_CPP11_GENERIC -g
```

with the following error message
```
etl_test.cpp:5:7: error: call to member function 'push_back' is ambiguous
    q.push_back(20);
    ~~^~~~~~~~~
/home/prasenjit/dev/etl/include/etl/deque.h:1627:10: note: candidate function
    void push_back(parameter_t item)
         ^
/home/prasenjit/dev/etl/include/etl/deque.h:1641:10: note: candidate function
    void push_back(T&& item)
         ^
1 error generated.
```

The file builds fine with release 14.0.0.

annius, 2018-12-17:
(Using clang 6)

jwellbelove, 2018-12-17:
Fixed v14.5.1

closed 2018-12-17 by jwellbelove