ETLCPP/etl #115: Compile time endianness

shyun3, 2019-02-18:
Instead of determining endianness at run time as is done [currently][etl], could you consider determining it at compile time? For example, Boost uses compiler macros to [determine endianness][predef] and uses these when determining [whether to byte swap][conversion]. The Boost example should cover the mainstream compilers and I know that Keil defines the `__BIG_ENDIAN` macro. I haven't used other embedded toolchains like CCS but if they don't define any endian macros then I suppose you could fall back to the current run time method.

Maybe also consider using intrinsics for byte swapping (also used by [Boost][intrinsic])?

[etl]: https://github.com/ETLCPP/etl/blob/790ba1308a11e77b6fc53900398d13f1c0ad6673/include/etl/endianness.h#L70-L104
[predef]: https://github.com/boostorg/predef/blob/develop/include/boost/predef/other/endian.h
[conversion]: https://github.com/boostorg/endian/blob/develop/include/boost/endian/conversion.hpp
[intrinsic]: https://github.com/boostorg/endian/blob/develop/include/boost/endian/detail/intrinsic.hpp

jwellbelove, 2019-02-18:
I thought about compile time endianness but I didn't have a chance to investigate all of the different compilers at the time. It would be a useful addition. I'll put it on my TODO list.

LanguidSmartass, 2019-02-26:
Hi.
I just wanted to share the knowledge about this topic, since I did some digging into it in a recent past.

I'm kind of sure that it is not doable purely at compile time. Since integers-to-pointers casts (c-style or reinterpret_cast<>) cannot be used in constant expressions and accessing 'uninitialized' members of unions issues a compilation error.

Basically to achieve compile-time endian check there must be a possibility for the developer to lookup the internal memory layout of types, which seems to be impossible.

There are, however, two things that probably can help 
1. Many compilers optimize the runtime union-approach into a single integer/boolean value (0/1);
2. Upcoming C++ 20 std::endian, which is basically a set of enum class values hardcoded for each implementation of the STL.

But the caveats are:
1. Not every compiler may do this optimization, here is a snippet: https://godbolt.org/z/bYl3K1 -- pretty much every compiler inlines the '1' result except for MSVC (if you can check this code on a big endian machine, that would be awesome, because I didn't manage to find an online big-endian compiler and a runtime environment);
2. You may not have the C++20 STL in the nearest future (or at all) for some platforms, but this is just a wild guess.

My point is that if ETL is striving to be **compile-time guaranteed** (TMP and constexpr) that this feature is probably unimplementable. It's a shame, because I myself would like to have that.

shyun3, 2019-02-26:
Currently, we can rely on compiler macros to determine endianness at compile time. For example, GCC defines the macros `__BYTE_ORDER__`, `__ORDER_LITTLE_ENDIAN__`, and `__ORDER_BIG_ENDIAN__`. I'm sure other compilers have similar facilities. If not, then we could fall back to the current approach.

GatorQue, 2019-02-27:
I recommend looking at the following file from the Qt core library as a potential example for determining Endianness at compile time:
http://code.qt.io/cgit/qt/qtbase.git/tree/src/corelib/global/qprocessordetection.h

Xenoamor, 2019-09-20:
This method is optimised away at compile time:
https://github.com/tatewake/endian-template/blob/ea8392fd8f2a9930a0ea51e5ce612e562540634c/tEndian.h#L150

closed 2020-03-28 by jwellbelove