ETLCPP/etl #118: IAR test project not correct

ben-edna, 2019-02-20:
When trying to build [IAR test project](https://github.com/ETLCPP/etl/tree/master/test/iar), compilation fails due to missing several source files. 

These probably were renamed to `test_.*cpp` or not? 

```
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\crc16_ccitt.cpp" 
Error while running C/C++ Compiler 
crc16.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\crc16.cpp" 
Error while running C/C++ Compiler 
crc32.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\crc32.cpp" 
Error while running C/C++ Compiler 
crc16_kermit.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\crc16_kermit.cpp" 
Error while running C/C++ Compiler 
crc8_ccitt.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\crc8_ccitt.cpp" 
Error while running C/C++ Compiler 
crc64_ecma.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\crc64_ecma.cpp" 
Error while running C/C++ Compiler 
pearson.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\pearson.cpp" 
Error while running C/C++ Compiler 
error_handler.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\etl\error_handler.cpp" 
Error while running C/C++ Compiler 
test_compile.cpp  
Fatal Error[Pe1696]: cannot open source file "SOME_DIR\test_compile.cpp" 
```

jwellbelove, 2019-02-20:
The '.cpp' files no longer exist as the library was changed to 'header only' a while ago. They need to be removed from the project.
I don't have the IAR compiler on this machine, as I'm away on holiday in Spain for about two more weeks.

ben-edna, 2019-02-20:
Thanks for the quick response, and happy holiday 🌞. 
I can make an attempt at fixing it if you would welcome the PR?

RallyTronics, 2019-03-05:
I can confirm that ETL compiles fine without the .cpp files in the IAR project ( Cortex M4F ).

closed 2020-03-29 by jwellbelove