ETLCPP/etl #121: How do you actually install this library ?

GeorgeFlorian, 2019-03-21:
Hello !

I am trying to use this library on PlatformIO IDE, but I'm having no luck. 
I'm using it with `lib_deps = 930` inside `platformio.ini`.

I've read here: https://www.etlcpp.com/setup.html and I made `etl_profile.h` with 
```
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_THROW_EXCEPTIONS
#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP

#endif
```

Compiling gives me the following error:
```
In file included from .piolibdeps/Embedded Template Library_ID930/src/binary.cpp:31:0:
.piolibdeps/Embedded Template Library_ID930/src/platform.h:49:25: fatal error: etl_profile.h: No such file or directory
```

jwellbelove, 2019-03-21:
You don't need to include the **etl_profile.h** explicitly. It's already included by **platform.h**.
If **platform.h** can't see it then the header path must be included in the compiler's search paths.
**platform.h** should not be in the ETL's **src** sub-directory.
**ecl_timer.c** is only required if you are writing a pure C application; it's not really part of the C++ ETL proper (It's just an extra I threw in for C programmers)

The cpp files were removed from the library a while ago when it went to a 'header only' implementation.
You can delete all of the cpp files in **src**.

maxgerhardt, 2019-03-21:
@GeorgeFlorian Because the library.properties of this repository wasn't updated, PlatformIO still registers 10.21.2 from March 2018 as the most recent version, (see [here](https://platformio.org/lib/show/930/Embedded%20Template%20Library)). You can try the newest git version by doing `lib_deps = https://github.com/ETLCPP/etl.git` in your `platformio.ini`

jwellbelove, 2019-03-21:
library.properties was a hangover from when the library's directory layout was Arduino compatible. I didn't realise that anything else actually used it!

GeorgeFlorian, 2019-03-22:
Ok !

So I've uninstalled the library and then put `lib_deps = https://github.com/ETLCPP/etl.git` inside `platformio.ini`.

Here is the project's folder:

https://i.postimg.cc/J4q59wTd/21345.png

As you can see, I still can't use headers from the ETL library because the program can't find them.

jwellbelove, 2019-03-22:
You need to set the path to the ETL's "include" directory as one of the projects addition include paths so that you can say
#include "etl/deque.h"

GeorgeFlorian, 2019-03-22:
> You need to set the path to the ETL's "include" directory as one of the projects addition include paths so that you can say
> #include "etl/deque.h"

Thank you for the fast reply !

I have no idea how to do that ! I will search through PlatformIO documentation. I already posted a thread on their forums.

maxgerhardt, 2019-03-22:
@GeorgeFlorian 
> I have no idea how to do that ! I will search through PlatformIO documentation. I already posted a thread on their forums.

I think he means to replace `#include <dequeue.h>` with `#include <ets/dequeue.h>`.  Though you may try to add include paths by adding `build_flags -I .piolibdeps/Embedded Template Library_ID930/include/etl` ([docs](http://docs.platformio.org/en/latest/projectconf/section_env_build.html#build-flags)).

jwellbelove, 2019-03-22:
@maxgerhardt 
Yes, that's what I meant.

jwellbelove, 2019-03-22:
You may want to set the library path to exclude the final 'etl' so that you can easily disabiguate headers with the same name.
#include "etl/algorithm.h"
#include "xyz/algorithm.h"

ristomatti, 2019-04-15:
@GeorgeFlorian Thank you for posting this question and of course @jwellbelove for the answer! This inspired me to try to make ETL work with Arduino (using PlatformIO) the 3rd time and finally got it working.

I had no interest in using an old version so I've never tried the version that's supposed to work. The key for me was John's comment about the files in src being optional and that ecl_timer.h was not required. What I did in the end was:
1. Cloned the repo to `lib/ETL` under the project folder
2. Deleted `lib/ETL/src/c/ecl_timer.c`
3. Created `include/etl_profile.h` with minimal configuration (I haven't yet read the documentation on the macros) and 
4. Added this build flag to platformio.ini under the `[env:board_id]` section:
```
build_flags = -I ./lib/ETL/include
```

After this I was able to run the message router example and play around with unordered map seemingly without issues. I did the testing in an ESP32 board but it seemed to at least compile for ESP8266 also. 

It's a bit sad that this fascinating project is actually possible to setup to be run with Arduino (if you count ESP's with an Arduino core as ones) but it is very difficult to understand the required steps to someone like me who's only C/C++ experience is from 2 years of playing around with electronics and Arduino. I understand this library might not be targeted to my kind of an audience but I wouldn't have found myself here unless ETL was listed in the Arduino library index.

ristomatti, 2019-04-15:
@jwellbelove Possibly you could reconsider your stance on Arduino support? I doubt it would be too difficult for a seasoned C++ developer like you. :) You could just skip the classic Arduino library repository and focus on providing the library only through PlatformIO. It seems to be gaining more and more traction amongst the Arduino users.

One can dream at least right? :)

GeorgeFlorian, 2019-04-15:
> @GeorgeFlorian Thank you for posting this question and of course @jwellbelove for the answer! This inspired me to try to make ETL work with Arduino (using PlatformIO) the 3rd time and finally got it working.
> 

It's great that I was able to help !
I've stopped trying make it work, because I did not need it in the end. It's too big of a library for the things I needed.

jwellbelove, 2019-04-15:
@ristomatti I shall certainly look at PlatformIO support, if that is a popular option. I only dropped Arduino support as it demanded a directory layout that I didn't want to use. If you could do a pull request when you have finalised your PlatformIO setup then I can include it in the library. I could add a page specific to setting a project up for that on the web site.

ristomatti, 2019-04-15:
@GeorgeFlorian I can understand. I really wanted this to work and have tried twice earlier, spending an evening trying to figure out what do I need to do. I've got almost an obsession to find a way to make use of the tips that I've learned while reading Christopher Kormanyos' excellent book _Real Time C++: Effective Object Oriented and Template microcontroller programming_. All the other similar libraries I've found  have required CMake or some other system for building then that I are just too much for me at the moment (for example EASTL). Every time I've tried googling I've ended up here.

I have not yet tried if ETL would compile or make sense on micros like the classic Atmel's but on a ESP32 the memory overhead didn't seem to be that bad at all. Given the fact I'm a big fan of abstractions and elegant code, I think I'll be using ETL for many of my projects in the future.

Glad you found a solution that worked for you anyhow!

jwellbelove, 2019-04-15:
@ristomatti I'm sorry that I wasn't able to put as much effort into helping you as I would have liked, but the day job (the one that pays the bills) has been taken up a lot of my time recently (big delivery deadline!) and my time at the weekends has been mostly dedicated to 'chilling out'.

ristomatti, 2019-04-15:
@jwellbelove Just yesterday Andreas Spiess released [https://www.youtube.com/watch?v=0poh_2rBq7Eurl](a YouTube video) demonstrating how to set it up for Arduino development. He's channel is very popular amongst similar minded people like me. It seemed to have a very good initial  response from the viewers. Many seem to have had the same frustration I initially had when I tried Arduino IDE. As a developer who's used to modern IDE's, I wouldn't have got into the whole Arduino thing (and through that learning some C/C++) unless I wouldn't have come across PlatformIO by accident. It's an actual developer tool and targeting other platforms also, like ESP-IDF and Mbed.

And hey, no need to be sorry! I don't recałl ever even posting a question here as I thought this might be "impossible" to adapt to Arduino in a reasonable effort. I've got the understanding also this is a personal project of yours and can definately understand you can only put a limited amount of time supporting this. Especially if your day job is also focused around embedded development or consulting, it's no wonder you want to do something else on your free time. For me this is just refreshing after a workday developing a huge B2B ecommerce system in Java/React.

I would love to do a PR but I'm afraid my skill level is not there yet. Otherwise it would be enough to just add this repo's git address in the list of dependencies but I don't know how should the build configuration be adjusted to make it exclude the "offending" .c file that's breaking the build other then deleting it. I could possibly figure out how to add the include folder in the build path but I'm not 100% sure. The problem is that the library dependencies to a project get automatically cloned to a folder `.piolibdeps` with a changing hash in the directory name.

I also believe the new directory structure is better exactly due to the reasons mentioned on some closed issue predating the change. It's better to be able to include the files with the etl/ prefix.

BTW I've also run across your old post to Arduino forums about ETL and reading the bashing from grey beards made me sad. But it was somehow understandable as setting this up has not exactly been "Arduino easy". :) 

P.S. If you happen to watch the video I linked, the part where he talks about installing Arduino IDE first is misleading. It is not required at all. Need to give some slack to a 60+y old engineer learning and teaching the younger generation about stuff he has learned during his life...

ristomatti, 2019-04-15:
Although my current setup would likely be reproducible as just written instructions, it would beat the purpose of the PlatformIO dependency system which would normally do the cloning automatically. For each library there's an installation section but I'm not sure if it's just 100% automatically generated. Otherwise it would be a good place to put this additional configuration that was required. I believe ETL is already in the index but it's likely scrubbed from the regular Arduino library index somehow unless you've added it.

PlatformIO allows you to set the target platforms (avr, STM32, ESP8266 etc) and frameworks (Arduino, Mbed etc) supported by your library. So theoretically it should be possible to gain also users from other "camps" of developers also, if that is something you'd like.

jwellbelove, 2019-04-15:
@ristomatti I've just finished adding 'to_string' functionality to the library and I need to update the website to suit, but after that I could look at setting up PlatformIO and get some first hand experience of it.

"Need to give some slack to a 60+y old engineer"
Careful, I'm not far off 60 myself! ;-)

ristomatti, 2019-04-15:
Haha, point taken. Yes it was your profile picture that led me to think you've got to me a lot younger than Andreas. :) I'm myself reaching "only" 40 but now after two years of trying to learn C/C++ (and electronics) from scratch has made me feel like a junior developer again. Having been spoiled for years with high level languages, all of this seems extremely complex and often times it's difficult to even know where to start.

It would probably help if I'd try to write some C++ for my various Raspberry Pi's for example as my go-to higher level languages can be too resource hungry for some purposes. There I'd at least get a faster feedback loop than when flashing a firmware but there I do not have the comfort of the Arduino API's to help me, and all the OS level complexity comes to my way. Then again it feels like too big of a sidetrack from my main interest of creating networked devices for automation and monitoring purposes both home and at our summer cottage. :)

ristomatti, 2019-04-15:
> I've just finished adding 'to_string' functionality to the library and I need to update the website to suit,

I happened to notice this yesterday evening but have not yet looked up into what it can be used for but I reckon it has something to do with converting other than strings into strings...

> but after that I could look at setting up PlatformIO and get some first hand experience of it.

It'd be awesome if you would willing to take the time! It might be fun/interesting also from a professional point of view, who knows.

In case you don't know anything about PlatformIO, some background info if your interested:

Their (quite ambitious) goal is to make it possible to develop on various embedded (IoT oriented) platforms/frameworks with the IDE of your liking (currently Atom, CLion, CodeBlocks, Eclipse, Emacs, NetBeans, QtCreator, Sublime Text, Vim, Visual Studio, Visual Studio Code).

It's basically just a set of Python scripts that do all the work in the background. All the features are available from the CLI tool but they've relatively recently build deeper integration into some editors like Atom and Visual Studio Code (which is what they support the best at the moment) to support debugging.

Debugging is only available for the subscription users but they have some pretty interesting other features that come with the subscription, like being able to flash, debug and monitor (by serial) boards remote over the network/cloud. So you could for instance have a dedicated server with various targeted devices attached and then run automated test suites on new firmware from a separate development laptop. 

I'm on their non-commercial subscription ($2.99/month) myself to be able to use the debugging feature but I'm also curious to experiment with the remote features at some point. I honestly would never have managed to setup STM32 or ESP32 debugging with all of the related tools with their insanely detailed configurations without it. Now it was pretty much just reading their documentation, choosing a debugger supported by them for the target platform and adding a couple of lines to the project config file and I was step debugging some Arduino code like I do with Java at work. :sunglasses:

ristomatti, 2019-04-23:
@jwellbelove Hi! Did you notice PR #124 ? I know you're busy so not to hurry, only checking.

jwellbelove, 2019-04-23:
Yes, I saw it, but thanks for reminding me, I'll look at it very soon 👍

ristomatti, 2019-04-23:
@jwellbelove If you have any specific questions regarding setting up PlatformIO, please feel free to contact me via email for example (firstname.lastname@gmail.com). It should be straightforward though and my PR has the project settings. Thanks again for looking into this! I've also been very busy and I'm anxiously waiting for the first suitable moment to experiment more with ETL.

puzrin, 2019-09-22:
@ristomatti, what are reasons to restrict lib to esp & arduino in your PIO librari config? That does not seems to solve any problems and restricts use on `native`, `stm32`, other platforms & frameworks.

IMO it would be safe to drop `platforms` and `frameworks` properties at all. What do you think?

ristomatti, 2019-09-22:
@puzfrin Arduino is the only framework I have experience on and I was only successful in getting it to build on `esp8266` and `esp32` from the platforms I had available for testing at the time. It's been a while but if I remember correctly I was not able to get it to compile for classic 8bit Atmel boards as an example. I believe I also tested with some STMF103 boards. At least not without additional dependencies like ArduinoSTL. At that time PlatformIO supported defining environment specific library dependencies `platformio.ini` but all libraries were still downloaded into the same directory which caused compilation to fail on platforms not needing the added dependencies, even if they weren't `#include`d. Nowadays when dependencies are downloaded to env specific dirs this might not be an issue.

I personally don't like libraries listing all frameworks or platforms that it hasn't been tested at all. I wouldn't want to maintain such a library either.

puzrin, 2019-09-22:
Ok, thanks for comments. I've created new issue with proposal of less restrictive changes & fixes https://github.com/ETLCPP/etl/issues/151. It also refers to sample project as proof that everything (?) works. It's used with "native" platform, without framework at all.

Note, i'm not C programmer, and may miss some points. Focused only on PIO-related changes. Not familiar what is ArduinoSTL and why it's needed. If this issue still exists, let me know. I just copied code from tutorial, and did not dived into details.

jwellbelove, 2019-09-22:
@puzrin I'm thinking that I shall remove the ECL components from the library completely.
ArduinoSTL was included as the Arduino development platform did not come with an STL implementation. It may now not be required if the project defines ETL_NO_STL. I haven't experimented with an Arduino for some time. I've not had very much experience with PIO yet, so I've been relying on ristomatti's expertise on this.

puzrin, 2019-09-22:
Sorry if i say something stupid, i'm not C programmer. Demo, refered in #151 works without `ETL_NO_STL` option. But i only copied first available tutorial from your website into `main.cpp`. If you show me better sample with build problems, i will try to fix. PIO comes with gcc, i thought it should contain stl by default.

jwellbelove, 2019-09-22:
Yes, GCC comes with the STL, but the compiler supplied with the Arduino development platform didn't. Things may have changed since I last looked.

puzrin, 2019-09-22:
If i understand right, you mean Arduino IDE. In this case, PIO is not affected anyhow. In PIO, arduino is just a library with well-known API. Compiler is full-featured gcc, without ugly restrictions.

Since `etl` is transparent to "everything", i just cleared PIO config to remove restrictions, and added required changes to demo's config to make things work. I have absolutely no idea what is `/src/c/ecl_timer.c` for, but build pass, and executable works as expected.

ristomatti, 2019-09-23:
@jwellbelove I noticed ETL 14.31.0 just appeared in PlatformIO library index, nice! I also encountered the syntax error in `library.json` which you've apparently already fixed in 14.31.1. It just hasn't updated on the library index.

I saw you had deleted the files I had excluded in the build flags which is also nice. I decided to test if it works with my old test project (I used `lib_deps =  https://github.com/ETLCPP/etl.git` to get the newest version). Unfortunately it does not compile any longer with the updated version. But it does if I add the build flags back:

```json
  "build": {
    "flags": [
      "-I include"
    ]
  }
```

Was there a specific reason why you decided to remove these?

jwellbelove, 2019-09-23:
I removed it at the suggestion of Vitaly as he said it did nothing! Obviously it does!

puzrin, 2019-09-23:
Let's wait until PIO registry fetch new version with correct config, and i will test everything at once.

https://github.com/puzrin/etl_check/blob/master/platformio.ini#L22

This should help right now if you have problems. But I checked everything before suggest changes.

My goal is to make etl available in pio for all, in clear and simple way. If i missed something - that will be fixed.

puzrin, 2019-09-23:
https://github.com/puzrin/etl_check/blob/master/platformio.ini#L22

If you can't wait, this line should help. But i tested everything before suggest changes.

Let's wait until PIO fetch new version with fixed config, and i will recheck all at once.

puzrin, 2019-09-23:
Hm... it really fails. Probably, cross-issue with last changes (src folder deleted).

This change fixed things for me:

```json
,
  "build": {
    "includeDir": "include"
  }
```

Also, `repository` field does not conforms docs. Correct should be

```
  "repository": {
    "type": "git",
    "url": "https://github.com/ETLCPP/etl.git"
  },
```

@jwellbelove could you update? I hope, this will be final.

puzrin, 2019-09-23:
https://github.com/puzrin/etl_check - here is updated demo, if someone wish to test.

After first failed build, go to `/.pio/libdeps/native/etl` and apply fixes above. Then it works.

If more issues exists - let me know.

ristomatti, 2019-09-23:
I assume `"includeDir": "include"` is basically the same as what I had added there originally. `master` branch of ETL has had it in it's `library.json` since May. Your original example just used that version. It was based on what I suggested to be added here: https://github.com/ETLCPP/etl/pull/124.

puzrin, 2019-09-24:
In theory, my variant is more descriptive in context of PIO. But result is the same, you are right.

> Your original example just used that version.

No, it used version from cloned repo with [this file](https://github.com/puzrin/etl/commit/caa5da8ec4c4e1c87e56e7eac11fd13c228a3561). I don't like problems with tagged releases, and really tested all as much as were possible.

puzrin, 2019-09-26:
14.31.2 works again :). I also did PR with primitive demo.

IMO this issue can be closed.

ristomatti, 2019-09-26:
@GeorgeFlorian I've gobbled together two examples of using the just updated ETL with Arduino: https://github.com/ristomatti/arduino-etl-examples.

GeorgeFlorian, 2019-09-27:
I thank you all for your answers. I really do ! But this thread is old and I ended up not using this library.

closed 2019-09-27 by GeorgeFlorian

ristomatti, 2019-09-27:
Well at least the issue resulted in many others being able to use the library. :)