ETLCPP/etl #123: improve etl::function

creibetanz, 2019-04-12:
Hi, I would like to suggest to update etl::function to accept all kinds of return values (not just void) and any number of parameters (instead of one).
Also I would like to change the syntax from etl::function<TObject, TParameter> to etl::function<TObject, TFunction> to be able to do things like:
etl::function<MyClass, int(float, int, bool)>.
This seems easier to read, than something like etl::function<MyClass, float, int, bool, int> or similar and it is also closer to the syntax of std::function.

I would suggest an implementation like:

```cpp
#include <iostream>

namespace etl {
template <typename TFunction>
class ifunction {
};
template <typename TReturnValue, typename ... TArgs>
class ifunction<TReturnValue(TArgs ...)> {
public:
    virtual TReturnValue operator ()(TArgs ... args) const = 0;
};

template <typename TObject, typename TFunction>
class function : public ifunction<TFunction> {
};

template <typename TObject, typename TReturnValue, typename ... TArgs>
class function<TObject, TReturnValue(TArgs ...)> : public ifunction<TReturnValue(TArgs ...)> {
public:
    using TFunction = TReturnValue(TArgs ...);
    function(TObject& object_, TFunction(TObject:: * p_function_))
        : p_object(&object_),
        p_function(p_function_) {
    }

    virtual TReturnValue operator ()(TArgs ... args) const {
        return (p_object->*p_function)(args...);
    }
private:
    TObject* p_object;
    TFunction(TObject:: * p_function);
};

template <typename TReturnValue, typename ... TArgs>
class function<void, TReturnValue(TArgs ...)> : public ifunction<TReturnValue(TArgs ...)> {
public:
    using TFunction = TReturnValue(TArgs ...);
    function(TFunction(* p_function_))
        : p_function(p_function_) {
    }

    virtual TReturnValue operator ()(TArgs ... args) const {
        return (*p_function)(args...);
    }
private:
    TFunction(* p_function);
};

}

// tests

class MyClass {
public:
    void foo_ret_void_param_int(int i) {
        std::cout << i << std::endl;
    }
  	void foo_ret_void_param_void() {
        std::cout << "void foo_ret_void_param_void(void)" << std::endl;
    }
    int foo_ret_int_param_float_float(float f1, float f2) {
      	std::cout << "int foo_ret_int_param_float_float(float f1, float f2)" << std::endl;
        std::cout << f1 << std::endl;
        std::cout << f2 << std::endl;
        return f1 + f2;
    }
};
void call_void_int(etl::ifunction<void(int)> * callee) {
    (*callee)(23);
}
void call_int_float_float(etl::ifunction<int(float, float)> * callee) {
    std::cout << (*callee)(3.0, 4.0) << std::endl;
}
void call_void_void(etl::ifunction<void()> * callee) {
    (*callee)();
}

bool my_free_function(bool a, bool b) {
    std::cout << a << std::endl;
    std::cout << b << std::endl;
    std::cout << "bool my_free_function(bool b)" << std::endl;
    return a | b;
}

void call_free_func(etl::ifunction<bool(bool, bool)> * callee) {
     std::cout << (*callee)(false, true) << std::endl;
}

int main() {
    MyClass my_inst;
    etl::function<MyClass, void(int)> func_void_int(my_inst, &MyClass::foo_ret_void_param_int);
    etl::function<MyClass, int(float, float)> func_int_float_float(my_inst, &MyClass::foo_ret_int_param_float_float);
    etl::function<MyClass, void()> func_void_void(my_inst, &MyClass::foo_ret_void_param_void);
    
    // I would be very happy, if anyone has an idea how to get rid of the following void
    etl::function<void, bool(bool, bool)> free_func_bool_bool(&my_free_function);

    call_void_int(&func_void_int);
    call_int_float_float(&func_int_float_float);
    call_void_void(&func_void_void);
    call_free_func(&free_func_bool_bool);
}
```

Of course, this is not all the code, that would be required. I would need to implement this for all those kind of consts, volatiles and references and write the tests.
It would be an implementation, that is a lot more flexible than the one right now.

This would be a breaking API change.
Please let me know, if you would like to merge such a thing.

creibetanz, 2019-04-12:
Oh and of course, this would not be C++03 compatible and making it C++03 compatible would be almost impossible.

jwellbelove, 2019-04-12:
I thought about the idea creating an improved version of etl::function using C++11 techniques a few weeks ago and did some experiments but I didn't have the time to get very far. I liked the idea of making it more like std::function, but didn't have the time to work out how it was actually implemented. Any solution would have to include all of the compile time object/function address variant and of course not use any dynamic memory (which I am led to believe can occur when using std::function & bind).
It would probably be better to create a whole new set of classes under a different name, such as 'delegate' to allow the old etl::function to continue to exist.

swagggpickle, 2019-04-12:
I have played around with this idea as well, what could work and allow it to still be c++03 compatible is template specialization. This is a very basic concept that could be expanded:
```
namespace mstl
{
  template<typename Functor> struct function;
  template<typename Functor> struct function_invoker;
};
#include "Function_0.h"
#include "Function_1.h"
```

```
// Function_0.h
namespace mstl 
{

template<typename R>
  struct cb_base0<R>
  {
    virtual R run() = 0;
  }
  
  template<>
  struct cb_base0<void>
  {
    virtual void run() = 0;
  }
  
  template<typename R, typename T>
  struct cb0 : public cb_base0<R>
  {
    typedef R(T::*method_ptr_t)();
    cb0(T* inst_ptr, R(T::*method)()) : inst(inst_ptr), fun(method)
    R run()
    {
       return inst->*fun();
    }
    T* inst;
    method_ptr_t fun;
  };

  template<typename R> 
  struct function<R()>
  {
    public:
    typedef R return_type;
    typedef return_type (*function_type)();
    typedef function_invoker<function_type> invoke_type;
    typedef return_type (*cb_type)(void* inst_ptr);
    typedef function<R()> self_type;
    typedef cb_base0<R>   cb_base_t;
   

    function() : inst_ptr(NULL), cb_ptr(NULL), cb(NULL) {};
    function(function_type ptr_in) : inst_ptr(reinterpret_cast<void*>(ptr_in)), cb_ptr(&func), cb(NULL){};
    template<typename T>
    function(T* inst_ptr_in, R(T::*method)())
    {
      cb = new cb1<R, T>(inst_ptr_in, method);
      inst_ptr = static_cast<void*>(cb);
      cb_ptr = &func2;
    }

    return_type operator()() const
    {
      return (*cb_ptr)(inst_ptr);
    }

    operator bool () const
    {
      return cb_ptr != 0;
    };

    bool operator!() const
    {
      return !(operator bool());
    };

    private:
    static return_type func(void* inst)
    {
      function_type temp = reinterpret_cast<function_type>(inst);
      return temp();
    }
    static return_type func2(void* inst)
    {
      cb_base_t* temp = stat<cb_base_t*>(inst);
      return temp->run();
    }

   // member variables
    private:
    void* inst_ptr;
    cb_type cb_ptr;
    cb_base_t* cb;
};
};
```

```
namespace mstl 
{
  template<typename R, typename T1>
  struct cb_base1<R>
  {
    virtual R run(T1 a1) = 0;
  }
  
  template<typename T1>
  struct cb_base1<void, T1>
  {
    virtual void run() = 0;
  }
  
  template<typename R, typename T, typenam T1>
  struct cb1 : public cb_base1<R, T1>
  {
    typedef R(T::*method_ptr_t)(T1);
    cb1(T* inst_ptr, R(T::*method)(T1)) : inst(inst_ptr), fun(method){};
    R run(T1 a1)
    {
       return inst->*fun(a1);
    }
    T* inst;
    method_ptr_t fun;
  };

  template<typename R, typename T1> 
  struct function<R(T1)>
  {
    public:
    typedef R return_type;
    typedef return_type (*function_type)(T1);
    typedef return_type (*cb_type)(void* inst_ptr, T1 a1);
    typedef function<R(T1)> self_type;
    typedef cb_base1<R, T1>   cb_base_t;
   

    function() : inst_ptr(NULL), cb_ptr(NULL), cb(NULL) {};
    function(function_type ptr_in) : inst_ptr(reinterpret_cast<void*>(ptr_in)), cb_ptr(&func), cb(NULL){};
    template<typename T>
    function(T* inst_ptr_in, R(T::*method)(T1))
    {
      cb = new cb1<R, T, T1>(inst_ptr_in, method);
      inst_ptr = static_cast<void*>(cb);
      cb_ptr = &func2;
    }

    return_type operator()(T1 a1) const
    {
      return (*cb_ptr)(inst_ptr, a1);
    }

    operator bool () const
    {
      return cb_ptr != 0;
    };

    bool operator!() const
    {
      return !(operator bool());
    };

    private:
    static return_type func(void* inst, T1 a1)
    {
      function_type temp = reinterpret_cast<function_type>(inst);
      return temp(a1);
    }
    static return_type func2(void* inst, T1 a1)
    {
      cb_base_t* temp = stat<cb_base_t*>(inst);
      return temp->run(a1);
    }

   // member variables
    private:
    void* inst_ptr;
    cb_type cb_ptr;
    cb_base_t* cb;
};
```

creibetanz, 2019-04-12:
As I said, it is only almost impossible to do it with C++03. With template specialization it is kind of possible, but a whole lot of work. You need to write all the stuff for every number of parameters. And the code will get even less readable.

What do you mean with "all of the compile time object/function address variant"?

I do not think it is a good idea to create a new class like delegate.
Especially if the new implementation works even more like std::function.

jwellbelove, 2019-04-13:
etl::function has variants (e.g. etl::function_imp) that takes *all* of the object and function address information at *compile* time. This is something that std::function does not do. (edited)

jwellbelove, 2019-04-13:
@swagggpickle 
I can see that you are using type erasure in your solution as does std::function.
You are also using 'new' to allocate from the heap. cb = new cb1<R, T, T1>(inst_ptr_in, method);
This is not allowed in the ETL. The ETL guarantees not to use the heap.

jwellbelove, 2019-04-13:
@creibetanz I actually regret calling it etl::function because it does not work exactly like std::function.
Originally it was named etl::callback, as that was the sort of application it was intended for. 'delegate' seemed to me to be a better generic term than 'callback' or 'function'.

creibetanz, 2019-04-15:
Hm, I worked a bit on it and here is the outcome:
I took the "tests" from https://en.cppreference.com/w/cpp/utility/functional/function and http://www.cplusplus.com/reference/functional/function/function/.

What works:
free functions, member functions, data members, bool, no dynamic memory

What doesn't work:
- lambdas (quite impossible I guess, as they are compiler implementation specific function objects)
- Function objects (kind of possible I guess, if you add them as an extra template argument, but syntax and usage quite confusing)
- no etl::bind, as this would result in such a function object
- a lot of const/volatile/reference combinations are missing, but I can add them
- not C++03 compatible (not really "hard" to do for like 3 Arguments, but still a lot of work and I can already use C++11, so ...)
- (code not really clean :D)

Regarding the naming:
I would suppose to really call such an implementation "etl::function", as it is pretty close to std::function.
If you are fine with this and tell me I can do a pull request and you will merge something like this, I'm willing to do the whole implementation including tests for C++11 and all kind of const/volatile/reference implementation. Maybe I could also do an implementation for one Parameter for C++03.

Any suggestions are highly welcome.

``` cpp
#include <iostream>

namespace etl {

template <typename TFn>
class base_function {
};
template <typename TObj, typename ... TArgs, typename TRet>
class base_function<TRet(TObj&, TArgs ...)> {
public:
    virtual TRet operator ()(TObj p_inst_, TArgs ... args) const = 0;
};

template <typename TMemFn>
class function : public base_function<TMemFn> {
};
template <typename TObj, typename ... TArgs, typename TRet>
class function<TRet(const TObj&, TArgs ...)> : public base_function<TRet(TObj&, TArgs ...)> {
public:
    typedef TRet (TObj::*TMemFn)(TArgs ... args);
    function() :
        p_memfn(nullptr) {
    }
    function(TMemFn p_memfn_) :
        p_memfn(p_memfn_) {
    }
    TRet operator ()(TObj p_inst_, TArgs ... args) const {
        return (p_inst_.*p_memfn)(args...);
    }
    template <typename TCtorArg>
    TRet operator ()(TCtorArg&& ctor_arg_, TArgs ... args) const {
        TObj temp_inst(ctor_arg_);
        return (temp_inst.*p_memfn)(args...);
    }
private:
    TMemFn p_memfn;
};

template <typename TObj, typename ... TArgs, typename TRet>
class function<TRet(TObj&, TArgs ...)> : public base_function<TRet(TObj&, TArgs ...)> {
public:
    typedef TRet (TObj::*TMemFn)(TArgs ... args);
    function() :
        p_memfn(nullptr) {
    }
    function(TMemFn p_memfn_) :
        p_memfn(p_memfn_) {
    }
    typedef TRet (TObj::*TMem);
    function(TMem p_mem_) :
        p_mem(p_mem_) {
    }
    TRet operator ()(TObj p_inst_, TArgs ... args) const {
        if (!p_mem) {
            return (p_inst_.*p_memfn)(args...);
        } else {
            return (p_inst_.*p_mem);
        }
    }
    template <typename TCtorArg>
    TRet operator ()(TCtorArg&& ctor_arg_, TArgs ... args) const {
        TObj temp_inst(ctor_arg_);
        return (temp_inst.*p_memfn)(args...);
    }
private:
    TMemFn p_memfn;
    TMem p_mem = nullptr;
};
template <typename ... TArgs, typename TRet>
class function<TRet(TArgs ...)> : public base_function<TRet(TArgs ...)> {
public:
    typedef TRet (*TFn)(TArgs ... args);
    function() :
        p_fn(nullptr) {
    }
    function(TFn p_fn_) :
        p_fn(p_fn_) {
    }
    function(base_function<TFn> bind_obj) {
    }
    TRet operator ()(TArgs ... args) const {
        return (*p_fn)(args...);
    }
    operator bool() const{
        return (p_fn != nullptr);
    }
private:
    TFn p_fn;
};

}

// tests

struct Foo {
    Foo(int num) : num_(num) {}
    void print_add(int i) { std::cout << num_+i << '\n'; }
    int num_;
};
void print_num(int i) { std::cout << i << '\n'; }
 
// a function:
int half(int x) {return x/2;}
// a function object class:
struct third_t {
  int operator()(int x) {return x/3;}
};
// a class with data members:
struct MyValue {
  int value;
  int fifth() {return value/5;}
};

void sampleFunction() {
    std::cout << "This is the sample function!\n";
}
void checkFunc( etl::function<void()> &func ) {
    // Use operator bool to determine if callable target is available.
    if(func) {
        std::cout << "Function is not empty! Calling function.\n";
        func();
    } else {
        std::cout << "Function is empty. Nothing to do.\n";
    }
}

int main() {    
    // store a free function
    etl::function<void(int)> f_display = print_num;
    f_display(-9);

    // store a call to a member function
    etl::function<void(const Foo&, int)> f_add_display= &Foo::print_add;
    const Foo foo(314159);
    f_add_display(foo, 1);
    f_add_display(314159, 1);

    etl::function<int(int)> fn1 = half;                    // function
    etl::function<int(int)> fn2 = &half;                   // function pointer
    // stl::function<int(int)> fn3 = third_t();               // function object
    // std::function<int(int)> fn4 = [](int x){return x/4;};  // lambda expression
    // stl::function<int(int)> fn5 = std::negate<int>();      // standard function object

    std::cout << "fn1(60): " << fn1(60) << '\n';
    std::cout << "fn2(60): " << fn2(60) << '\n';
    //std::cout << "fn3(60): " << fn3(60) << '\n';
    //std::cout << "fn4(60): " << fn4(60) << '\n';
    //std::cout << "fn5(60): " << fn5(60) << '\n';

    // stuff with members:
    etl::function<int(MyValue&)> value = &MyValue::value;  // pointer to data member
    etl::function<int(MyValue&)> fifth = &MyValue::fifth;  // pointer to member function

    MyValue sixty {60};

    std::cout << "value(sixty): " << value(sixty) << '\n';
    std::cout << "fifth(sixty): " << fifth(sixty) << '\n';

    etl::function<void()> f1;
    etl::function<void()> f2( sampleFunction );
 
    std::cout << "f1: ";
    checkFunc( f1 );
 
    std::cout << "f2: ";
    checkFunc( f2 );
}
```

jwellbelove, 2019-04-15:
I would be open to a pull request as long as users of the current etl::function are not affected in any way (i.e. must be 100% backward compatible). It must not use the C++11 version automatically if the compiler is set to >=C++11 if in doing so it will break existing users code.

jwellbelove, 2019-08-06:
The ETL now has a delegate class based on https://www.codeproject.com/Articles/1170503/The-Impossibly-Fast-Cplusplus-Delegates-Fixed.
I've improved on it by adding more compile time creation options.
etl::delegate is for C++11 and above only.

closed 2019-08-06 by jwellbelove