ETLCPP/etl #126: Add `emplace()` method to `etl::optional`

annius, 2019-04-16:
This feature request would add parity to C++17 `std::optional::emplace()`.

https://en.cppreference.com/w/cpp/utility/optional/emplace

ristomatti, 2019-04-18:
@annius I just happened to read ETL's "Get involved" page earlier today and thought I'd comment that the need is likely already acknowledged.

>There are a number of jobs outstanding at the time of writing that could usefully be taken up whole, or in part, by someone else.
> 1. Add C++03 compatible C++11 'emplace' functions to the rest of the ETL containers.
> 2. Add C++11 variadic 'emplace' functions to all containers, selected by a pre-processor switch for ETL_CPP11_SUPPORTED

Quote from here: https://www.etlcpp.com/get_involved.html

annius, 2019-04-18:
:+1: 

I am happy to reduce some of the tech debt and write `etl::optional::emplace`; should be fairly straightforward as there is one under `etl::vector`. My only hesitation was that this would need a unit test and it would take a while for me to write one (I've only uses Google Test before).

annius, 2019-05-02:
Looks like it's been added; closing this.

closed 2019-05-02 by annius