ETLCPP/etl #127: variant_pool vended the same address for two allocations

phillipjohnston, 2019-04-19:
I try to allocate two uint16_t variables from an etl::variant_pool that supports uint8_t and uint16_t. In this scenario, the variant pool returned the same address for both allocations. This happened with gcc-arm-none-eabi (latest) + C++17 with debug symbols and no optimizations enabled.

Pool declaration (128 is an arbitrarily high value):

```
/// Static memory pool used for I2C transactions.
etl::variant_pool<128, uint16_t, uint8_t> i2c_pool_;
```

I try to allocate the two values:

```
auto r = i2c_pool_.create<uint16_t>(READ_RANGE_REG);
auto out = i2c_pool_.create<uint16_t>();
```

And they both end up with the same address:

```
(gdb) p r
$31 = (unsigned short *) 0x20006528
(gdb) p out
$32 = (unsigned short *) 0x20006528
```

Here's a print-out of the pool before allocation. Note that `p_next` is 0x20006528.

```
(gdb) p i2c_pool_
$23 = {static MAX_SIZE = 128, pool = {<etl::ipool> = {
      p_buffer = 0x20006510 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15488> "\001\017",
      p_next = 0x20006528 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15512> "(e", items_allocated = 34,
      items_initialised = 128, ITEM_SIZE = 4, MAX_SIZE = 128}, static SIZE = 128, static ALIGNMENT = <optimized out>, static TYPE_SIZE = 2, buffer = {{{data = "\001\017\000 ",
          etl_alignment_type = 536874753}}, {{data = "\000\001\000 ", etl_alignment_type = 536871168}}, {{data = "\000\002\000 ", etl_alignment_type = 536871424}}, {{
          data = "\264e\000 ", etl_alignment_type = 536896948}}, {{data = "\000\345\000 ", etl_alignment_type = 536929536}}, {{data = "\003e\000 ",
          etl_alignment_type = 536896771}}, {{data = "(e\000 ", etl_alignment_type = 536896808}}, {{data = "\000\003\000 ", etl_alignment_type = 536871680}}, {{
          data = "\000\004\000 ", etl_alignment_type = 536871936}}, {{data = "\000\005\000 ", etl_alignment_type = 536872192}}, {{data = "\000\006\000 ",
          etl_alignment_type = 536872448}}, {{data = "\000\a\000 ", etl_alignment_type = 536872704}}, {{data = "\000\b\000 ", etl_alignment_type = 536872960}}, {{
          data = "\000\t\000 ", etl_alignment_type = 536873216}}, {{data = "\000\n\000 ", etl_alignment_type = 536873472}}, {{data = "\000\v\000 ",
          etl_alignment_type = 536873728}}, {{data = "\000\f\000 ", etl_alignment_type = 536873984}}, {{data = "\000\r\000 ", etl_alignment_type = 536874240}}, {{
          data = "\000\016\000 ", etl_alignment_type = 536874496}}, {{data = "\000\017\000 ", etl_alignment_type = 536874752}}, {{data = "\000\020\000 ",
          etl_alignment_type = 536875008}}, {{data = "\000\021\000 ", etl_alignment_type = 536875264}}, {{data = "\000\022\000 ", etl_alignment_type = 536875520}}, {{
          data = "\000\023\000 ", etl_alignment_type = 536875776}}, {{data = "\000\024\000 ", etl_alignment_type = 536876032}}, {{data = "\000\025\000 ",
          etl_alignment_type = 536876288}}, {{data = "\000\026\000 ", etl_alignment_type = 536876544}}, {{data = "\000\027\000 ", etl_alignment_type = 536876800}}, {{
          data = "\000\030\000 ", etl_alignment_type = 536877056}}, {{data = "\000\031\000 ", etl_alignment_type = 536877312}}, {{data = "\000\032\000 ",
          etl_alignment_type = 536877568}}, {{data = "\000\033\000 ", etl_alignment_type = 536877824}}, {{data = "\000\034\000 ", etl_alignment_type = 536878080}}, {{
          data = "\000\035\000 ", etl_alignment_type = 536878336}}, {{data = "\000\036\000 ", etl_alignment_type = 536878592}}, {{data = "\000\037\000 ",
          etl_alignment_type = 536878848}}, {{data = "\000 \000 ", etl_alignment_type = 536879104}}, {{data = "\000!\000 ", etl_alignment_type = 536879360}}, {{
          data = "\000\"\000 ", etl_alignment_type = 536879616}}, {{data = "\000#\000 ", etl_alignment_type = 536879872}}, {{data = "\000$\000 ",
          etl_alignment_type = 536880128}}, {{data = "\270e\000 ", etl_alignment_type = 536896952}}, {{data = "\274e\000 ", etl_alignment_type = 536896956}}, {{
          data = "\300e\000 ", etl_alignment_type = 536896960}}, {{data = "\304e\000 ", etl_alignment_type = 536896964}}, {{data = "\310e\000 ",
          etl_alignment_type = 536896968}}, {{data = "\314e\000 ", etl_alignment_type = 536896972}}, {{data = "\320e\000 ", etl_alignment_type = 536896976}}, {{
          data = "\324e\000 ", etl_alignment_type = 536896980}}, {{data = "\330e\000 ", etl_alignment_type = 536896984}}, {{data = "\334e\000 ",
          etl_alignment_type = 536896988}}, {{data = "\340e\000 ", etl_alignment_type = 536896992}}, {{data = "\344e\000 ", etl_alignment_type = 536896996}}, {{
          data = "\350e\000 ", etl_alignment_type = 536897000}}, {{data = "\354e\000 ", etl_alignment_type = 536897004}}, {{data = "\360e\000 ",
          etl_alignment_type = 536897008}}, {{data = "\364e\000 ", etl_alignment_type = 536897012}}, {{data = <incomplete sequence \370\145\000\040>,
          etl_alignment_type = 536897016}}, {{data = <incomplete sequence \374\145\000\040>, etl_alignment_type = 536897020}}, {{data = "\000f\000 ",
          etl_alignment_type = 536897024}}, {{data = "\004f\000 ", etl_alignment_type = 536897028}}, {{data = "\bf\000 ", etl_alignment_type = 536897032}}, {{data = "\ff\000 ",
          etl_alignment_type = 536897036}}, {{data = "\020f\000 ", etl_alignment_type = 536897040}}, {{data = "\024f\000 ", etl_alignment_type = 536897044}}, {{
          data = "\030f\000 ", etl_alignment_type = 536897048}}, {{data = "\034f\000 ", etl_alignment_type = 536897052}}, {{data = " f\000 ", etl_alignment_type = 536897056}}, {
        {data = "$f\000 ", etl_alignment_type = 536897060}}, {{data = "(f\000 ", etl_alignment_type = 536897064}}, {{data = ",f\000 ", etl_alignment_type = 536897068}}, {{
          data = "0f\000 ", etl_alignment_type = 536897072}}, {{data = "4f\000 ", etl_alignment_type = 536897076}}, {{data = "8f\000 ", etl_alignment_type = 536897080}}, {{
          data = "<f\000 ", etl_alignment_type = 536897084}}, {{data = "@f\000 ", etl_alignment_type = 536897088}}, {{data = "Df\000 ", etl_alignment_type = 536897092}}, {{
          data = "Hf\000 ", etl_alignment_type = 536897096}}, {{data = "Lf\000 ", etl_alignment_type = 536897100}}, {{data = "Pf\000 ", etl_alignment_type = 536897104}}, {{
          data = "Tf\000 ", etl_alignment_type = 536897108}}, {{data = "Xf\000 ", etl_alignment_type = 536897112}}, {{data = "\\f\000 ", etl_alignment_type = 536897116}}, {{
          data = "`f\000 ", etl_alignment_type = 536897120}}, {{data = "df\000 ", etl_alignment_type = 536897124}}, {{data = "hf\000 ", etl_alignment_type = 536897128}}, {{
          data = "lf\000 ", etl_alignment_type = 536897132}}, {{data = "pf\000 ", etl_alignment_type = 536897136}}, {{data = "tf\000 ", etl_alignment_type = 536897140}}, {{
          data = "xf\000 ", etl_alignment_type = 536897144}}, {{data = "|f\000 ", etl_alignment_type = 536897148}}, {{data = "\200f\000 ", etl_alignment_type = 536897152}}, {{
          data = "\204f\000 ", etl_alignment_type = 536897156}}, {{data = "\210f\000 ", etl_alignment_type = 536897160}}, {{data = "\214f\000 ",
          etl_alignment_type = 536897164}}, {{data = "\220f\000 ", etl_alignment_type = 536897168}}, {{data = "\224f\000 ", etl_alignment_type = 536897172}}, {{
[... 135 lines cut ...]
          data = "\030f\000 ", etl_alignment_type = 536897048}}, {{data = "\034f\000 ", etl_alignment_type = 536897052}}, {{data = " f\000 ", etl_alignment_type = 536897056}}, {
        {data = "$f\000 ", etl_alignment_type = 536897060}}, {{data = "(f\000 ", etl_alignment_type = 536897064}}, {{data = ",f\000 ", etl_alignment_type = 536897068}}, {{
          data = "0f\000 ", etl_alignment_type = 536897072}}, {{data = "4f\000 ", etl_alignment_type = 536897076}}, {{data = "8f\000 ", etl_alignment_type = 536897080}}, {{
          data = "<f\000 ", etl_alignment_type = 536897084}}, {{data = "@f\000 ", etl_alignment_type = 536897088}}, {{data = "Df\000 ", etl_alignment_type = 536897092}}, {{
          data = "Hf\000 ", etl_alignment_type = 536897096}}, {{data = "Lf\000 ", etl_alignment_type = 536897100}}, {{data = "Pf\000 ", etl_alignment_type = 536897104}}, {{
          data = "Tf\000 ", etl_alignment_type = 536897108}}, {{data = "Xf\000 ", etl_alignment_type = 536897112}}, {{data = "\\f\000 ", etl_alignment_type = 536897116}}, {{
[... cut ...]
          etl_alignment_type = 536897264}}, {{data = "\364f\000 ", etl_alignment_type = 536897268}}, {{data = <incomplete sequence \370\146\000\040>,
          etl_alignment_type = 536897272}}, {{data = <incomplete sequence \374\146\000\040>, etl_alignment_type = 536897276}}, {{data = "\000g\000 ",
          etl_alignment_type = 536897280}}, {{data = "\004g\000 ", etl_alignment_type = 536897284}}, {{data = "\bg\000 ", etl_alignment_type = 536897288}}, {{data = "\fg\000 ",
          etl_alignment_type = 536897292}}, {{data = "\020g\000 ", etl_alignment_type = 536897296}}}, static ELEMENT_SIZE = 4}}
```

phillipjohnston, 2019-04-19:
I see you just merged new changes a few days ago. I was using commit `35883fc37045f729e79fd660d9c238dcbb599fbc`. Will re-test with master.

phillipjohnston, 2019-04-19:
Still happens with the latest commit.

Interestingly, the first time through this function the addresses are different:

```
(gdb) p r
$33 = (unsigned short *) 0x2000651c <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15500>
(gdb) p out
$34 = (unsigned short *) 0x20006528 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15512>
(gdb) p i2c_pool_
$35 = {static MAX_SIZE = 128, pool = {<etl::ipool> = {
      p_buffer = 0x20006510 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15488> "\001\017",
      p_next = 0x2000652c <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15516> "0e", items_allocated = 7,
      items_initialised = 9, ITEM_SIZE = 4, MAX_SIZE = 128}, static SIZE = 128, static ALIGNMENT = <optimized out>, static TYPE_SIZE = 2, buffer = {{{data = "\001\017\000 ",
          etl_alignment_type = 536874753}}, {{data = "\001e\000 ", etl_alignment_type = 536896769}}, {{data = "\000\000\000 ", etl_alignment_type = 536870912}}, {{
          data = "\000\226\000 ", etl_alignment_type = 536909312}}, {{data = "\000\345\000 ", etl_alignment_type = 536929536}}, {{data = "\000e\000 ",
          etl_alignment_type = 536896768}}, {{data = "\000\000\000 ", etl_alignment_type = 536870912}}, {{data = "0e\000 ", etl_alignment_type = 536896816}}, {{
          data = "4e\000 ", etl_alignment_type = 536896820}}, {{data = "\000\000\000", etl_alignment_type = 0}} <repeats 119 times>}, static ELEMENT_SIZE = 4}}
```

The second time through (and all times following), they are the same:

```
$36 = (unsigned short *) 0x20006528 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15512>
(gdb) p out
$37 = (unsigned short *) 0x20006528 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15512>
(gdb) p i2c_pool_
$38 = {static MAX_SIZE = 128, pool = {<etl::ipool> = {
      p_buffer = 0x20006510 <embvm::VirtualPlatformBase<nRF52DK_FrameworkDemoPlatform, embvm::DriverRegistry<12u, 4u> >::inst()::inst_+15488> "\001\017",
      p_next = 0x20009600 "@B\017", items_allocated = 36, items_initialised = 48, ITEM_SIZE = 4, MAX_SIZE = 128}, static SIZE = 128, static ALIGNMENT = <optimized out>,
    static TYPE_SIZE = 2, buffer = {{{data = "\001\017\000 ", etl_alignment_type = 536874753}}, {{data = "\000\001\000 ", etl_alignment_type = 536871168}}, {{
          data = "\000\002\000 ", etl_alignment_type = 536871424}}, {{data = "\264e\000 ", etl_alignment_type = 536896948}}, {{data = "\000\345\000 ",
          etl_alignment_type = 536929536}}, {{data = "\003e\000 ", etl_alignment_type = 536896771}}, {{data = "\000\000\000 ", etl_alignment_type = 536870912}}, {{
          data = "\000\003\000 ", etl_alignment_type = 536871680}}, {{data = "\000\004\000 ", etl_alignment_type = 536871936}}, {{data = "\000\005\000 ",
          etl_alignment_type = 536872192}}, {{data = "\000\006\000 ", etl_alignment_type = 536872448}}, {{data = "\000\a\000 ", etl_alignment_type = 536872704}}, {{
          data = "\000\b\000 ", etl_alignment_type = 536872960}}, {{data = "\000\t\000 ", etl_alignment_type = 536873216}}, {{data = "\000\n\000 ",
          etl_alignment_type = 536873472}}, {{data = "\000\v\000 ", etl_alignment_type = 536873728}}, {{data = "\000\f\000 ", etl_alignment_type = 536873984}}, {{
          data = "\000\r\000 ", etl_alignment_type = 536874240}}, {{data = "\000\016\000 ", etl_alignment_type = 536874496}}, {{data = "\000\017\000 ",
          etl_alignment_type = 536874752}}, {{data = "\000\020\000 ", etl_alignment_type = 536875008}}, {{data = "\000\021\000 ", etl_alignment_type = 536875264}}, {{
          data = "\000\022\000 ", etl_alignment_type = 536875520}}, {{data = "\000\023\000 ", etl_alignment_type = 536875776}}, {{data = "\000\024\000 ",
          etl_alignment_type = 536876032}}, {{data = "\000\025\000 ", etl_alignment_type = 536876288}}, {{data = "\000\026\000 ", etl_alignment_type = 536876544}}, {{
          data = "\000\027\000 ", etl_alignment_type = 536876800}}, {{data = "\000\030\000 ", etl_alignment_type = 536877056}}, {{data = "\000\031\000 ",
          etl_alignment_type = 536877312}}, {{data = "\000\032\000 ", etl_alignment_type = 536877568}}, {{data = "\000\033\000 ", etl_alignment_type = 536877824}}, {{
          data = "\000\034\000 ", etl_alignment_type = 536878080}}, {{data = "\000\035\000 ", etl_alignment_type = 536878336}}, {{data = "\000\036\000 ",
          etl_alignment_type = 536878592}}, {{data = "\000\037\000 ", etl_alignment_type = 536878848}}, {{data = "\000 \000 ", etl_alignment_type = 536879104}}, {{
          data = "\000!\000 ", etl_alignment_type = 536879360}}, {{data = "\000\"\000 ", etl_alignment_type = 536879616}}, {{data = "\000#\000 ",
          etl_alignment_type = 536879872}}, {{data = "\000$\000 ", etl_alignment_type = 536880128}}, {{data = "\270e\000 ", etl_alignment_type = 536896952}}, {{
          data = "\274e\000 ", etl_alignment_type = 536896956}}, {{data = "\300e\000 ", etl_alignment_type = 536896960}}, {{data = "\304e\000 ",
          etl_alignment_type = 536896964}}, {{data = "\310e\000 ", etl_alignment_type = 536896968}}, {{data = "\314e\000 ", etl_alignment_type = 536896972}}, {{
          data = "\320e\000 ", etl_alignment_type = 536896976}}, {{data = "\000\000\000", etl_alignment_type = 0}} <repeats 80 times>}, static ELEMENT_SIZE = 4}}
```

phillipjohnston, 2019-04-19:
Once I trigger this scenario in my code, it appears that I'm always vended these duplicate addresses. Happens in a separate function as well, using the same pool.

phillipjohnston, 2019-04-19:
90% sure this is a thread issue on my end.

phillipjohnston, 2019-04-20:
Well, I put locks around all create/destroy calls and still ended up seeing the same behavior. Hm.

jwellbelove, 2019-04-20:
Hi Phillip, I'll take a look at it this morning.

jwellbelove, 2019-04-20:
I've tried various new scenarios in the unit tests, but I don't seem to be able to reproduce the problem.
etl::variant_pool uses etl::pool, and that class is used for all of the containers except vector & deque. I've had no reports of issues with those containers.
Can you step through **etl::pool::allocate_item()** and watch the values of **p_value**, **p_next** and **items_allocated**?
At the moment I'm not able to see how **p_next** can end up with the same address for two consecutive calls to **create**, unless there has been a **destroy** between the two.

The pool operates as a linked list, where unused pool elements contain the address of the next free element. New elements are allocated from the front of the list (**p_next**)  and released to the front.
Therefore, if an item is created/destroyed/created the returned addresses will be the same.

I have run the unit tests under MS Visual Studio 2017 and GCC 8.1.0

phillipjohnston, 2019-04-22:
Thanks for checking John. I also though it was a threading problem, but was really confused when locking did not help. 

After a few long hours of debugging, I narrowed it down to a problem with lambda captures that I don't yet understand. I was capturing the pointer in the lambda and passing it to destroy<> later in a callback function. When I changed the functions to destroy via an argument supplied to the callback function, everything works great. 

I don't have a great hypothesis for why this was causing duplicate pointers to be vended at a later time...

phillipjohnston, 2019-04-26:
Closing this, I think it's a problem on my end.

closed 2019-04-26 by phillipjohnston

phillipjohnston, 2019-04-30:
Just to close the loop here for posterity: the issue was using a callback and capturing stack variables by reference. I needed to capture the stack variables by copy. Accessing the variables in a callback lambda was grabbing random values from the stack, since I had left the frame of the original call.

jwellbelove, 2019-04-30:
Glad you got to the bottom of it.