ETLCPP/etl #130: Usage of ::new

embeddedteam103, 2019-05-07:
Is there any reason for usage of `::new` instead of `new` through the library?

It seems like IAR can't recognize that..
We're using `etl` alongside `mbed-os`, so it might cause the issue as `mbed-os` overrides
`operator new(std::size_t count)`
**however**, it does not override `operator new(std::size_t count, void *)`
so we believe it just a compiler specific bug.

Either way, it would be helpful to replace `::new` with `new` if it doesn't hurt functionality.

jwellbelove, 2019-05-07:
I'm pretty sure that I used 'new' originally, but an issue was raised about changing it to '::new'. I'm sure there was a very good reason, but I can't remember what it was. I'll look back over the issues to see if I can find it.

crespum, 2019-05-31:
I'm not sure if it's related to this issue, but I can't compile this simple example:
```cpp
#include "etl/queue.h"

int test() {
    etl::queue<int, 10> queue;
    queue.push(2);
    return queue.size();
}
```

I'm getting the following error:
```
In file included from Src/test.cpp:1:
etl/include/etl/queue.h: In instantiation of 'void etl::iqueue<T, MEMORY_MODEL>::push(etl::iqueue<T, MEMORY_MODEL>::parameter_t) [with T = int; unsigned int MEMORY_MODEL = 2; etl::iqueue<T, MEMORY_MODEL>::parameter_t = int]':
Src/test.cpp:5:17:   required from here
etl/include/etl/queue.h:315:7: error: no matching function for call to 'operator new(sizetype, int*)'
       ::new (&p_buffer[in]) T(value);
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
<built-in>: note: candidate: 'void* operator new(unsigned int)'
<built-in>: note:   candidate expects 1 argument, 2 provided
make: *** [Makefile:210: build/test.o] Error 1
```

Any idea?

jwellbelove, 2019-05-31:
I think I've seen this issue before. Can you try adding **#include <new>** to the list of includes in queue.h?
Which compiler and platfor are you working with?

crespum, 2019-05-31:
Thanks @jwellbelove. That include fixes the error. Do you know what's going on?

I'm using GCC for ARM (8-2018-q4-major). The board is an STM32F4 Disovery and this is my `etl_profile.h`:

```c
#ifndef __ETL_PROFILE_H__
#define __ETL_PROFILE_H__

#define ETL_VERBOSE_ERRORS
#define ETL_CHECK_PUSH_POP
#define ETL_ISTRING_REPAIR_ENABLE
#define ETL_IVECTOR_REPAIR_ENABLE
#define ETL_IDEQUE_REPAIR_ENABLE
#define ETL_CALLBACK_TIMER_USE_ATOMIC_LOCK
#define ETL_NO_STL

// ARM6 compiler
#if defined(ETL_NO_STL)
#include "etl/profiles/armv6_no_stl.h"
#else
#include "etl/profiles/armv6.h"
#endif

#endif
```
The compiler command is:
```
arm-none-eabi-g++ -c -mcpu=cortex-m4 -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -DUSE_HAL_DRIVER -DSTM32F407xx -IInc -IDrivers/STM32F4xx_HAL_Driver/Inc -IDrivers/STM32F4xx_HAL_Driver/Inc/Legacy -IDrivers/CMSIS/Include -Ietl/include  -Og -Wall -fdata-sections -ffunction-sections -g -gdwarf-2 -MMD -MP -MF"build/test.d"  -std=c++11 -fabi-version=0 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -fno-threadsafe-statics -Wa,-a,-ad,-alms=build/test.lst Src/test.cpp -o build/test.o
```

jwellbelove, 2019-05-31:
The error was because there was no definition for the 'placement new' operator. I should probably add the header to all of the files that use placement new.

crespum, 2019-06-02:
This might be a stupid question, but why does `queue.push()` need standard `new` if the ETL doesn't use dynamic memory allocation?

radarjones, 2019-06-03:
@crespum 

https://en.cppreference.com/w/cpp/language/new#Placement_new

ETL uses fixed block memory pools that are allocated at compile time.  When you put an item into any ETL container it allocates a block of memory from the container's internal pool. Placement new is then use to  construct the object at the give address allocated from the pool. 

See also https://en.wikipedia.org/wiki/Memory_pool

closed 2020-03-28 by jwellbelove