ETLCPP/etl #133: Comparison operators not in namespace

jpan127, 2019-06-25:
Hi, I recently started using these containers `etl::{ set , unordered_set , map , unordered_map }` in larger structures / containers.  However, these outer containers require the comparison operators of these etl containers.

The compiler could not find these operators for `etl::{ set , map}`, I believe because the operators are in the global namespace not the `etl` namespace so ADL fails.

Maybe other containers are namespaced like this too, I did not check.  Please let me know if you agree, I am happy to make a PR!

jwellbelove, 2019-06-25:
Yes, I put all of the non-member operators in the global namespace (some time ago). I had a read up on ADL (to be honest, I hadn't really looked too hard at ADL rules up until now) and I can see that I should have put them in the ETL namespace.

jpan127, 2019-06-26:
Cool! So would you like me to assist in making the change?

jwellbelove, 2019-06-27:
That would be a great help. It should be a fairly simple cut & paste job. We just have to keep track of what's been converted. We should keep an updated list of converted classes, either here or on Slack, while the work is in progress.

jwellbelove, 2019-06-27:
Can you make sure you make the changes in a 'feature' or 'hotfix' branch, rather than 'master', thanks.

jpan127, 2019-06-28:
Made a PR in a feature branch, please take a look whenever!

jwellbelove, 2019-06-28:
I may not be able to look until Moday, as I'm off to a party in on a farm in Portesham, Dorset this weekend.

jpan127, 2019-06-29:
Have fun!!

closed 2019-06-29 by jpan127