ETLCPP/etl #136: CI pipeline

jpan127, 2019-06-28:
When running the tests for my last PR, I noticed there is no continuous integration pipeline.  Have you considered setting up Travis?

jwellbelove, 2019-06-28:
I looked at travis a while ago, but I never had the time to figure out how to get it set up.
The library tests are run on Appveyor on every push to master (Appveyor uses the VS project file) and I think they are offering a linux setup now.
There is a clone of the library on Gitlab and the guys there have set up a travis CI.

jpan127, 2019-06-29:
I see, I have never used Appveyor since I don't do much development on Windows.  For linux based continuous integration, Travis is great.

I would recommend enabling PR based integration, for example on my most recently merged PR, we didn't know for sure if it builds or not (what if it doesn't build on Windows and only builds on my Ubuntu?).

If you want to enable Appveyor with Linux, I probably can't help but that would be great!  If you want to enable Travis with Linux, I can help with the scripting side, but the owner of the repository has to link Travis to his Github + the repository.  (Pretty seamless, just create a Travis account and it will take you through all the steps)

jwellbelove, 2019-06-29:
I'm pretty sure I've already setup an account on Travis, just never used it yet. I've spent most of my life creating embedded applications using Windows base tools, so I don't have a lot of experience in setting up a Linux environment. I have a Ubuntu VM on my home machine which I sometimes use to check compilation of the ETL. I don't use it often as I lose patience trying to figure out what command line 'incantations' I have to use to get it to compile/install/update what I want. I'm not unfamiliar with CLI's. I spent over a decade workng with computers long before GUI's even existed. It's just that whenever I try to do something on Linux, I find there's another half dozen *other* things I have to figure out before I can get any work done. Searching online for help usually comes up with solutions that either don't seem to apply to my 'distro', or require me to learn even more new stuff to understand the 'solution'.

jwellbelove, 2019-06-29:
I found the Travias account. Looks like I set it up about two years ago.
Another thing that has delayed the setting up of a Linux CI is that I've never, in my 30 years of programming, had to deal with CMake or Unix make. I used some version of make files on DOS back in the early 1990's, but I've not had to bother with them since.

jpan127, 2019-07-10:
Yea, linux does have a more of a steep learning curve at first, but from my limited experience going from windows to ubuntu, the freedom/flexibility has increased drastically.  CMake is trivial to set up, I believe someone else is already maintaining it for this repository?

Well anyways, if you would like help setting this up, you can always let me know!

peter-moran, 2019-08-28:
Another option is to perform cross-platform testing using docker

jwellbelove, 2020-03-31:
I set up a CircleCI account a while ago and I'm trying to get this up and running. Getting the configuration files right has been a bit of a trial. Even my wife, whose a Unix/Solaris admin, couldn't figure out where I was going wrong. I shall keep trying...

puzrin, 2020-07-17:
https://docs.github.com/en/actions/building-and-testing-code-with-continuous-integration/setting-up-continuous-integration-using-github-actions

You could do CI direcly on github. I use Travis for along time. But, to be honest, for multi-os testing Gihub Actions seems more convenient and fast.

https://github.com/actions/starter-workflows/issues/209 - this sample should be ok for start with cmake/ctest. Then extend matrix with more OS-es, compilers and so on.

jwellbelove, 2020-07-17:
Yes, I'd heard that GitHub had introduced CI, but I haven't had time to check it out yet. CI is successfully running on Appveyor and CircleCI at the moment.

puzrin, 2020-07-17:
As fas as i understand, some of regressions are caused by lack of runing tests with previous gcc & clang versions.

I could prepare draft in fork (or PR), if someone could explain how to force run cmake with specific toolchain under linux.

jwellbelove, 2020-07-28:
We now have a Github Action for compiling under Ubuntu Linux

finger42, 2020-07-29:
From the Github Actions  the following machines are supported:

- Windows Server 2019 
- Ubuntu 20.04 
- Ubuntu 18.04 
- Ubuntu 16.04 
- macOS Catalina 10.15

puzrin, 2020-09-17:
https://github.com/ETLCPP/etl/blob/master/.github/workflows/main.yml#L45-L49

IMO saving artefacts not required. And those are overriden by each job. Can anyone explain why this was added?

puzrin, 2020-09-17:
https://github.com/puzrin/etl/tree/workflow-rework

I restructured workflow.

- Removed garbage (artefacts upload)
- AFAIK we need compiler tests, and here are no OS-dependent code. So testing in multiple os-es the same compiler not needed (but can be added).
- It's not good idea to run CI tests on `master` only. Removed restriction.

Found problem:

- gcc-4.8, gcc-5, gcc-6 can not be tested now (fails with message C++ 17 support required).

Many embeded compilers still based on gcc-4.8 & gcc-5. IMO it would be nice no make tests work with those somehow (may be resricted subset).

Feel free to reuse that file as you wish or ask any questions.

```yaml
name: CI
on:
  push:
  pull_request:

jobs:
  build-gcc-linux:
    runs-on: ${{ matrix.os }}
    strategy:
      fail-fast: false
      matrix:
        os: [ ubuntu-18.04, ubuntu-20.04 ]
        # Tests fail with 6, 5 & 4.8 (not supported)
        #gcc_version: [ 10, 9, 8, 7, 6, 5, '4.8' ]
        gcc_version: [ 10, 9, 8, 7 ]
        exclude:
          # Exclude duplicated versions
          - { os: 'ubuntu-20.04', gcc_version: '4.8' }
          - { os: 'ubuntu-20.04', gcc_version: 5 }
          - { os: 'ubuntu-20.04', gcc_version: 6 }
          - { os: 'ubuntu-18.04', gcc_version: 7 }
          - { os: 'ubuntu-18.04', gcc_version: 8 }
          - { os: 'ubuntu-18.04', gcc_version: 9 }
          - { os: 'ubuntu-18.04', gcc_version: 10 }

    steps:
      - name: Install compiler
        run: |
          sudo apt-get install -y gcc-${{ matrix.gcc_version }} g++-${{ matrix.gcc_version }}

      - name: Checkout
        uses: actions/checkout@v2

      - name: Build and run
        run: |
          export CC=gcc-${{ matrix.gcc_version }}
          export CXX=g++-${{ matrix.gcc_version }}
          cmake -D BUILD_TESTS=ON ./
          make
          ./test/etl_tests

  build-clang-linux:
    runs-on: ${{ matrix.os }}
    strategy:
      fail-fast: false
      matrix:
        os: [ ubuntu-20.04 ]
        clang_version: [ 10, 9, 8 ]

    steps:
      - name: Install compiler
        run: |
          sudo apt-get install -y clang-${{ matrix.clang_version }}

      - name: Checkout
        uses: actions/checkout@v2

      - name: Build and run
        run: |
          export CC=clang-${{ matrix.clang_version }}
          export CXX=clang++-${{ matrix.clang_version }}
          cmake -D BUILD_TESTS=ON ./
          make
          ./test/etl_tests
```

closed 2020-10-10 by jwellbelove