ETLCPP/etl #139: Dangling reference in etl::delegate with capturing lambdas

JoelFilho, 2019-08-09:
The binding of const lvalue reference to an rvalue reference in the constructor `delegate(const TLambda& instance)` invokes undefined behavior when passed a capturing lambda (possibly crashes at optimization levels greater than 0).

I imagine similar behavior could happen in `create<T, TReturn(T::*)(TParams...)>` as well, but I believe the lambda case is probably more susceptible to happen in real code, especially since it's constructible from the `=` operator.

Reproducible example (+[execution on Compiler Explorer](https://godbolt.org/z/Nnst-P)): 

```c++
void external_call(etl::delegate<void(int)> d){
    std::cout << "  external_call: ";
    d(1);
}

int main() {
  // lvalue case
  bool ok = false;
  std::cout << "lvalue:" << std::endl;
  auto l = [&](int i) {
    ok = true;
    std::cout << "  d1(" << i << ")" << std::endl;
  };
  auto d1 = etl::delegate<void(int)>(l);
  d1(0);
  external_call(d1);

  // rvalue case
  ok = false;
  std::cout << "rvalue:" << std::endl;
  auto d2 = etl::delegate<void(int)>([&](int i) {
    ok = true;
    std::cout << "  d2(" << i << ")" << std::endl;
  });
  d2(2);
  external_call(d2);
}
```

jwellbelove, 2019-08-10:
I'm not by any means an expert on rvalue references and lambdas and the code was a slightly modified version of an article on CodeProject, so I only have a tentative understanding of why the code goes wrong with capturing lambdas. I imagine the error may be something to do with the original lambda going out of scope. Is that true?
Do you have any siggestions on how this may be fixed (other than noting that using delegate this way is undefined behaviour)?

JoelFilho, 2019-08-10:
> I imagine the error may be something to do with the original lambda going out of scope. Is that true?

Yes, since it's a temporary object, it's destroyed right after the delegate construction, as there's no lifetime extension. 

> Do you have any suggestions on how this may be fixed (other than noting that using delegate this way is undefined behaviour)?

I was trying some alternatives, and the simplest way I found is changing the signature of the constructor to:

```c++
template <typename TLambda>
    delegate(TLambda& instance);
```

This way, it should prevent the r-value case from happening (substitution would fail), but still deduce constness from lambdas declared const. Same solution should apply in the `create<TLambda>` function.

---

And I checked the `create<T, TReturn(T::*)(TParams...) const>(const T&)` case, the same problem can also happen in the r-value case, although a little harder to happen in real code:

```c++
struct S { void run() const {} };
etl::delegate<void()>::create<S, &S::run>(S{});
```

The solution for this one is probably cleaner, since the parameter T is explicitly declared and the compiler won't deduce a forwarding/universal reference:

```c++
template <typename T, TReturn(T::*Method)(TParams...) const>
    static delegate create(T&& instance) = delete;
```

jwellbelove, 2019-08-10:
```
template <typename TLambda>
    delegate(TLambda& instance);
```

Unfortunately this gets called when instead of the copy constructor in this test...
```
      Test test;

      etl::delegate<void(int, int)> d1 = etl::delegate<void(int, int)>::create<Test, &Test::member_int>(test);
      etl::delegate<void(int, int)> d2 = d1; // Lambda constructor called.

      CHECK(d1 == d2); // FAILS
```

JoelFilho, 2019-08-10:
True. Sorry about that, I didn't run the tests.

That happens because `TLambda&` deducing `delegate` is a better fit for `delegate&` than `const delegate&`.

Explicitly defining the copy constructor and the copy assignment operator for the non-const `delegate&` would be a quick solution. e.g.:

```c++
    //*************************************************************************
    // Copy constructor.
    //*************************************************************************
    delegate(const delegate& other)
    {
      other.invocation.clone(invocation);
    }

    delegate(delegate& other)
    {
      other.invocation.clone(invocation);
    }
```

Other way would be using SFINAE in the TLambda cases to guarantee they're not being called with instances of delegate, but that would bring the type_traits dependency.

jwellbelove, 2019-08-13:
Yes, it looks like that is true.

NickolasLapp, 2020-03-21:
Did a fix for this ever land? I just ran into the same issue with capturing lambdas.

jwellbelove, 2020-03-21:
I'm afraid I haven't looked at this issue for a while.

jwellbelove, 2020-03-25:
After giving it some thought (and a bit of experimentation) I'm not sure how capturing lambdas, that go out of scope, can ever work with `etl::delegate` (unless I'm misunderstanding how lambdas work).

etl::delegate can not make a copy of the lambda. The internal function stubs use a common 'invocation' structure that only stores a `void*` to the function (whether it be global, static, member or lambda).

JoelFilho, 2020-03-25:
You're right, @jwellbelove. 

`etl::delegate` is supposed to be non-owning, unlike `std::function`, which uses heap allocations along with type erasure for an owning solution. `etl::delegate`'s solution is closer to [`tl::function_ref`](https://github.com/TartanLlama/function_ref). 

The lambda case is similar to the "member function caller" overload of `delegate`: The user is supposed to manage the lifetime of the object for as long as the delegate is valid. 

My concern when creating this issue was the rvalue case, where there was no documentation about it, nor anything in the API preventing it. 

Note that allowing the rvalue case is a perfectly valid choice for an API: it allows you to immediately invoke the lambda by passing to a generic interface. That was the decision they took in `tl::function_ref`. 

But, if `etl::delegate` is supposed to be stored for future calls, I believe disallowing the passing of rvalue capturing lambdas directly on the API would be the right choice. I also believe the solution you committed at the time would work. 

---

A side note: the use case behind this issue was the a "task scheduler" I developed, similar to FreeRTOS's software timer functionality, using the same delegate base for representing tasks. In my case, the objects were stored for future execution, so it made sense to prevent the rvalue lambda case.

jwellbelove, 2020-03-25:
I haven't been able to come up with a solution yet that either compiles or passes the unit tests.

closed 2020-03-29 by jwellbelove