ETLCPP/etl #144: Automatically perform `conan upload` on each release

peter-moran, 2019-08-28:
To make #143 work more smoothly, we should have the current CI framework perform `conan upload` on each release.

Does the CI have hooks to do this?

jwellbelove, 2020-03-28:
I've just had a quick look at this.
I've installed conan on my PC and I've experimented with variations of 'conan upload' but no luck so far.
Assuming I'm in the 'etl' directory which contains a 'conanfile.py', what CLI syntax should I be using?

peter-moran, 2020-04-07:
First of all, if you want to make a first-class conan package that all conan users can easy install by default (this is very easy for a header only library like this), then I would love to see this package on `conan-center`, the default conan repository. In this setup the conanfile will exist in their repository and they will automatically run CI steps to upload packages to the repository. The [instructions for getting started with this are here](https://blog.conan.io/2019/09/04/A-new-binary-building-service-for-Conan-center-packages.html).

If, instead, you want to use a different conan package repository, you will need to:
1. Choose/obtain a conan repository to push to. You can either:
    1. Use [Bincrafters Public Conan Repository](https://bincrafters.github.io/2017/06/06/using-bincrafters-conan-repository/) which I think you can upload to for free
    2. Make your own account on Bintray and add a conan repository under your account (this may not be free).
2. Set up the conan remote (both methods above should say how, but basically you would do `conan remote add <name> <url>`
3. Create the `etl` conan package by (from the project root) running `conan create . <username>/<channel>`. The username and channel are optional, but help prevent any collisions between multiple people trying to package the same version of the same package. For testing, you might want to do `conan create . jwellbelove/testing`
4. Now that you have a package to upload, you can call conan upload to your repository: `conan upload '*etl*' -r <your_remote_name>`

jwellbelove, 2020-04-07:
I'll check, but I'm pretty sure I created a repository on Bintray some time ago.

jwellbelove, 2020-04-07:
Yes, I did.
https://bintray.com/etl

jwellbelove, 2020-04-07:
conan create returns withe error `ERROR: conanfile didn't specify version`
I assume `get_version_from_git_tag() `is returning 'None'

jwellbelove, 2020-04-07:
By forcing the version to the latest "17.4.2" `conan create` ran without errors.

jwellbelove, 2020-04-07:
Also success with `conan remote add embedded-template-libray https://api.bintray.com/conan/etl/ETL`

jwellbelove, 2020-04-07:
Got as far as `conan upload '*embedded-template-library*' -r embedded-template-library`
but conan refuses to accept my Bintray username and/or password.

jwellbelove, 2020-04-07:
Can I add you to the Bintray organisation?